zwave-js/zwave-js #2981: Fibaro RGBW 441 switch has no color wheel in HA (missing Multilevel Switch value IDs for endpoint 0)

bart1970, 2021-07-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Configuration file is missing compat flag

What do you observe?
No color wheel in HA

What did you expect to happen?
Have a entity with a color wheel

Steps to reproduce the behavior:
Add fibaro FGRGBW (version 441) to your system

Solution:
Add: 
`,"compat": {`
`                 // This device supports RGBW control on the root endpoint`
`		"preserveRootApplicationCCValueIDs": true
	}`
to the file devices/0x010f/fgrgbw.json

For corrrect file and tests, see also:
[https://github.com/home-assistant/core/issues/47888](url)

### Device information

Manufacturer:  Fibaro
Model name:  FGRGBW
Node ID in your network: 7

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 7.10.1
`zwavejs2mqtt` branch: 5.2.1

### Did you change anything?

no

### If yes, what did you change?

Solution described above

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Worked in old Zwave integration

### Attach Driver Logfile

Fixed by adding correct file (see also [https://github.com/home-assistant/core/issues/47888](url))

2021-07-05T18:12:00.629Z CNTRLR [Node 007] User-provided device config loaded

AlCalzone, 2021-07-06:
@zwave-js-bot add compat flag to 0x010f:0x0900:0x1000:0.0
```json
// This device exposes the RGBW control on the root endpoint
"preserveRootApplicationCCValueIDs": true
```

AlCalzone, 2021-07-06:
Should be fixed in today's nightly config or the next official release.
You'll need to re-interview the device probably.

closed 2021-07-06 by AlCalzone

Kurisutian, 2021-07-06:
Does this also fix the 442 device?

bart1970, 2021-07-06:
> Does this also fix the 442 device?
No, the update is for the 441. For me, the 442 is working as it should be, so that must be another problem.

Extreme79, 2021-07-15:
For me not solved unfortunately. At least apparently.
All updated and rebooted, new entity is discovered (supporting RGBw, but it's unavailable and any call to light.turn_on fail)
[image]
[zwave_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6823872/zwave_dump.zip)

gzone156, 2021-07-15:
> For me not solved unfortunately. At least apparently.
> All updated and rebooted, new entity is discovered (supporting RGBw, but it's unavailable and any call to light.turn_on fail)
> [image]
> [zwave_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6823872/zwave_dump.zip)

For me it's ok with latest releases (zwave-js 7.12 and HA 2021.7.2), either without doing a node re-interview.

[image]

Extreme79, 2021-07-15:
> > For me not solved unfortunately. At least apparently.
> > All updated and rebooted, new entity is discovered (supporting RGBw, but it's unavailable and any call to light.turn_on fail)
> For me it's ok with latest releases (zwave-js 7.12 and HA 2021.7.2), either without doing a node re-interview.
> You have to turn on the light first to get the color wheel

Sure I know I have to turn on.. but the RGBw entity is shown as unavailable and any try to turn on fail.
The other entity works and turn on the device (without the color control).
This is driving me crazy.
Can you show my your fibaro params? What's your fw revision (mine is 22.22, damn fibaro that refuse to release fw binaries)?

If your works.. mine must works 🤐

Extreme79, 2021-07-17:
No way. I removed the integration, deleted the entities, rebooted, done all of this many times.. but same result.
All the endpoints are discovered.. but the RGBw one is not responding. I want to cry.
When it say that is unavailable without showing the palette, anyway, moving the slider change the brightness!! 🥴😫

bart1970, 2021-07-17:
Not at home, so did not yet got the possibility to test. Will let you know as soon as I have.

bart1970, 2021-07-20:
@Extreme79, tested it today. Removed my additional config file so that I'm, using the one from the latest zwave-js-mqtt now. For me it works. Unfortunately, I have to little knowledge on the exact working of the zwave to help you on why your endpoint stays unavailable.

Extreme79, 2021-07-22:
I'll try again when I'll be back to home.
It's so strange, I' starting to suspect that something is messing in assistant registry but I'm unsure, I probably need someone expert in home assistant.

PR #2982 (2021-07-06): fix(config): preserve root endpoint values on Fibaro RGBW 441
fixes: #2981