zwave-js/zwave-js #2995: Missing device configuration: Schlage BE469NX

orrious, 2021-07-08:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Schlage BE469NX

### Manufacturer ID

0xXXXX

### Product Type

0xXXXX

### Product ID

0xXXXX

### Is the device listed on the Z-Wave Alliance website?

This specific model is not listed

### Do you have a manual?

https://www.schlage.com/content/dam/intg-feat/Documents/24060857.pdf

orrious, 2021-07-08:
Log of re-interview of device with silly logging:

https://paste.ubuntu.com/p/QgDPV5XZDv/

AlCalzone, 2021-07-08:
> ### Manufacturer ID
> 
> 0xXXXX
> ### Product Type
> 
> 0xXXXX
> ### Product ID
> 
> 0xXXXX
> ### Firmware Version
> 
> _No response_

Can't do much without this information 🤷🏻‍♂️

orrious, 2021-07-09:
The issue was resolved by placing the zwave controller closer to the device.  The device then responded properly.

closed 2021-07-09 by orrious