zwave-js/zwave-js #2998: GE Jasco 46202 missing parameter to configure LED behavior

kurtzettel, 2021-07-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

I have a bunch of GE/Jasco switches. One of them is a dimmer:
https://github.com/zwave-js/node-zwave-js/blob/5539dd48f967e74f8afe131dd896f952269e42c1/packages/config/config/devices/0x0063/14295_46204_zw3011.json#L34

That dimmer lets me configure the LED behavior in the UI because of parameter 3. I also have some switches, the GE Jasco 46202:

https://github.com/zwave-js/node-zwave-js/blob/5539dd48f967e74f8afe131dd896f952269e42c1/packages/config/config/devices/0x0063/46202.json

These allow me to “Invert Switch” in the Home Assistant UI but not set the “LED Light” behavior. I’m guessing because the second device config doesn’t have parameter 3 defined. I can see the device supports it because when I interrogate it I get this in the log:
```
2021-07-09T21:51:41.564Z CNTRLR « [Node 021] received information for parameter #3:
                                  parameter name:      LED Light
                                  value format:        SignedInteger
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           3
                                  default value:       2
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
```

What do you observe?
I can configure the LED Behavior on my GE/Jasco 14295 / 46204 / ZW3011 but not on my GE Jasco 46202

What did you expect to happen?
LED Behavior is configurable in the Home Assistant UI.

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: GE/JASCO
Model name: 46202
Node ID in your network: 8, 20, 21

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

[... 92 lines cut ...]
                                  value format:        SignedInteger
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           1
                                  default value:       0
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
2021-07-10T16:29:46.884Z CNTRLR » [Node 008] querying parameter #19 information...
2021-07-10T16:29:47.700Z CNTRLR « [Node 008] received information for parameter #19:
                                  parameter name:      Alternate Exclusion
                                  value format:        SignedInteger
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           1
                                  default value:       0
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
2021-07-10T16:29:47.706Z CNTRLR » [Node 008] querying parameter #4 value...
2021-07-10T16:29:47.771Z CNTRLR   [Node 008] Interviewing Association...
2021-07-10T16:29:47.771Z CNTRLR » [Node 008] querying number of association groups...
2021-07-10T16:29:47.817Z CNTRLR « [Node 008] supports 3 association groups```

AlCalzone, 2021-07-11:
@kurtzettel would you be willing to provide a PR for the necessary change?

kurtzettel, 2021-07-11:
> @kurtzettel would you be willing to provide a PR for the necessary change?

I don't have this in a dev environment. I only have it in Home Assistant in a supervised container. Since I can't test it, I didn't feel comfortable doing the PR.

blhoward2, 2021-07-15:
That parameter does not exist on all GE switches/dimmers, and it isn't listed for yours. It does exist on some where it isn't documented, however. Has this ever worked on other platforms? Can you give us some background as to why you think this should be possible? Happy to add it if it works but we're not going to add it if it doesn't actually work.

kurtzettel, 2021-07-18:
I am able to configure this behavior with SmartThings with no addon. In SmartThings they have a configuration called "LED Indicator" with the options: "When On", "When Off", "Never".  The only thing I see in my log is that it is included:
```
2021-07-10T16:29:45.908Z CNTRLR » [Node 008] querying parameter #3 information...
2021-07-10T16:29:46.329Z CNTRLR « [Node 008] received information for parameter #3:
                                  parameter name:      LED Light
                                  value format:        SignedInteger
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           3
                                  default value:       2
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
```

I am running the add-on Z-Wave JS 0.1.29. I was running 0.1.28 when I logged this issue.

Can you direct me to any documentation on how to test this with zwavejs2mqtt? Would that require me to setup something besides home assistant on my pi?

kurtzettel, 2021-07-18:
[image]

blhoward2, 2021-07-18:
If you switch to the zwavejs2mqtt addon there is an advanced option under the node that can set any value on any parameter, unbounded by the device files. You can switch using the same integration and without losing anything but you'd need to see the HA zwave discord channel for help doing so. It's easy to do I just don't know the exact steps.

DaDre, 2021-07-20:
I can confirm everything @kurtzettel is saying. I have the same 46202 switches that I can't set LED settings on but was able to back when I had SmartThings. I have several of them, had the settings set as I wanted to in SmartThings and those carried over to HA, but now bought another one to expand and figured out that I can't set its settings. Came looking for an answer, and found this recently open issue. 

I can further confirm that I'm able to control the LED settings on 14295 / 46204 / ZW3011 GE Smart Toggle Dimmers:
[image]

Hoping the above pull request gets implemented and resolves this. Thanks, @kurtzettel  for submitting this.

AlCalzone, 2021-07-20:
> those carried over to HA, but now bought another one to expand and figured out that I can't set its settings

That sounds like these devices have two different fingerprints (manufacturer ID, product type and ID, `0xabcd:0x1234:0x5678`). Can you share those please?

blhoward2, 2021-07-20:
Yes think he means the parameter was set and he didn't reset the device.

DaDre, 2021-07-20:
The latest one I bought is showing `0x0063-0x4944-0x3131` my older switches appear to be showing `0x0063-0x4952-0x3137`

AlCalzone, 2021-07-21:
Okay, these are two different config files:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0063/46202.json
and
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0063/14287.json

@DaDre Does param 3 exist on both?

DaDre, 2021-07-21:
> Does param 3 exist on both?

@AlCalzone I'm certain it does, but not sure how to fully validate it.

AlCalzone, 2021-07-21:
Zwavejs2mqtt has a field where you can manually enter a param number and read or write it

DaDre, 2021-07-23:
Was able to read param 3, which was set to the value of 2. Pushed value of 0 to it and the new switch lit right up:

[image]

AlCalzone, 2021-07-23:
Thanks, PR is merged

PR #3049 (2021-07-23): fix(config): add missing LED Light param to GE/Jasco 46202 and 14292
Add LED param for 46202

Closes #2998
closed 2021-07-23 by AlCalzone