zwave-js/zwave-js #3001: Switch reacts very slow since zwavejs2mqtt update from 4.x.x to 5.2.1

Jandokus, 2021-07-11:
Recently I update zwavejs2mqtt from 4.x.x to 5.2.1. After the update all my devices reacts very slow. Because I have almost only devices from GreenWave (Single and Multi PowerNodes), I cannot determine if this is device specific.
When I was using version 4.x.x there was no problem.

This afternoon I send a ON command at 17:50:52 and my lamp will be switched to ON at 17:55:40 (almost 5 minutes later).

[image]

My log file:

2021-07-06 17:50:52.889 INFO ZWAVE: Calling api writeValue with args: [
 {
    nodeId: 22,
    commandClass: 37,
    endpoint: 0,
    property: 'targetValue'
  },
  true,
  {},
  [length]: 3
]
2021-07-06 17:55:40.532 INFO ZWAVE: Node 22: value updated: 37-0-currentValue false => true
2021-07-06 17:55:40.535 INFO ZWAVE: Success zwave api call writeValue undefined
2021-07-06 17:55:41.742 INFO ZWAVE: Node 22: value updated: 50-0-value-66049 0 => 11.3
2021-07-06 17:55:43.242 INFO ZWAVE: Node 22: value updated: 50-0-value-66049 11.3 => 4.7
2021-07-06 17:55:53.745 INFO ZWAVE: Node 22: value updated: 50-0-value-66049 4.7 => 5
2021-07-06 17:56:11.215 INFO ZWAVE: Node 22: value updated: 50-0-value-66049 5 => 5.1
2021-07-06 17:56:12.391 INFO ZWAVE: Node 22: value updated: 37-0-currentValue true => true
2021-07-06 17:56:49.242 INFO ZWAVE: Node 22: value updated: 50-0-value-66049 5.1 => 5.3

Hopefully somebody reconize this problem and want to help me.

Markourai, 2021-07-11:
@AlCalzone here are the log in pdf (I hope it's better). 
[zwavejs_ko_e5efe5e.pdf](https://github.com/zwave-js/node-zwave-js/files/6796917/zwavejs_ko_e5efe5e.pdf)
[zwavejs_ok_ee4343f.pdf](https://github.com/zwave-js/node-zwave-js/files/6796918/zwavejs_ok_ee4343f.pdf)

AlCalzone, 2021-07-11:
@Markourai just download the logfile:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=download-zip

Markourai, 2021-07-20:
@AlCalzone 
Sorry for this late answer, here are the logfiles
[zwavejs2mqtt_ok.zip](https://github.com/zwave-js/node-zwave-js/files/6846318/zwavejs2mqtt_ok.zip)
[zwavejs2mqtt_ko_e5efe5e.log](https://github.com/zwave-js/node-zwave-js/files/6846322/zwavejs2mqtt_ko_e5efe5e.log)
.

AlCalzone, 2021-07-20:
> zwavejs2mqtt_ko_e5efe5e.log

That's not the driver logfile

Markourai, 2021-07-21:
Here it is.
[zwavejs2mqtt_ko_e5efe5e.zip](https://github.com/zwave-js/node-zwave-js/files/6853884/zwavejs2mqtt_ko_e5efe5e.zip)

AlCalzone, 2021-07-21:
@Markourai I'll need some more context please. Which node am I looking for and how does working vs. not working show itself?

Markourai, 2021-07-22:
@AlCalzone 
For both versions, I switched Node 16, 9 and 14 for the tests.

I've put a polling of 60s on all devices.

AlCalzone, 2021-07-22:
Your node 8 seems to cause problems. Interestingly the node itself seems to have changed its behavior between the logs:

**OK:** zwave-js queries endpoint 3, the node responds from endpoint 3
```
2021-07-12T16:14:25.101Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      14
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 3
                                    └─[MeterCCGet]
                                        rate type: Consumed
                                        scale:     W
...
2021-07-12T16:14:25.163Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      3
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       W
                                        rate type:   Consumed
                                        value:       0
                                        time delta:  9 seconds
                                        prev. value: 0
```

**NOK:** zwave-js queries endpoint 3, the node responds from the root endpoint, which is unrelated from the driver's perspective. Note that the outgoing command is identical, but the response is not.
```
2021-07-21T08:03:44.896Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      13
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 3
                                    └─[MeterCCGet]
                                        rate type: Consumed
                                        scale:     W
...

2021-07-21T08:03:44.961Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       W
                                        rate type:   Consumed
                                        value:       0
                                        time delta:  9 seconds
                                        prev. value: 0
```

Each of those attempts causes a pause of 10 seconds and since you seem to poll all endpoints, the node alone blocks all other polls for almost a minute. Then Node 8 is scheduled again.

We're working around that for the 6-port plug, could it be that your's is the 5-port one? Please share the device ID.

Markourai, 2021-07-22:
@AlCalzone yes node 8 is a 6-port greenwave NP240/NP242

Device ID: 
153-4-3 (0x0099-0x0003-0x0004)

AlCalzone, 2021-07-22:
Then it doesn't look like the compat flag is applied. Can you export the node information for node 8?
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-nodejson

Markourai, 2021-07-24:
@AlCalzone here is the export for node 8 (6 port plug)
[node_8.json.zip](https://github.com/zwave-js/node-zwave-js/files/6872498/node_8.json.zip)

AlCalzone, 2021-07-25:
Damn, that doesn't contain what I'm looking for. I've asked @robertsLando to add that info to the dumps.

Markourai, 2021-08-01:
Hi @AlCalzone,
I’ve tried latest tag yesterday or the day before and now it seems to work as before (and wanted).

robertsLando, 2021-08-02:
DeviceConfig is now there

AlCalzone, 2021-08-10:
@Markourai can you make a new export with the latest version?

closed 2021-08-21 by zwave-js-assistant[bot]