zwave-js/zwave-js #3011: Missing events/notifications/updates from Yale lock (YDM3168 with zwave module 2)

SuRGeoNix, 2021-07-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

_No updates/notifications/events will fire during unlocking/locking etc._

What do you observe?

_Nothing was received on low level (zwave-js logs). Even trying on higher level (HA dev tools) to listen for events (such as *, zwave_js_event, zwave_js_notification and zwave_js_value_notification) nothing fired._

**HA forum reference: https://community.home-assistant.io/t/z-wave-js-notifications-are-not-working/321234**

What did you expect to happen?

_See alarmtype, alarmlevel notifications._

Steps to reproduce the behavior:

_Try to unlock the door with a user pin and see if any notifications will be received._

### Device information

Manufacturer: Yale
Model name: YDM3168
Node ID in your network: 31

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 7.10.1
`zwavejs2mqtt` branch: 5.2.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2021-07-08.log](https://github.com/zwave-js/node-zwave-js/files/6815057/zwavejs_2021-07-08.log)
[zwave_js_dump.log](https://github.com/zwave-js/node-zwave-js/files/6815062/zwave_js_dump.log)

AlCalzone, 2021-07-14:
I don't see anything regarding lock/unlock notifications either. The device responds with its current status during the interview and when actuated via Z-Wave, but that's it. Doesn't look like something that can be fixed in the driver - maybe there's configuration on what the device is supposed to send via Z-Wave?

SuRGeoNix, 2021-07-14:
Tried to find more information on the ydm3168 manual and zwave module 2 with no success. So I guess is the device that doesn't send the notifications and I should try to contact yale for that? The lock has normal and advanced mode. I wasn't able to change user codes if I leave it in normal, requires to leave it in the advanced mode. Maybe there is another trick to configure it to send the notifications to the controller. Possible by settings exclusively the node id?

[image]

AlCalzone, 2021-07-14:
> Maybe there is another trick to configure it to send the notifications to the controller

If there is, it is probably under `Configuration` or by setting up a different association. Contacting support is probably a good idea.

SuRGeoNix, 2021-07-14:
By adding the controller (node 1) as new association to node 31 (yale lock) did the trick., seems that I receive the alarm events now Didn't test much as, I will close this for now and I will give an update if it is required after adding some automation / triggers etc. Thanks @AlCalzone

closed 2021-07-14 by SuRGeoNix

AlCalzone, 2021-07-14:
Was the association missing before or did you change it?

SuRGeoNix, 2021-07-14:
It was missing

AlCalzone, 2021-07-14:
That's weird, zwave-js is supposed to add it when interviewing the node. Maybe there was a connection problem.

SuRGeoNix, 2021-07-14:
I did that to other locks and had the same. I had it in close distance and interview had finished in less than 15'' sucessfully. Let me know if it helps to send you any logs by redoing this.

AlCalzone, 2021-07-14:
`v8.0.0` is going to revamp the association setup, so if it persists with that version, I'd be interested in seeing a log of a re-interview.

privatesam, 2021-11-01:
@SuRGeoNix would you mind explaining how you created the association between the lock and controller?  I'm using the yale keyless lock with the zwave module 2 aswell and I'm struggling to get the events too.  Admittedly I'm using zwavejs2mqtt and when I try and create the association I don't have anything under 'grou' to select:

[image]

SuRGeoNix, 2021-11-01:
Hey @privatesam, I think I have seen that when the interview was not completed successfully. Maybe re-interview (in close distance for secure protocols)?

privatesam, 2021-11-01:
@SuRGeoNix Ah! Hadn't thought of that.  A quick glance at zwavejs2mqtt control panel is showing that ` security:none ` so possibly thats something to do with what the device is reporting or not reporting to the zwave network - will investigate further when I get home.  Thanks so much for the idea and quick reply!