zwave-js/zwave-js #3012: DanaLock V3 keeps going from Alive to Dead

Nutti85, 2021-07-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

No idea

What do you observe?

Device drops out seemlingly at random

What did you expect to happen?

That the device keeps itself alive

### Device information

Manufacturer: 

Poly-control

Model name: 

Danalock V3-BTZE

Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Can't find it, but probably the latest as of date.

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

The device had no issues when running the old Home Assistant Z-wave integration that is now deprecated

### Attach Driver Logfile

```
2021-07-14T10:04:05.101Z CNTRLR   [Node 014] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
2021-07-14T10:04:09.119Z CNTRLR   [Node 014] Node 14 did not respond after 3 attempts, it is presumed dead
2021-07-14 12:04:09.151 INFO ZWAVE: Node 14 is now dead
2021-07-14T10:04:09.151Z CNTRLR   [Node 014] The node is now dead.
2021-07-14 12:04:16.455 ERROR ZWAVE-SERVER: Z-Wave error Failed to send the command after 3 attempts (Status NoAck)
ZWaveError: Failed to send the command after 3 attempts (Status NoAck)
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2271:23)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2405:28)
    at DoorLockCCAPI.set (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:294:21)
    at Proxy.DoorLockCCAPI.<computed> (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:140:15)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:752:14)
    at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:44)
    at Object.node (/opt/node_modules/@zwave-js/server/dist/lib/server.js:37:91)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:91:99)
    at WebSocket.<anonymous> (/opt/node_modules/@zwave-js/server/dist/lib/server.js:44:45)
    at WebSocket.emit (events.js:375:28)
2021-07-14T10:04:16.480Z CNTRLR » [Node 014] pinging the node...
2021-07-14T10:04:20.019Z CNTRLR   [Node 014] Node 14 did not respond after 1 attempts, it is presumed dead
2021-07-14T10:04:20.023Z CNTRLR   [Node 014] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-07-14 12:04:20.024 ERROR ZWAVE-SERVER: Z-Wave error The message cannot be sent because node 14 is dead
ZWaveError: The message cannot be sent because node 14 is dead
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2207:12)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2405:17)
    at DoorLockCCAPI.set (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:294:3)
    at Proxy.DoorLockCCAPI.<computed> (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:140:4)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:752:4)
    at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:33)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:91:62)
2021-07-14T10:04:20.028Z CNTRLR » [Node 014] pinging the node...
2021-07-14T10:04:23.576Z CNTRLR   [Node 014] Node 14 did not respond after 1 attempts, it is presumed dead
2021-07-14T10:04:23.601Z CNTRLR   [Node 014] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-07-14 12:04:23.603 ERROR ZWAVE-SERVER: Z-Wave error The message cannot be sent because node 14 is dead
ZWaveError: The message cannot be sent because node 14 is dead
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2207:12)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2405:17)
    at DoorLockCCAPI.set (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:294:3)
    at Proxy.DoorLockCCAPI.<computed> (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:140:4)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:752:4)
    at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:33)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:91:62)
2021-07-14T10:04:23.611Z CNTRLR » [Node 014] pinging the node...
2021-07-14T10:04:27.127Z CNTRLR   [Node 014] Node 14 did not respond after 1 attempts, it is presumed dead
2021-07-14T10:04:27.129Z CNTRLR   [Node 014] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-07-14 12:04:27.130 ERROR ZWAVE-SERVER: Z-Wave error The message cannot be sent because node 14 is dead
ZWaveError: The message cannot be sent because node 14 is dead
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2207:12)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2405:17)
    at DoorLockCCAPI.set (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:294:3)
    at Proxy.DoorLockCCAPI.<computed> (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:140:4)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:752:4)
    at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:33)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:91:62)
2021-07-14T10:04:27.161Z CNTRLR » [Node 014] pinging the node...
2021-07-14 12:04:30.664 INFO APP: GET /health/zwave 200 0.925 ms - 1875
2021-07-14T10:04:30.726Z CNTRLR   [Node 014] Node 14 did not respond after 1 attempts, it is presumed dead
2021-07-14T10:04:30.728Z CNTRLR   [Node 014] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack)
2021-07-14 12:04:30.731 ERROR ZWAVE-SERVER: Z-Wave error The message cannot be sent because node 14 is dead
ZWaveError: The message cannot be sent because node 14 is dead
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2207:12)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:2405:17)
    at DoorLockCCAPI.set (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:294:3)
    at Proxy.DoorLockCCAPI.<computed> (/opt/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:140:4)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:752:4)
    at Function.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:33)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:91:62)
2021-07-14T10:04:47.011Z DRIVER   Dropping message because the driver is not ready to handle it yet.
```

AlCalzone, 2021-07-14:
**Status NoAck** means that the Z-Wave stick received no acknowledgement from the device that the command we sent it was received. There can be several reasons for this, but none that the driver software has any influence over. The most common are bad routes (the node receives the message, but doesn't know where to reach the stick - can be caused by moved/removed nodes) and radio interference (causing the ACK to get lost).
You can try:
* putting the controller on an USB extension cord, away from metallic surfaces (including USB ports)
* healing the network
* adding mains-powered nodes between the controller and the node

Nutti85, 2021-07-14:
> **Status NoAck** means that the Z-Wave stick received no acknowledgement from the device that the command we sent it was received. There can be several reasons for this, but none that the driver [...]

Well, this was working perfectly before moving to Zwave-JS and I have changed nothing else so that is weird.

closed 2021-07-18 by zwave-js-assistant[bot]