zwave-js/zwave-js #3016: ZWA005 config not working with Firmware 2.16

halytech, 2021-07-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Trying to use vendor documented config params for TriSensor ZWA005 doesn't map right
What do you observe?
Many config items are incorrect (causing incorrect or unusable settings when adjusted)
What did you expect to happen?
I expected to be able to use vendor provided params on this device.
Vendor documentation on currently released params (especially the temp reporting time and temp reporting change levels):
[https://help.aeotec.com/support/solutions/articles/6000195459-trisensor-user-guide-](https://help.aeotec.com/support/solutions/articles/6000195459-trisensor-user-guide-)

### Device information

Manufacturer: Aeotec
Model name: Trisensor ZWA005
Node ID in your network: 27

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 7.12.0
`node-zwave-js` branch: release? (HA latest)
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Can't locate log in HA release, can provide the network JS dump.json tho if requested

AlCalzone, 2021-07-14:
> Many config items are incorrect

Can you be a bit more specific? Which params are wrong, what did you expect them to be like, what are they like?

halytech, 2021-07-14:
Example:

Configuration parameter for "Temperature automatic report based on time" is supposed to be parameter number 23 per the vendor associated documentation.  In the config file that's in the repo (node-zwave-js/packages/config/config/devices/0x0371/zwa005.json), there is no parameter 23 and instead it's listed as parameter 15.  This appears to be incorrect for several other parameters as well (basically any of the config parameters above 13 seem to be wrong).

AlCalzone, 2021-07-14:
@blhoward2 got any insight? You touched that file a while ago.

blhoward2, 2021-07-15:
This is odd, and something is off in their documentation. The user guide does say, as an example, that parameter 14 is the battery report LED. But the engineering sheet says that is temperature unit. Both the 2.15 version (https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/6102433840/original/ES%20-%20TriSensor.pdf?response-content-type=application%2Fpdf&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS6FNSMY2XLZULJPI%2F20210715%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210715T025451Z&X-Amz-Expires=300&X-Amz-SignedHeaders=host&X-Amz-Signature=5cbc19a55c8365b4474ca38e48ee628c89172a95146eaf0b06226861d358fd9c) and the 2.25 version https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/6102433840/original/ES%20-%20TriSensor.pdf?response-content-type=application%2Fpdf&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS6FNSMY2XLZULJPI%2F20210715%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210715T025451Z&X-Amz-Expires=300&X-Amz-SignedHeaders=host&X-Amz-Signature=5cbc19a55c8365b4474ca38e48ee628c89172a95146eaf0b06226861d358fd9c

blhoward2, 2021-07-15:
@halytech I'm not sure which is off, their user guide or their engineering sheet. Have you tried setting them as they are currently in the config file to see if it works even if it doesn't match the user guide?

halytech, 2021-07-15:
@blhoward2 My barometer that lead me here is attempting to use the temperature change trigger on the device.  It currently does not update more than once an hour (which mirrors the expectation of the default value for param 23 of 3600 seconds).   I can't set param 23 to see if indeed it works because I can't get at that param through my implementation in Home Assistant when I try to use the direct service call to set a config param.  There's an advanced option to send a command direct but I don't know the values to pass in to that service through HA to set the value and observe if the reporting frequency increases.

BTW my device firmware is 2.16 if that helps.

blhoward2, 2021-07-15:
Ok. We work with Aeotec and I emailed them last night. We'll get clarity from their engineers.

halytech, 2021-07-15:
@blhoward2 Thank you! I'll give it a whirl and offer feedback on the behavioral changes of the device when I set that value for further confirmation

blhoward2, 2021-07-17:
Apparently, I'm blind. There are two numbers for each parameter in the engineering sheet, and for the first 13 it is 1->1. But at 14 it starts jumping numbers so you have to look at the chart, not the number next to the parameter name.

PR #3033 (2021-07-17): feat(config): correct Aeotec ZWA005 parameter numbers
Closes #3016
closed 2021-07-17 by AlCalzone