zwave-js/zwave-js #3019: Options for integrating zwave-js logging with the parent application's logs

RoboPhred, 2021-07-14:
I am building out a project that makes use of zwave-js as one of many home automation focused plugins.  However, when running, zwave-js uses its internal logging to dump human readable logs straight to the terminal.

I want to implement my own logging solution for my project, and would like a way to capture zwave-js log messages and integrate it with my solution.  This would allow my repl to work while zwave-js is running, and permit me to capture zwave log messages and display them on the web ui.

Ideally, I would be able to receive log packets from a zwave-js event, and re-emit them into my logging framework.  Raw text would be acceptable, but if zwave-js uses key/value ancillary data along with the messages that would be even better.

AlCalzone, 2021-07-14:
You should be able to realize that by using a custom transport.
https://zwave-js.github.io/node-zwave-js/#/api/driver?id=zwaveoptions and https://zwave-js.github.io/node-zwave-js/#/api/driver?id=logconfig
You'll probably want these options:
```
logConfig: {
  enabled: false, // silences the internal loggers
  transports: [ /* your custom transport here */ ]
}
```
Your transport can be any `winston` transport. If you want to preserve the default logging format of zwave-js, take a look at this comment:
https://github.com/zwave-js/node-zwave-js/issues/2142#issuecomment-805675415

Here's an example of a `logfmt` transport:
https://github.com/zwave-js/log-transports/blob/master/packages/logfmt/src/index.ts

marcus-j-davies, 2021-07-14:
@RoboPhred,

To give you some context here, I do this with the NR plugin - which works great to 'inject' custom log entries.

Creating the logger instance (for your Applciation), and instatiate  a custom log transport for it.
https://github.com/zwave-js/node-red-contrib-zwave-js/blob/6d7a350001a990a76567069178a5abcb81e5b7f0/zwave-js/zwave-js.js#L664

Setup logConfig options for Z-Wave JS, specifically  `DriverOptions.logConfig.transports` which will contain your custom transport.
https://github.com/zwave-js/node-red-contrib-zwave-js/blob/6d7a350001a990a76567069178a5abcb81e5b7f0/zwave-js/zwave-js.js#L699

'Injecting' a custom log entry
https://github.com/zwave-js/node-red-contrib-zwave-js/blob/6d7a350001a990a76567069178a5abcb81e5b7f0/zwave-js/zwave-js.js#L30

Hope it helps.