zwave-js/zwave-js #3036: Aeotec ZW096 parameters 83 and 84 behaving oddly

sjthespian, 2021-07-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I am attempting to use bulk_set_partial_config_parameters to set parameters 83 and 84 on an Aeotec ZW096 via. Node Red. 

What do you observe?
The values I am setting are not the same values that end up in those parameters, whether I use the mask values or the text labels.

What did you expect to happen?
The values set in the service call to be the same values that appear in the parameters on the device.

Steps to reproduce the behavior:
  1. Make a bulk_set_partial_config_parameters service call to a ZW096 such as the one below.
  2. Reload the configuration parameters from the device and see that they do not match what you just set
ex.
```yaml
service: zwave_js.bulk_set_partial_config_parameters
target:
  device_id: 5aa1f54fcc01a6cbf0868d1ae4932555
data:
  parameter: '83'
  value:
    'Night Light: Red Color': 200
    'Night Light: Green Color': 100
    'Night Light: Blue Color': 50
```
Values that were shown in the device configuration via. the web UI after the above service call:
Blue: 100
Green: 200
Red: 0

Similar behavior can be seen with parameter 84, in my testing I never see the "Night Light: Red Brightness" value set to anything. It appears that the either the mask is not being applied properly or it has an issue with red values with the high-bit set making them negative integers.

These configuration parameters work as expected when set via. the HA web UI configuration interface. This also worked properly in the old deprecated ZWave system, it only started failing after updating my Node Red flows after converting to ZWave-JS.

Attached is a debug log from this morning as I was testing out various values. The device  I am working with is Node 008.

### Device information

Manufacturer: Aeotec
Model name: ZW096
Firmware: 1.3
Node ID in your network: 8

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 7.12.0
`zwavejs2mqtt` branch: N/A

Home Assistant 2021.7.3
Driver Version: 7.12.0
Server Version: 1.8.0
Home ID: 4216457015
Nodes ready: 12 / 13

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/6835093/zwave_js.log)

AlCalzone, 2021-07-17:
Our config has a value size of 3:
https://github.com/zwave-js/node-zwave-js/blob/178f39619183673d55f947f80b6b74f6b261e4c2/packages/config/config/devices/0x0086/templates/aeotec_template.json#L1691

The manual says 4:
[image]

where "Value 1" seems to be the MSB, so our bit masks are shifted by 1.

/cc @blhoward2

blhoward2, 2021-07-26:
The engineering sheet says size 3. https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/6102433596/original/ES%20-%20Smart%20Switch%206.pdf?response-content-type=application%2Fpdf&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS6FNSMY2XLZULJPI%2F20210726%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210726T045425Z&X-Amz-Expires=300&X-Amz-SignedHeaders=host&X-Amz-Signature=8f183e1652a18a4d74d1e3687fb87366c5c50a8c61c10074019066053222ecb4

AlCalzone, 2021-07-26:
@blhoward2 in the product manual that can be found here: https://products.z-wavealliance.org/products/3119

blhoward2, 2021-07-28:
I sent an email to Aeotec checking on this.

blhoward2, 2021-07-29:
These should be four bytes. They are actually only using threes bytes but size 3 isn't valid in the spec. Format of 0xRRGGBB00.

AlCalzone, 2021-07-29:
So, are they using a 4-byte parameter but only occupy the high 3 bytes?

blhoward2, 2021-07-29:
Looks like it. He tested it and provided that format for 83. 84 is 0xGGYYRR00. 

He said the device should accept 0xRRGGBB and 0xGGYYRR but that's actually not according to the spec.

AlCalzone, 2021-07-29:
> He said the device should accept 0xRRGGBB and 0xGGYYRR but that's actually not according to the spec.

Probably works if we write a 3-byte value, but the response message says its 4 bytes and that means the partial param bitmasks select different bytes.

So, we should change that to 4 bytes and add `00` at the end of those bitmasks.

PR #3125 (2021-08-01): fix(config): correct values sizes for zw096/zw099
Closes #3036
closed 2021-08-01 by AlCalzone