zwave-js/zwave-js #3047: Merge config files for `M417_9E` and `HUSBZB-1`

AlCalzone, 2021-07-19:
https://github.com/zwave-js/node-zwave-js/discussions/3045

Looks like the update changed the product fingerprint from [0x0000:0x0001:0x0001](https://devices.zwave-js.io/?jumpTo=0x0000:0x0001:0x0001:0.0) to [0x0000:0x0003:0x0008](https://devices.zwave-js.io/?jumpTo=0x0000:0x0003:0x0008:0.0).

We should probably merge these two files into one, although I'm sure that someone will complain then that his `M417_9E` is identified as an `HUSBZB-1` 🤷🏻‍♂️

kpine, 2021-07-19:
The Silicon Labs UZB3 is also recognized as M417_9E. Perhaps a generic description, not HUSBZB-1 should be used.

AlCalzone, 2021-07-19:
Makes sense, we already do that for the Z-Wave 700 controller:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0000/700_series_controller.json

ProjectPatatoe, 2021-07-19:
Is whether or not zwave+ being supported or not part of the config or in the firmware? If all it is is just a fingerprinting issue, I wouldn't mind hex editing the fingerprint to what it is supposed to be to be correct.

ProjectPatatoe, 2021-07-19:
Is "0000 0003 0008" adjacent in the firmware? i found in the firmware that pattern and am considering correcting it there. However I seem to remember something about checksum in the hex files or something. How is that calculated if that is the case?

AlCalzone, 2021-07-19:
> Is whether or not zwave+ being supported or not part of the config or in the firmware?

That doesn't matter for controller sticks. All it provides to the driver is the radio functionality and some magic. We don't (and we can't) care if a controller is Z-Wave+. All you're getting from the config files for them is a nice label.

ProjectPatatoe, 2021-07-20:
So I was poking around in the SDK provided by Silicon Labs and found their product type id enum and their product id enums. It looks like perhaps it is actually correct now and the husbzb-1 being a 0x0001,0x0001 was maybe not intended? I'm wondering if maybe Nortek Control didn't properly set the id values when they originally released the stock firmware.

I did a quick search for what a "M417_9E" is supposed to be and... well I think they may have made a mistake on their firmware or identifiers. It seems like it is supposed to be [this ](http://www.heimantech.com/product/?type=detail&id=39) and you can see the m417_9e when you get their installation guide pdf. It looks like they just took the sdk and compiled it without changing anything or something

from the silabs SDK
```c
/**
 * Standard enum Product type Id
 */
typedef enum _PRODUCT_TYPE_ID_ENUM_
{
  PRODUCT_TYPE_ID_ZWAVE_ZIP_GATEWAY = 1,
  PRODUCT_TYPE_ID_ZWAVE,
  PRODUCT_TYPE_ID_ZWAVE_PLUS
} eProductTypeID;
/**
 * Standard enum Product Plus Id's
 */
typedef enum _PRODUCT_PLUS_ID_ENUM_
{
  PRODUCT_ID_DoorLockKeyPad = 0x0001,
  PRODUCT_ID_SwitchOnOff = 0x0002,
  PRODUCT_ID_SensorPIR = 0x0003,
  PRODUCT_ID_InclusionController = 0x0004,
  PRODUCT_ID_MyProductPlus = 0x0005,
  PRODUCT_ID_SecureSensorPIR =  0x0006,
  PRODUCT_ID_SecureSwitchOnOff = 0x0007,
  PRODUCT_ID_SerialAPIPlus = 0x0008,
  PRODUCT_ID_ProdTestDUT = 0x0009,
  PRODUCT_ID_ProdTestGen = 0x000A,
  PRODUCT_ID_PowerStrip = 0x000B,
  PRODUCT_ID_WallController = 0x000C,
  PRODUCT_ID_ZIRC = 0x21ac
} eProductPlusID;
```

I'm trying to find out how to compile the firmware and not just reuse their precompiled version but I can't seem to figure out how to compile for the sd3503 chip ):

AlCalzone, 2021-07-20:
> I'm wondering if maybe Nortek Control didn't properly set the id values when they originally released the stock firmware.

Very likely - that seems to have happened a couple of times.

> Seems like it might be safe to label this type/id as a generic zwave+ controller.

I agree. Worst case a label for some old device is wrong.

closed 2021-08-02 by AlCalzone