zwave-js/zwave-js #3051: Missing device configuration: 0x019b:0x0004:0x000f:4.1

AlCalzone, 2021-07-20:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Unknown

### Manufacturer ID

0x019b

### Product Type

0x0004

### Product ID

0x000f

### Firmware Version

4.1

### Do you have a manual?

Reported by Telemetry:

capability | value
-- | --
associationGroups | ["Lifeline"]
supportsAGI | true
supportsConfigCCV3 | false
supportsZWavePlus | true
zWavePlusVersion | 1

AlCalzone, 2021-07-20:
If anyone plans on helping out with this file, please read https://zwave-js.github.io/node-zwave-js/#/config-files/using-telemetry-data first.

ralphvandenhoudt, 2021-12-29:
I've bought tis Heatit Z-Push Wall Controller. I want to help with the files for this controller but don't know where to start?

Is adding a Device Configuration File sufficient ?

The Z-Wave product info: https://products.z-wavealliance.org/products/4028

Greetings Ralph

AlCalzone, 2021-12-30:
Hope that answers some of your questions. I'll let the bot create an import from the Z-Wave Alliance DB which can serve as a starting point.

@zwave-js-bot import config 4028 from zwa

AlCalzone, 2023-02-01:
fixed by https://github.com/zwave-js/node-zwave-js/pull/3987

closed 2023-02-01 by AlCalzone