zwave-js/zwave-js #3052: Missing device configuration: 0x0384:0x0200:0x000a:2.7

AlCalzone, 2021-07-20:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Unknown

### Manufacturer ID

0x0384

### Product Type

0x0200

### Product ID

0x000a

### Firmware Version

2.7

### Do you have a manual?

Reported by Telemetry:

capability | value
-- | --
associationGroups | ["Lifeline","Basic set"]
supportsAGI | true
supportsConfigCCV3 | false
supportsZWavePlus | true
zWavePlusVersion | 1

AlCalzone, 2021-07-20:
If anyone plans on helping out with this file, please read https://zwave-js.github.io/node-zwave-js/#/config-files/using-telemetry-data first.

blhoward2, 2021-07-31:
@zwave-js-bot import config 2906 from zwa

PR #3127 (2021-09-18): feat(config): add Aibase Water Leak Sensor
fixes: #3052

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-09-18 by AlCalzone