zwave-js/zwave-js #3054: Missing device configuration: 0x0258:0x0003:0x608d:3.80

AlCalzone, 2021-07-20:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Unknown

### Manufacturer ID

0x0258

### Product Type

0x0003

### Product ID

0x608d

### Firmware Version

3.80

### Do you have a manual?

Reported by Telemetry:

capability | value
-- | --
associationGroups | ["Lifeline","Sensor Basic Set","Sensor notifi rep","Sensor notifi rep"]
supportsAGI | true
supportsConfigCCV3 | false
supportsZWavePlus | true
zWavePlusVersion | 1

AlCalzone, 2021-07-20:
If anyone plans on helping out with this file, please read https://zwave-js.github.io/node-zwave-js/#/config-files/using-telemetry-data first.

onkytonk, 2021-07-22:
I just received these Neo Coolcam PIRs, and came here to request for them to be added to the database.

They have all the same parameters as the below device, except for one additional.
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0258/motion_sensor_2.json

I'm very new to this, if I knew how to add it myself, I would.
Please forgive me if I'm going about this the wrong way.

Thank you.

I've attached images of the manual. 
[image]
[image]
[image]
[image]
[image]
[image]

AlCalzone, 2021-07-22:
> I just received these Neo Coolcam PIRs

To make sure, do those have the fingerprint `0x0258:0x0003:0x608d`?

onkytonk, 2021-07-22:
> To make sure, do those have the fingerprint `0x0258:0x0003:0x608d`?

Yes, thank you.
Please let me know if you require any more information. :)

This is what is showing up in zwavejstomqtt in Home Assistant:

[image]

PR #3072 (2021-07-22): feat(config): add NAS-PD03Z Motion Sensor 3, template other motion sensors
This motion sensor model number is NAS-PD01Z, but it seems there are multiple versions. Hence, the naming scheme with versions.

fixes: #3054
closed 2021-07-22 by AlCalzone