zwave-js/zwave-js #3061: Fibaro Roller Shutter 3

sefki21, 2021-07-20:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Fibaro Roller Shutter 3 gets recognized as window instead of shutter.

What do you observe?
Fibaro Roller Shutter 3 gets recognized as window instead of shutter.

### Device information

Manufacturer: Fibaro
Model name: Fibaro Roller Shutter 3 
Node ID in your network: 10

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:  0.1.29

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

.

kpine, 2021-07-20:
Do you have a link to an HA issue or something which referred you here? The classification of the device (window vs. shutter) is controlled by HA, not node-zwave-js.

sefki21, 2021-07-20:
didn't raise it to HA as during some searching I found this https://github.com/zwave-js/node-zwave-js/issues/1225 which got resolved here as well. Means I should raise this to HA ?

kpine, 2021-07-20:
You'll have to clarify and provide more details about what you mean by "recognized as window instead of shutter" then. I assumed you meant the device type. If you are talking about the name of the product, etc. then this is the place.

sefki21, 2021-07-20:
Yeah, I'm talking about the device class within HA, but I assumed zwave-js does specify this setting. 

[image]

I mean I can edit the device class in HA front end via "customize" but this does not change the device class HA exposes to  HomeKit, why it seems a fix at the root of this issue is needed.

Still node-js issue or should I raise this to HA?

kpine, 2021-07-20:
The device class has nothing to do with this project so you should submit the issue to HA.

sefki21, 2021-07-20:
ok. thanks.

closed 2021-07-20 by sefki21