zwave-js/zwave-js #3066: [bug]AEON Labs / MultiSensor 6 firmware update not working

mschiessl, 2021-07-21:
### 

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker

zwavejs2mqtt: 5.4.0
zwave-js: 8.0.3

### Describe the bug
When trying to send a firmware update towards [AEON Labs / MultiSensor 6](https://devices.zwave-js.io/?jumpTo=0x0086:0x0002:0x0064:0.0) i am getting the following error messages:

First attempt
```text
Failed to start the update: Firmware target #0 not found on this node! beginFirmwareUpdate undefined
```
on other attempts i am getting the following error
```text
Failed to start the update: A firmware upgrade is already in progress! beginFirmwareUpdate undefined
```

Trying to abort the update i am receiving this but without any change on the second err message
```text
Success zwave api call abortFirmwareUpdate undefined
```

### To Reproduce
- Go to the [Aeotec website](https://help.aeotec.com/support/solutions/articles/6000036562-how-to-update-multisensor-6-z-wave-firmware-) and fetch the latest [EU firmware](https://help.aeotec.com/helpdesk/attachments/6130549819) (for my use case)
- extract the received rar file
- Go to the webinterface, pick the device ID (tested on several Multisensors - so lets use zwave-js/zwavejs2mqtt#40 for now) 
- Click the advanced button within the device details subpage
- select Firmware update 
- set Target: "0" (to hit the chip)
- Select the extracted .exe file from the firmware (found that zwave.js is able to extract aeotec firmware from it)
- click the begin button

See error message on botton & observer log

I waited ~ 1.5 hours for the update to complete without success 

### Expected behavior
The regarding deviceID is getting the firmwareupdate through  (after some time)

AlCalzone, 2021-07-21:
Thanks, I think I know why this happens. Should be fixed in the next patch.

mschiessl, 2021-07-21:
@AlCalzone  - thanks that was a seamlessly quick one :D thanks for the heads up.
Let me know if you want me to run a preflight test with a self built image. Happy to verify upfront a release

PR #3068 (2021-07-21): fix: logic for validating firmware update target
fixes: #3066
closed 2021-07-21 by AlCalzone