zwave-js/zwave-js #3067: CT101 no longer automatically updating temperature values

Hilo4321, 2021-07-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Updating to v8

What do you observe?

CT101 thermostats no longer update temperature values without refreshing.

What did you expect to happen?

Values should update automatically when changing.

Steps to reproduce the behavior:

Observe that values change upon refreshing, but are otherwise static (from zwavejs2mqtt)

### Device information

Manufacturer: Radio Thermostat Company of America
Model name: CT101
Node ID in your network: 7, 43, 54

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.0.1
`zwavejs2mqtt` branch: 5.3.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

updated to v8

### Did this work before?

Yes (please describe)

### If yes, where did it work?

prior to v8. I keep it updated regularly, but am not sure of exact version. Would have been latest within the last 2 weeks.

### Attach Driver Logfile

[zwavejs_2021-07-21.log](https://github.com/zwave-js/node-zwave-js/files/6857430/zwavejs_2021-07-21.log)

AlCalzone, 2021-07-21:
Please update to 8.0.3, re-interview the device and see if that changes things.

Hilo4321, 2021-07-21:
Thank you, I've updating everything and will monitor and update.

Hilo4321, 2021-07-21:
OK, I realized I could just force the temp change manually. 

No change in temp is shown in zwavejs2mqtt (or HA), but I do the the changes reported in the log (node 054 @ 2021-07-21T17:17). Surprised to also see humidity reporting there as well, as it that has never worked without a refresh.
[zwavejs_2021-07-21 (1).log](https://github.com/zwave-js/node-zwave-js/files/6857695/zwavejs_2021-07-21.1.log)

AlCalzone, 2021-07-21:
Could you test the device file please?
https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

[ct101.json.txt](https://github.com/zwave-js/node-zwave-js/files/6858404/ct101.json.txt) (remember to rename back to .json)

Hilo4321, 2021-07-21:
Seems that humidity is working on its own, now, but temp doesn't even register in the log (set for the minimum .5F change)
[zwavejs_2021-07-21_1.log](https://github.com/zwave-js/node-zwave-js/files/6858714/zwavejs_2021-07-21_1.log)

AlCalzone, 2021-07-22:
```
2021-07-21T20:34:16.832Z CNTRLR   [Node 054] Embedded device config loaded
                                             ^^^^^^^^
```
Doesn't look like it loaded the new config file.

Hilo4321, 2021-07-22:
[Here's](https://github.com/zwave-js/node-zwave-js/files/6860310/CT101DebugInfo.txt) the Debug Info for Node 54.

I created [docker]/zwavejs2mqtt/store/config and placed ct101.json there. I restarted the container, re-interviewed and had the results above. Also did a full stop and up.

I'm not sure what I'm missing to get this to trigger.

AlCalzone, 2021-07-22:
Hmm, I fixed a bug related to config files in https://github.com/zwave-js/node-zwave-js/releases/tag/v8.0.4 - maybe that's what happens for you here. I'll ask @robertsLando to release a new version of z2m with that driver version.

robertsLando, 2021-07-22:
Release is out wait half an hour for docker release to be available for all archs

Hilo4321, 2021-07-22:
The embedded device config is still being used. 

Additionally, all devices are unknown per #1469 in zwavejs2mqtt.
[zwavejs_2021-07-22.log](https://github.com/zwave-js/node-zwave-js/files/6862449/zwavejs_2021-07-22.log)

AlCalzone, 2021-07-22:
Yeah, next patch is already on the way - this time with regression tests 😇

AlCalzone, 2021-07-22:
https://github.com/zwave-js/node-zwave-js/releases/tag/v8.0.5

Hilo4321, 2021-07-22:
No change, unfortunately. Relevant detail is right at the end. 

[zwavejs_2021-07-22.log](https://github.com/zwave-js/node-zwave-js/files/6863290/zwavejs_2021-07-22.log)

AlCalzone, 2021-07-22:
Ah damn, that seems to be another bug. Now we don't even know if it is using the correct file.
Can you try mounting the file I gave you (as .json) directly into
```
/usr/src/app/store/.config-db/devices/0x0098/ct101.json
```
and restart?

Hilo4321, 2021-07-22:
I don't have /usr/src/app/ on Pi/Docker.

AlCalzone, 2021-07-22:
That's the path inside the docker container.

Hilo4321, 2021-07-22:
Alright, took a minute to figure out how to paste into the container. 

Looks good! Temp updated successfully!

I'm perhaps a little too excited about it, but looks like humidity is working now as well.

Should I avoid restarting/stopping the container until the fix is pushed?

Sidenote: looks like the existence of .../store/config is what triggers the embedded/user configs?

AlCalzone, 2021-07-22:
You can leave it as is for now. I'll incorporate the changes into the next version (probably 8.0.6 / z2m 5.4.3 [?]) - just don't forget to remove the volume mount then.

The embedded vs. user config check is fixed in #3078.

Hilo4321, 2021-07-22:
Thanks very much!

PR #3079 (2021-07-22): fix(config): add root report mapping to CT101
fixes: #3067
closed 2021-07-22 by AlCalzone