zwave-js/zwave-js #3084: Issue with Aeon Labs - Home Energy meter after update to 5.4.2 (Latest version as of 7/23 - Docker) from 5.2.2

kan84, 2021-07-23:
Hi,

I have been using HEM v1 for sometime with zwavejs2mqtt + HA without any issue. I have set group 1 reporting for both clamps and set it as 30 seconds. This setup has been working fine on 5.2.2, as soon as I updated to 5.4.2 HEM stopped updating the values automatically. If I refresh from zwavejs2mqtt dashboard by click refresh the values are fetched otherwise it just does not refresh.

I made sure old configuration values for Group 1 are still there like 30 sec refresh which I tried changing few times to kickstart HEM. Restarted all of the devices but still did not work. Eventually I came back to 5.2.2. Is there any change in HEM v1 driver or anything else which might be causing this issue?

Thanks!

kan84, 2021-07-23:
I have attached two days log. I updated it yesterday and it stopped reporting Node 24 (HEM) or rather updates were not flowing in zwavejs unless you do it manually. I rolled it back and it started working, tried couple of times updating to latest and then rolling back. 
5.2.2 Node 24 reports every 30 secs
5.4.2 Node 24 does not report though config set for group 1 is 30 secs, updated couple of times did not work. When i manually refresh i can see in the logs correct value is being fetched.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6870190/zwavejs2mqtt-store.zip)

AlCalzone, 2021-07-23:
Can you re-interview the device and make a log of that too?

kan84, 2021-07-23:
I believe you want it for 5.4.2 and not 5.2.2? Will update to 5.4.2 and interview. The log which I had sent should have some interview stuff as well as thats the first thing I tried.

kan84, 2021-07-23:
Attached 
[Manual_Refresh.log](https://github.com/zwave-js/node-zwave-js/files/6870507/Manual_Refresh.log)
[Interview.log](https://github.com/zwave-js/node-zwave-js/files/6870508/Interview.log)
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6870509/zwavejs2mqtt-store.zip)

github-0, 2021-07-28:
Same behavior with zwavejs 8.0.5 and HEM Gen 5 (ZW095): manual refresh works but automatic updates are not happening. I'm a new HA/zwavejs user so cannot comment which version still worked, but I know the HEM works since it is sending automatic updates with Openhab and its Z-Wave implementation.

AlCalzone, 2021-07-28:
@zwave-js-bot add compat flag to 0x0086:0x0002:0x0009:0.0
```
// This device reports compound values on the root endpoint
preserveRootApplicationCCValueIDs: true
```

AlCalzone, 2021-07-28:
The endpoint changes in v8 cause the root endpoint to be hidden. This was the case before, but we incorrectly mapped the reports to endpoint 1, so you would see the values change. The bot should create a PR in a few seconds which fixes that.

AlCalzone, 2021-07-28:
@github-0 I'll also need a driver log of a re-interview from you.

PR #3110 (2021-07-28): fix(config): preserve root endpoint values for Aeotec DSB09 
fixes: #3084

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-07-28 by AlCalzone