zwave-js/zwave-js #3089: Add undocumented/hidden parameter for No. of channels to RGBGenie/Sunricher LED controllers

AlCalzone, 2021-07-26:
### Discussed in https://github.com/zwave-js/node-zwave-js/discussions/3086

<div type='discussions-op-text'>

<sup>Originally posted by **jhein05** July 23, 2021</sup>
Hi,

I found that some of the Sunricher/RGBGenie LED controllers have an undocumented parameter (4) that allows you to change the controller type. The options are 1: monochrom (1 channel) 2: CCT (tunable white; 2 channels); 3: RGBW (4 channels).

This definitely works for mine RGBGenie ZW-1001 identified as Sunricher ZV9101
It is highly likely that is also works for RGBGenie ZW-1005 

I am not a developer, so while I can probably hack it into my configuration files, it would surely be not good enough for your standards. I guess I just leave this here :)

</div>

jhein05, 2021-07-26:
OK, guide me through it :)

The actual change is easy, because I can just copy the parameter 4 part from a similar model, it is more the Github workflow I need help with. I am assuming I have to fork the repository, make the change locally, sync back, and then create a PR. Am I assuming correctly?

AlCalzone, 2021-07-26:
You'll have to start with a fork and ideally a branch in that fork.
You can then either edit the files in the branch and create a PR for that branch or work locally.

If you work locally, you might want to read this first: https://zwave-js.github.io/node-zwave-js/#/development/intro
You can then perform the checks locally by executing `yarn lint:config`, which will show you if you made a mistake.