zwave-js/zwave-js #3090: [question] Missing endpoints with version 5.x

febalci, 2021-07-26:
I have qubino ZMNHCD shutter switches. These normally has only Endpoint 0. Whenever i plug in its optional temperature sensor, then it has Endpoint 0.1 and 2. This was working fine with zwavejs2mqtt 4.x. But in 5.x, as i see from "Migrating to v8" notes of node-zwave-js; it has "Ignore unnecessary endpoints" feature now. [(Migrating to v8)](https://zwave-js.github.io/node-zwave-js/#/getting-started/migrating-to-v8?id=ignore-unnecessary-endpoints-instead-of-the-root-device) I need to add:

```
"preserveEndpoints": [0, 1, 2]
```

but i do not know where to put it. I am using a docker container for zwavejs2mqtt. 3 of 12 Qubino ZMNHCD i have temp sensors connected; so 3 of them are multi endpoints and the remaining 9 are single endpoint.

Can anyone tell me how i can use this preserveEndpoints flag for the multiple endpoint issue in these 3 nodes? Thanks a lot...

AlCalzone, 2021-07-26:
Please share a driver log (level debug) of a re-interview.
Also, do you know which endpoint the sensor will appear on? Did you re-include your devices after adding the temperature sensor?

febalci, 2021-07-26:
In zwavejs2mqtt 4.x, i had 2 endpoints; Endpoint 0 lifeline and Endpoint 2 Temperature Sensor. Now, in 5.4.2, as default it comes as Endpoint 0 lifeline and Endpoint 0 Temperature Sensor. Funny thing is, when i delete Endpoint 0 Temperature Sensor From Groups, everything works perfectly fine even the temperature sensor sensor readings. So, you can simply disregard this problem; it works fine... Thanks...

closed 2021-07-26 by febalci

AlCalzone, 2021-07-26:
> Now, in 5.4.2, as default it comes as Endpoint 0 lifeline and Endpoint 0 Temperature Sensor.

Yep, that's the intention. If there is no duplicate functionality between endpoints, they are not really necessary.

> Funny thing is, when i delete Endpoint 0 Temperature Sensor From Groups, everything works perfectly fine even the temperature sensor sensor readings.

What do you mean by this? I'd like to make sure that we don't have to edit a configuration to make this work.

febalci, 2021-07-26:
I think there is no configuration change necessary. The way you did was right but as i understand the way Qubino does it is weird. In any case leaving it as it is better IMHO, cause it may create disturbance in other people's qubino usage. Thank you for the great work; my zwave network was never this stable...