zwave-js/zwave-js #3093: Logic-Group ZIF 5030

cbho, 2021-07-26:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Logic-Group ZIF 5030

### Manufacturer ID

0x0234

### Product Type

0x0003

### Product ID

0x0010A

### Firmware Version

1.1

### Do you have a manual?

[zif5030_datablad-3.pdf](https://github.com/zwave-js/node-zwave-js/files/6878771/zif5030_datablad-3.pdf)
[zif5030-users-manual-en.pdf](https://github.com/zwave-js/node-zwave-js/files/6878772/zif5030-users-manual-en.pdf)

I have made the .json-file for this, but can't get it to work properly.
Probably due to the structure in the configuration database for Logic Group products, that include a lot of templating.

Would be happy to provide the files for review.

AlCalzone, 2021-07-26:
> Would be happy to provide the files for review.

Yeah, go ahead. We can help you through it.

> I have made the .json-file for this, but can't get it to work properly.
Probably due to the structure in the configuration database for Logic Group products, that include a lot of templating.

If you want to test them in the user-config dir, you need to copy the template file and potentially the master template aswell. Make sure to use the same structure:
```
templates/master_template.json
0x0234/templates/logic_group_template.json
0x0234/your-file.json
```

cbho, 2021-07-28:
Adding the folder structure (and identifying an error in the productId :-) ) did the trick (for recognizing the device).

### Status
- The ten relay switches are now available - and controllable - in the control panel.

### Challenges
**Endpoints**
The specification say:
> • Root device; an intersection of all the devices (endpoint 0).
> • 10 switch devices to control each of the relay outputs (endpoint 1 – 10).
> • 4 switch devices representing the digital inputs (endpoint 11 – 14).
> • 4 sensor devices representing the analogue inputs (endpoint 15 – 18).
Endpoints 11-14 are not "real" switches - they can not be switched by the user, they are in fact "[...] typically used for reporting the actual status of the input to the Controller for the Controller to visualize the input in its user interface."
Thus, while it is technically represented in the correct way in the controller, I think it is confusing (and unnecessary) that those endpoints are represented as switches - as they are not switchable by the user...

**Configuration**
All configuration parameters are present.
However, for the four sensor devices (endpoint 15-18, configuration parameters 10-48), it seems like all values (when refreshing the parameters in the interface) ends up at sensor 1.

**Naming**
Switches are named "[61-37-1-targetValue] Target value" etc.
In the file, the association groups are labelled "_"label": "Relay output 1 (Basic Report On / Off)"_, e.g.
Is it possible to label the switches also? Like "Relay 1" etc.?
(That could also be a cool addition to other Logic Group devices, that has a relay and four physical switches, if they could be named "Relay", "Switch 1", etc. instead of the more generic naming, that may be rather confusing sometimes)

Tried to add the config-files here, but for some reason I can't add json-files - but they can be found here: https://1drv.ms/u/s!AiJbSZ6oHDf7jKthYPrsGmw60OjdLw?e=eMFntb

AlCalzone, 2021-07-28:
> Is it possible to label the switches also? Like "Relay 1" etc.?

No. Maybe when I've solved #552 we'll be a bit more flexible here, but I got bigger fish to fry atm.

> However, for the four sensor devices (endpoint 15-18, configuration parameters 10-48), it seems like all values (when refreshing the parameters in the interface) ends up at sensor 1.

Which driver version are you on?

cbho, 2021-07-29:
Would be cool with the naming - but need to fix is more important than nice to improve ;-)

Driver etc:
zwavejs2mqtt: 5.4.2
zwave-js: 8.0.5

AlCalzone, 2021-07-29:
> zwave-js: 8.0.5

Ok I thought you might be on an old one. Can you share a driver log (loglevel debug) which shows the effect that all values end up on sensor 1?

cbho, 2021-07-29:
Sure.
Just thougt about, if I should try excluding and reincluding the device first? Not sure if that would do any real difference compared to re-interviewing the device?
(I did of course restart the container with zwavejs2mqtt a couple of times already)

AlCalzone, 2021-07-29:
> if I should try excluding and reincluding the device first?

No, unless you changed a parameter which changes the capabilities of the device, this is not necessary.

cbho, 2021-07-29:
It's node 61

[zwavejs_2021-07-29.log](https://github.com/zwave-js/node-zwave-js/files/6899265/zwavejs_2021-07-29.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6899266/zwavejs2mqtt.log)

AlCalzone, 2021-07-29:
Something is not right there. First, I don't see any log that the re-interview was started - just that all config paramters get queried.
Second, the device doesn't respond to the config paramters after no. 14. Are you sure you have the correct device/firmware?

cbho, 2021-07-29:
Sorry, didn't realise that you needed a re-interview...

New log files:
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6900313/zwavejs2mqtt.log)
[zwavejs_2021-07-29.log](https://github.com/zwave-js/node-zwave-js/files/6900267/zwavejs_2021-07-29.log)

The device is without doubt a ZIF 5030 and it has the newest available firmware.

AlCalzone, 2021-07-29:
> Sorry, didn't realise that you needed a re-interview...

Sorry, I got confused for a second here. I probably don't...
However none of your logs include the situation you described above.
> for the four sensor devices (endpoint 15-18, configuration parameters 10-48), it seems like all values (when refreshing the parameters in the interface) ends up at sensor 1.

The first is just the config parameter query and the second one is an interview that's not done yet. What I can see is that there are no endpoints 15-18 though:
```
2021-07-29T11:42:29.408Z CNTRLR « [Node 061] received response for device endpoints:
                                  endpoint count (individual): 14
                                  count is dynamic:            false
                                  identical capabilities:      false
                                  endpoint count (aggregated): 0
...
2021-07-29T11:42:30.155Z CNTRLR « [Node 061] received endpoints: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14
```
and that it only responds to the config parameters 1-7 and 10-14 - all the others result in a report for parameter 1. Your config file definitely has a lot more defined.

> The device is without doubt a ZIF 5030 and it has the newest available firmware.

Could you verify with the manufacturer please? For reference, this is the device fingerprint:
```
manufacturer id: 0x0234
product type:    0x0003
product id:      0x010e
firmware version 1.1
```

cbho, 2021-08-07:
Short update:
Communicating with the lead developer from Logic-Group - he'll look into to it when he's back from vacation.