zwave-js/zwave-js #3094: OTA Firmware Update Times Out for Zooz ZEN Series 700 Switches, But Then Believes A Firmware Update is Active After Declared Dead

ziptbm, 2021-07-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

OTA Firmware Updates for Zooz ZEN77 Series 700 Dimmer Switches

What do you observe?

Firmware attempts to take, but times out usually less than 1/4 of the way through the file transfers of firmware fragments.  I then try again (persistence is what worked on ZEN72s and ZEN71s, but it will not take on the ZEN77s), so it fails and I try again.  

"Node 45 firmware update FINISHED: Status -1, Time: undefined"

 After a few more attempts, the updates fail to take as ZwaveJS2MQTT believes that a firmware update is already happening.

What did you expect to happen?

Firmware to take.

Steps to reproduce the behavior:
  1. Go to 'Z-WAveJS2MQTT Control Panel within Home Assistant'
  2. Find ZEN77 Z-Wave Node
  3. Click on down arrow
  4. Click on Advanced
  5. Under Firmware update, click Begin
  6. Leave Target as "0" and click OK
  7. Select Firmware and open
 
 

### Device information

Manufacturer:  Zooz
Model name:  ZEN77
Node ID in your network:  45

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

ZwaveJS2MQTT via Home Assistant

version: 0.23.2
`node-zwave-js` branch: 8.0.5
`zwavejs2mqtt` branch: 5.4.2

### Did you change anything?
[... 508 lines cut ...]

The above was taken from the Debug menu within the ZwaveJS2MQTT web UI

and here is what's taken from the Z-Wave JS Logs option within Home Assistant.  It took some time to navigate over, so here's what's was gathered.  Please advise on what else specifically I should grab.

2021-07-26T15:00:52.441Z CNTRLR   [Node 045] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2021-07-26T15:01:01.746Z CNTRLR   [Node 045] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
2021-07-26T15:01:10.172Z CNTRLR   [Node 045] Node 45 did not respond after 3 attempts, it is presumed dead
2021-07-26T15:01:10.176Z CNTRLR   [Node 045] The node is now dead.
2021-07-26T15:01:10.187Z CNTRLR   [Node 045] Error while sending firmware fragment: Failed to send the command a
                                  fter 3 attempts (Status NoAck)
2021-07-26T15:01:10.187Z CNTRLR   [Node 045] Error while sending firmware fragment: Node 45 did not respond afte
                                  r 3 attempts, it is presumed dead
2021-07-26T15:01:10.251Z DRIVER   unexpected response, discarding...
2021-07-26T15:01:15.509Z DRIVER     no handlers registered!
2021-07-26T15:01:15.834Z CNTRLR   [Node 045] Firmware update timed out
2021-07-26T15:01:23.287Z CNTRLR   [Node 045] The node is now alive.
2021-07-26T15:01:23.287Z CNTRLR « [Node 045] Received Firmware Update Get, but no firmware update is in progress
                                  . Forcing the node to abort...
2021-07-26T15:01:56.388Z CNTRLR   [Node 045] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2021-07-26T15:02:05.544Z CNTRLR   [Node 045] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
2021-07-26T15:02:14.314Z CNTRLR   [Node 045] Node 45 did not respond after 3 attempts, it is presumed dead
2021-07-26T15:02:14.316Z CNTRLR   [Node 045] The node is now dead.
2021-07-26T15:02:22.133Z CNTRLR   [Node 045] The node is now alive.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/6879242/zwave_js.log)

AlCalzone, 2021-07-29:
> NoAck

You have a flaky connection. Multiple messages to the device got not response or their acknowledgements got lost and it looks like it took the device half a minute to reach the controller again afterwards.

Firmware updates transfer a LOT of data (in this case ~4000 messages with ~50 bytes each), so there are a lot of chances for things to go wrong.

Try bringing the switch closer to the controller (or the other way round) and try again.

johanschelin, 2021-08-12:
I also have this issue on a couple of Heltun devices where the connection is okey. Probably because the firmware is very large for their 5-relay devices. Can we give the firmware update process more time before timeout? What is the time-out / retry setting in zwjs now?

AlCalzone, 2021-08-12:
@johanschelin that depends - when/how does it time out? In general, the devices request each firmware fragment and `zwave-js` responds to that. At least in the log above, that transfer seems to fail intermittedly with `NoACK`, meaning that packets get lost and `zwave-js` retries sending the packet after a short delay. If that happens 3 times in a row, the device is marked `dead`.

We could probably just not do that for the firmware transfers. No marking as dead, no retrying, just rely on the device to request what it needs and abort if the timeout elapses. That timeout is currently 30s between each fragment, so the actual firmware size has no influence on that.

Please share a driver log where the firmware update fails during the process.

johanschelin, 2021-08-13:
Could be my large network that’s noisy and 30s timeout happens. Will do a debug logged try on Sunday evening and come back with the log 👍

Hämta Outlook för iOS<https://aka.ms/o0ukef>

blhoward2, 2021-08-19:
Please provide an update when you can. If need be I'll have Zooz send me one of these for testing.

johanschelin, 2021-08-19:
Hi!
I still have issues with OTA updates which i beleive comes from timeouts. See attached log where one update on node 54 failed after running for a while. The update started at 20:46 and ended at roughly 20:54
[zwavejs2mqtt-store (2).zip](https://github.com/zwave-js/node-zwave-js/files/7017679/zwavejs2mqtt-store.2.zip)
 with "Node 54 firmware update FINISHED: Status -1, Time: undefined" and the update did not go through

AlCalzone, 2021-08-19:
> Failed to send the command after 3 attempts (Status Fail)

I'm not entirely sure what causes this, but your stick sometimes isn't able to actually send any commands and seems to need a thinking break from time to time.

That problem is compounded by the fact that these breaks cause the node to request the same fragment multiple times. zwave-js queues the response without checking for duplicates, causing a pretty long queue with unnecessary commands.

johanschelin, 2021-08-22:
Yeah, I have also noticed strange "wait times" in the log shere it seems like nothing is happening.
Could this be related to my stick being a Aeotec 7-stick (a 700 chip)? I will contact Aeotec and see if there is a new firmware for example.

/J

blhoward2, 2021-08-22:
My Aeotec did ship with a very early version of the firmware. They don't distribute it. You need to download Simplicity Studio from SiLabs. It's included in that and you have to do the OTW upgrade from there.

johanschelin, 2021-08-22:
Ok, I have Simplicity Studio already, but couldnt find a way to upgrade the stick firmware when I checked (briefly) last week. Can you give me directions in the software how to find the OTW for the stick? 🙂

blhoward2, 2021-08-22:
It's like this though there may be a newer version. `SimplicityStudio\v5\developer\sdks\gecko_sdk_suite\v3.1\protocol\z-wave\Apps\bin\gbl\ZW_SerialAPI_Controller_7.15.4_256_EFR32ZG14_REGION_US.gbl`

I don't recall the exact button label but you have to add the zwave support which installs the file above and the control panel. Then you go into the zwave module and there is a button to upgrade.

blhoward2, 2021-08-22:
From the internet: https://www.silabs.com/documents/public/user-guides/INS13114-Z-Wave-PC-Based-Controller-v5-User-Guide.pdf

Page 4. Looks like the top right though it may be different in the newer version.

ziptbm, 2021-08-22:
Quick note regarding my original submission here, my controller is a Nortek husbzb‑1, so that’s a series 500 chipset I believe.  I’ll edit the original post to indicate this.

johanschelin, 2021-08-22:
Thanks - found that I have 7.11 and that 7.15 is available. Will test upgrading.

Hämta Outlook för iOS<https://aka.ms/o0ukef>