zwave-js/zwave-js #3095: Aeotec Temperature & Humidity Sensor (model ZWA009-B)

gitwazza, 2021-07-27:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Aeotec Temperature & Humidity Sensor

### Manufacturer ID

0x371

### Product Type

0x0202

### Product ID

0x0009

### Firmware Version

1.1

### Do you have a manual?

https://help.aeotec.com/support/solutions/articles/6000227918-a%C3%ABrq-temperature-and-humidity-sensor-user-guide-
[zwavejs_2021-07-27.log](https://github.com/zwave-js/node-zwave-js/files/6882816/zwavejs_2021-07-27.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6882817/zwavejs2mqtt.log)

AlCalzone, 2021-07-27:
@zwave-js-bot add fingerprint 0x0202:0x0009 to 0x0371:0x0002:0x0009

closed 2021-07-28 by AlCalzone

gitwazza, 2021-08-21:
Thanks AlCalzone for doing this. I can see my device now gets recognised as an "aërQ Temperature and Humidity Sensor".

PR #3098 (2021-07-28): fix(config): add fingerprint `0x0202:0x0009` to Aeotec ZWA009
fixes: #3095