zwave-js/zwave-js #3099: Power Entity has not been created during inclusion / Power Entity not available

je-mei, 2021-07-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
./.

What do you observe?
./.

What did you expect to happen?
Power-Entity available for FGR223 Fibaro Roller Shutter

Steps to reproduce the behavior:
  1. See result from home assistant / core issue: https://github.com/home-assistant/core/issues/53507

### Device information

Add-on with the issue: Z-Wave JS
Add-on release with the issue: 0.1.32
Last working add-on release (if known): included in home assistant the first first
Operating environment (OS/Supervised): Version core-2021.7.4, supervisor-2021.06.8, Home Assistant OS 6.1

-Roller shutter 2 working correctly, power entity existing
-another user uses the legacy zwave implementation, where the power entity also occurs

see https://github.com/home-assistant/core/issues/53507

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Add-on with the issue: Z-Wave JS
Add-on release with the issue: 0.1.32
Last working add-on release (if known): included in home assistant the first first
Operating environment (OS/Supervised): Version core-2021.7.4, supervisor-2021.06.8, Home Assistant OS 6.1

### Did you change anything?

no

### If yes, what did you change?

- migrating from fhem to home assistant
- including process in fhem (here everythink ran fine)
- behavior from the beginning

### Did this work before?

Yes (please describe)

### If yes, where did it work?

- within fhem, zwave roller shutter completly useable
- legacy version of zwave runs with power entity
[zwave_2021-07-26_interview_fgr223.log](https://github.com/zwave-js/node-zwave-js/files/6887404/zwave_2021-07-26_interview_fgr223.log)
[zwave_config_legacy_integration_by_another_user_where_everything_works_fine_FGRM223_Roller_Shutter_Controller_3.txt](https://github.com/zwave-js/node-zwave-js/files/6887410/zwave_config_legacy_integration_by_another_user_where_everything_works_fine_FGRM223_Roller_Shutter_Controller_3.txt)

- Detailed Information, see: https://github.com/home-assistant/core/issues/53507, the developer from core asked me to get in contact with you, referring to the existing issue opened yesterday.

Kind regards
Jens

### Attach Driver Logfile

[zwave_config_legacy_integration_by_another_user_where_everything_works_fine_FGRM223_Roller_Shutter_Controller_3.txt](https://github.com/zwave-js/node-zwave-js/files/6887414/zwave_config_legacy_integration_by_another_user_where_everything_works_fine_FGRM223_Roller_Shutter_Controller_3.txt)
[zwave_2021-07-26_interview_fgr223.log](https://github.com/zwave-js/node-zwave-js/files/6887415/zwave_2021-07-26_interview_fgr223.log)

kpine, 2021-07-27:
See my [latest comment](https://github.com/home-assistant/core/issues/53507#issuecomment-887680041) in the HA issue. Given the newly provided zwcfg*.xml file, I don't think there's any issue here.

AlCalzone, 2021-07-28:
I agree, the device is using the `Meter CC` to report its consumption and I see it being queried successfully in the log.

je-mei, 2021-07-28:
Hi AlCalzone,
hi kpine,

first of all thanks for your support! By the way I would like to mention, that some additional points are not clear for me:

1. Why is there a possibility for configurating "power reports", when they are not available in shutter 3 at all?

[image]

2.
Unfortunately, values for "Roller Shutter 3: Electric Consumed [W]" will not be send, but the kwH do. I opened and closed my shutters, but thats not represented in the graphic/log:

[image]

[image]

Kind regards
Jens

je-mei, 2021-07-28:
update:
exclusion of shutter 3
factory reset of shutter 3
inclusion of shutter 3
same phenomenon: no reporting of electric consumed W

[image]

although mulitple open and close of the shutter, no reporting:
[image]

je-mei, 2021-07-28:
[image]

kpine, 2021-07-28:
3600 seconds is one hour for the periodic report. Have you tried reducing it to something like 1 minute to see if anything is reported?

AlCalzone, 2021-07-28:
>     1. Why is there a possibility for configurating "power reports", when they are not available in shutter 3 at all?

I can't speak for how devices behave. Looking at the configuration you posted, the changes should come every 5W or 1%, depending on what is higher:
[image]

If they don't appear in the UI, looking at the driver logs could give you a hint whether the reports are received at all. If they aren't, either the messages don't reach the controller (in which case a heal could help) or they never get sent (which you can only figure out with a Zniffer next to the shutters).

je-mei, 2021-07-29:
try to change energy repots didn't have any effect to reporting
therefor in check system behvior when changing the values
error message based on "silly" log find attached

[image]

AlCalzone, 2021-07-29:
Hm, looks like that log doesn't actually use the `silly` loglevel - there should be more. @kpine do you know more about that?

The mention of "Application Status" could mean that the device didn't actually accept the command

kpine, 2021-07-29:
> Hm, looks like that log doesn't actually use the `silly` loglevel - there should be more. @kpine do you know more about that?

No, but it sounds like you got some attention in Discord. The funny thing is, we gathered the interview logs using the same procedure. That seemed to be successful. 🤷🏻‍♂️

je-mei, 2021-07-29:
its really strange :-( :-(

I was using shutter 2+3 in fhem first. next to the known bug "send no update for position" it worked really fine. I used the power/electric reports to trigger a zwave-update for the whole sensor. similar like the blueprint. and it worked. 

so i directly switched from fhem (excluding) to ha (reset, incluing) within 1 day, so hardware couldn't be the root cause. :-(

just made another update of the configuration for you: 
change report "on change" and "periodic". Here is the result:

[image]

Network heal already done a few days ago and many times ... . But finally it is sending no update:

[image]

Although open and close a minimum of twice a day.

By the way: Shutter 2 is also sending no energy, but power ;-)

[image]

shutter 2 sends only power, but no energy watt:
[image]
(screenshot during close progress)

[image]

[image]

If I could do anything to help give me some information what to do. 

Kind regards 
Jens

je-mei, 2021-07-29:
did a configuration change for shutter 2

[image]

[zwave_js-log (2).txt](https://github.com/zwave-js/node-zwave-js/files/6902169/zwave_js-log.2.txt)

je-mei, 2021-07-29:
thats all shutter 2:
[image]

je-mei, 2021-07-29:
shutter 2:
[image]

AlCalzone, 2021-07-29:
> change report "on change" and "periodic". Here is the result:

Looks like the device doesn't like you changing the parameters. I think that is because you're setting an unsupported value for the periodic report. Only 0 and 30-32400 are allowed.
[image]

I can confirm that my Roller Shutter 2 reports power via a **Multilevel Sensor (Power)** and **Electric Consumed [kWh]** with the configured frequency, but the **Electric Consumed [W]** never changes in the time it is moving. Maybe FHEM was polling? zwave-js doesn't do that automatically.

> next to the known bug "send no update for position" it worked really fine. I used the power/electric reports to trigger a zwave-update for the whole sensor

You do know that this isn't necessary in `zwave-js`, right? It uses the `Supervision CC` to get informed about the position changes of the Roller Shutter 3. Can you make a log where you move the Shutter 3?

je-mei, 2021-07-30:
1. log during change of config for shutter 3
2. 2. log during many open and closes incusive a screenshot of the graphic / cover

[during_change_parameter_shutter3-log.txt](https://github.com/zwave-js/node-zwave-js/files/6906737/during_change_parameter_shutter3-log.txt)
[image]
[image]
[after_many_open_and_closes-log.txt](https://github.com/zwave-js/node-zwave-js/files/6906738/after_many_open_and_closes-log.txt)

je-mei, 2021-07-30:
i will send you another log, where shutter is triggered by local shutter switch to open and close tomorrow if needed

je-mei, 2021-07-30:
arghh :-/ not many information stored in the log, thats different to the information displayed. 

i will do it again tomorrow and copy data from screen into text file

kpine, 2021-07-30:
@je-mei Are you using the log download button in HA, or copying and pasting from the window? The download button does not work properly, so you should copy and paste.

je-mei, 2021-07-30:
@kpine yes, forgotten :-(. used copy and paste from the window instead. 

please find attached "silly" logs and corresponding screenshots (same number link 01_* means same progress)

[00_change-device-configuration.txt](https://github.com/zwave-js/node-zwave-js/files/6908690/00_change-device-configuration.txt)
[image]
[image]
[01_postition_change.txt](https://github.com/zwave-js/node-zwave-js/files/6908691/01_postition_change.txt)
[image]
[image]
[image]
[image]
[03_one_time_down_one_time_up_by_local_swith_shutter3.txt](https://github.com/zwave-js/node-zwave-js/files/6908695/03_one_time_down_one_time_up_by_local_swith_shutter3.txt)
[image]
[image]

kpine, 2021-07-30:
In your original HA dump file, the meter value is associated with endpoint one. In your zwave-js debug logs, endpoint 0 is reporting the meter values. This would mean HA wouldn't see the updates.
```
              {
                "endpoint": 1,
                "commandClass": 50,
                "commandClassName": "Meter",
                "property": "value",
                "propertyKey": 66049,
                "propertyName": "value",
                "propertyKeyName": "Electric_W_Consumed",
                "ccVersion": 3,
                "metadata": {
                  "type": "number",
                  "readable": true,
                  "writeable": false,
                  "label": "Electric Consumed [W]",
                  "ccSpecific": {
                    "meterType": 1,
                    "rateType": 1,
                    "scale": 2
                  },
                  "unit": "W"
                },
                "value": 0.6
              },
```

```
2021-07-30T15:45:13.338Z SERIAL « 0x0112000400200a32022132058a00000000b600d9                          (20 bytes)
2021-07-30T15:45:13.341Z CNTRLR   [Node 032] [~] [Meter] value[66049]: 0.6 => 141.8                 [Endpoint 0]
2021-07-30T15:45:13.343Z SERIAL » [ACK]                                                                   (0x06)
2021-07-30T15:45:13.345Z DRIVER « [Node 032] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      141.8
                                      time delta: 0 seconds
```

Also, your original dump had the Shutter 3 as node 31, did you remove and add it? Could you provide a new dump file to confirm it's in the same situation?

je-mei, 2021-07-30:
did you remove and add it --> yes, I did.  I did everything to get the shutter run ;-). 

here my new reports
1. reinterview 
2. down and up via local switch
3. down and up via HA

[image]
[00_reinterview.txt](https://github.com/zwave-js/node-zwave-js/files/6909143/00_reinterview.txt)

####

[image]
[01_down_and_up_via_local_switch-.txt](https://github.com/zwave-js/node-zwave-js/files/6909146/01_down_and_up_via_local_switch-.txt)

####

[image]
[02_down_and_up_via_HA.txt](https://github.com/zwave-js/node-zwave-js/files/6909149/02_down_and_up_via_HA.txt)

kpine, 2021-07-30:
Could you please provide the [network dump](https://www.home-assistant.io/integrations/zwave_js/#get-a-dump-of-the-current-network-state). I just wanted to confirm that HA still sees only values for endpoints 1 and 2.

je-mei, 2021-07-30:
[zwave_js_dump-json.txt](https://github.com/zwave-js/node-zwave-js/files/6909244/zwave_js_dump-json.txt)

kpine, 2021-07-30:
@AlCalzone looks like another case of values for endpoints 1, etc, but reports on endpoint 0?

AlCalzone, 2021-07-30:
Yeah, that looks like another edge case

je-mei, 2021-07-31:
hi, fyi - behaviour of my shutter 2 in addition.
I have only two shutters. one shutter 3 and one shutter 2. shutter 2 runs better, but maybe there is the same issue occuring, but the power entity is running great ;-)
[reinterview_shutter2.txt](https://github.com/zwave-js/node-zwave-js/files/6911133/reinterview_shutter2.txt)
[up_down_manually_via_local_shutter.txt](https://github.com/zwave-js/node-zwave-js/files/6911134/up_down_manually_via_local_shutter.txt)
[up_down_manually_via_local_shutter.txt](https://github.com/zwave-js/node-zwave-js/files/6911135/up_down_manually_via_local_shutter.txt)
[reinterview_shutter2.txt](https://github.com/zwave-js/node-zwave-js/files/6911136/reinterview_shutter2.txt)
[Screenshots_before_and_during_progress.pdf](https://github.com/zwave-js/node-zwave-js/files/6911137/Screenshots_before_and_during_progress.pdf)

closed 2021-08-15 by AlCalzone

je-mei, 2021-09-12:
Roller Shutter 3 runs great now, thanks for your work!

Kind regards 
Jens

PR #3196 (2021-08-15): fix: detect shared assoc groups with root before testing assoc status
fixes: #3099