zwave-js/zwave-js #3111: Aeon Labs Home Energy Meter Gen5 not updating values automatically

github-0, 2021-07-29:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Opening a new issue as requested [here](https://github.com/zwave-js/node-zwave-js/issues/3084#issuecomment-888635081).

The HEM does not send automatic updates even though it is configured to do so. Values get updated with manual refresh. 

("I have checked the troubleshooting section and my problem is not described there. *" : Unfortunately I wasn't knowledgeable enough to complete the steps described [here](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/no-updates))

### Device information

Manufacturer: Aeon Labs
Model name: Home Energy Meter Gen5 (ZW095)
Node ID in your network: 8

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver version: 8.0.5

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Automatic updates are working with the same device and settings on Openhab and its Z-Wave implementation. 

### Attach Driver Logfile

[interview_zw095.txt](https://github.com/zwave-js/node-zwave-js/files/6899785/interview_zw095.txt)

AlCalzone, 2021-07-29:
Any chance you have access to zwavejs2mqtt? The official HA integration does not seem to include all log messages in the output, although you set it to debug.

github-0, 2021-07-29:
Hopefully this works better:

[zwavejs_2021-07-29.log](https://github.com/zwave-js/node-zwave-js/files/6900378/zwavejs_2021-07-29.log)

AlCalzone, 2021-07-29:
The associations etc. look okay, but for reference could you look up how they are configured in OpenHAB?

I don't see any reports come in after the interview, but this could be due to the configuration. You currently only have Power reporting enabled for the whole device (none of the single clamps) and these settings:
[image]

Maybe your power consumption is so stable that this relative change of 50W isn't reached?

github-0, 2021-07-29:
A bit of a setback happened. I loaded my Openhab instance just to see if the HEM is still sending automatic updates. It is not, and I'm not sure what caused this change. As far as I can remember, the only things I did were 
a) add HEM to HA using its Z-Wave integrations / addons 
b) just after your last post, change the Selective Reporting to disabled through zwavejs2mqtt control panel to make sure the thresholds wouldn't be a problem

In Openhab (through the Z-Wave control software Habmin), I also changed the Power Change Thresholds (options 4 and 8) to 1. With these settings the automatic updates for Power (Whole HEM) should happen every 5 seconds OR if the Selective Reporting would be enabled, after 1W or 1% of change. 

As this is the case, I'm guessing Openhab isn't much of a reference anymore. 

A bit of background from Openhab where things worked for two years prior to this: it was a PITA to get working even in there and judging by [this thread](https://community.openhab.org/t/solved-home-energy-meter-hem-g3-aeon-labs-gen5-zw095-do-not-report-automatically/44579), still is. [This](https://community.openhab.org/t/solved-home-energy-meter-hem-g3-aeon-labs-gen5-zw095-do-not-report-automatically/44579/57) is my comment of how I got it working. The latest posts link to a [tutorial](https://community.openhab.org/t/make-aeotec-home-energy-meter-send-automatic-updates/114287) that someone made which correlates with my own experience:
- HEM has a problem/is challenging with associations and the tools within Openhab do not set them in such a way that the automatic updates would work
- From Aeotec support:

> "There is an issue when using Multichannel Association which causes the HEM Gen5 to not report correctly. One way to resolve this issue is to remove the Multichannel Association of Node ID 01 endpoint 1 from Group 1. Then re-associate Node ID 01 to Group 1 using “Group Association”. Alternatively you could use “Multichannel Association” and assign Node ID 01 endpoint 0 to Group 1 which will work as well."

- Without understanding much from the above, redoing the association with 3rd party tools (in my case Zensys Tools) fixed the problem

Sorry about the lengthy post. Hopefully this will give some clues for next steps? I know I could probably try to do the magic with Zensys Tools again but would of course prefer a proper fix.

AlCalzone, 2021-07-29:
> Alternatively you could use “Multichannel Association” and assign Node ID 01 endpoint 0 to Group 1 which will work as well."

Hmm, but that's what `zwave-js` does:
```
2021-07-29T12:08:29.473Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[MultiChannelAssociationCCReport]
                                      group id:           1
                                      maximum # of nodes: 5
                                      node ids:
                                      endpoints:          1:0
```
Group 1 is assigned with a multi channel association to Node 1, endpoint 0. 

Not sure about HA, but zwavejs2mqtt lets you configure the associations to the full extent. I'm guessing that shows you a target of node 1 without endpoint now? If that is the case and it works, we can add configuration that forces zwave-js to use a node association instead of a multi channel association.

github-0, 2021-07-30:
With lack of better terms, I'm going to call what is presented in the Groups list as '0 Lifeline NodeID_1 None' as "None" and '0 Lifeline NodeID_1 0' (the default one) as "0".

After some testing: when None is added to the Groups in Z-Wave JS Control Panel so that both that and 0 are present, automatic updates start to work. Great! 

[image]

Log file when both None and 0 are in the Groups list and automatic updates are working: 
[zwavejs_2021-07-30.log](https://github.com/zwave-js/node-zwave-js/files/6906488/zwavejs_2021-07-30.log)

Not that I needed to do anything else, but after adding None I tried to remove 0 from the Groups and this is when things got weird (this is reproducible):
- when 0 is removed automatic updates stop working
-- if a new interview is done at this point 0 is back and None is gone
- if after removing 0, None is added again (even though it's still on the Groups list), automatic updates start to work but stop working when a new interview is done; after the interview only 0 is present and automatic updates no longer work

Interviewing when both 0 and None are present on the Groups list doesn't break anything: they both stay on the list and automatic updates still work which is good. 

I don't know if this information is of any use but it sure confused me for a good while.

AlCalzone, 2021-07-30:
Please don't re-interview after changing the associations manually.
I assume that you just need one of the two, question is which one.

github-0, 2021-07-30:
Adding "None" makes things work. Let me know if you need me to test or log anything else.

AlCalzone, 2021-07-30:
So to be clear, you now only have one association to node 1, endpoint none?
And this is working?

github-0, 2021-07-30:
No, I have both like the screenshot above. 

When association to node 1, endpoint 0 is removed automatic updates stop working. 

It is possible to achieve a scenario where the updates work and the control panel only shows association to node 1, endpoint none - however this seems fiddly as I have to re-add association to node 1, endpoint none even though it is still on the Groups list after removing association to node 1, endpoint 0.

AlCalzone, 2021-07-30:
> this seems fiddly as I have to re-add association to node 1, endpoint none even though it is still on the Groups list after removing association to node 1, endpoint 0.

I think that's actually the setup we need. Can you make a logfile of you trying to achieve that status? I'd like to see what exactly is being configured here.
Not that the UI is doing something weird here.

github-0, 2021-07-30:
- Removed all association groups
- Interview
- Association groups at this point:
 0	Lifeline	NodeID_1	0
- **Start logging**
- Add  _0	Lifeline	NodeID_1	None_ (updates start to work)
- Remove _0	Lifeline	NodeID_1	0_ (updates stop working)
- Add  _0	Lifeline	NodeID_1	None_ (updates start to work)
- Association groups at this point:
 0	Lifeline	NodeID_1	None

[zwavejs_2021-07-31.log](https://github.com/zwave-js/node-zwave-js/files/6910189/zwavejs_2021-07-31.log)

AlCalzone, 2021-08-01:
Okay, something strange is going on when you remove the association to endpoint 0, but it looks like I was on the right track.