zwave-js/zwave-js #3118: Group 3 missing at device NAS-WR01ZE

koburg, 2021-07-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

 Not Sure if "BUG REPORT" is the right place else please tell me where to put next time.

the NAS-WR01ZE  wall plug is missing an association group (3)
"GROUP 3 allows for Send Notification to associated devices in this group. This Group Support: NOTIFICATION_REPORT_V4" 

this is not visible to the device when added tried reinterview no luck. 

Firmware: 2.23

[image]

### Device information

NAS-WR01ZE
af Shenzhen Neo Electronics Co., Ltd.
Firmware: 2.23

Z-Wave-info
Node-id: 20
Nodestatus: I live (Alive)
Node klar: Ja (node ready : YES)

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

`zwavejs2mqtt` version 0.23.2

JS 
Driverversion: 8.0.5
Serverversion: 1.9.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6905250/zwavejs2mqtt.log)

koburg, 2021-07-30:
Manual 
[Smart-Plug User Guide V2.pdf](https://github.com/zwave-js/node-zwave-js/files/6905277/Smart-Plug.User.Guide.V2.pdf)

AlCalzone, 2021-07-30:
Please make a driver log, loglevel `debug` or `silly` and attach it here as a file. While it is logging, please re-interview the device. The log you posted is not a driver log. ==> https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

We don't have any associations specified in the [file for the WR01ZE](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0258/nas-wr01ze.json), so those two groups are the ones the device reported. Maybe the one you're looking for was added in another firmware version?

closed 2021-08-09 by zwave-js-assistant[bot]