zwave-js/zwave-js #3119: FGS211/FGS221 not return status.

Ascariota, 2021-07-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Hello,
JS Zwave - FGS211 / FGS221 not return status

since the last update of JS Zwave (Current version: 0.1.32 ) my FGS211 / FGS221 not return the status of one switch.
He works but I don't know if is switch on or off.

Exactly the same issue I have in april : #48744

I open an issue : https://github.com/home-assistant/core/issues/53515
and after analysis "Kpine" told me to open a bug report to have help.

### Device information

Manufacturer:  Fibaro 
Model name: FGS211/FGS221
Node ID in your network: NODE 06

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: JS ZWAVE
Current version: 0.1.34 

CORE Version
core-2021.7.4

SUPERVISOR Version
supervisor-2021.06.8

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It's work before Udpate.

### Attach Driver Logfile

Log after a re-interviewing Node06.

[lognodeZw.txt](https://github.com/zwave-js/node-zwave-js/files/6905296/lognodeZw.txt)

AlCalzone, 2021-07-30:
Thanks for the report. I think this is an edge case we haven't handled yet.

The node (endpoint 0) supports both Multi Channel and normal associations, but none of the extra endpoints do. By default, we try to setup normal associations on all endpoints and fall back to a multi channel association on endpoint 0 when it didn't work on the other endpoints.

However since setting up the associations on the extra endpoints never happens, this fallback doesn't happen either.

You can work around this for now by changing the association on the node (group 3) from target "Node 1" to "Node 1, Endpoint 0". Not sure if HA lets you do that yet, but zwavejs2mqtt will.

Ascariota, 2021-07-30:
Hello, 
Thanks for the help, I use JS Zwave integrated and I can't change nothing :(

My work around is an automation every minute with a zwave_js.refresh_value.
It's not the best solution but waiting solution it's do the job.

AlCalzone, 2021-07-30:
You can install the zwavejs2mqtt addon and use it for the change, then go back to the integration.

Ascariota, 2021-07-30:
AlCalzone, ok ZwaveJs2Mqtt run :)
I'm not sure to understand well (my English is not perfect) it's correct like this : 

[image]

Thx

AlCalzone, 2021-07-30:
@robertsLando shouldn't there be a target endpoint selection?

robertsLando, 2021-08-02:
Target endpoint should be displayed under the target node but just if group `multiChannel` is true

AlCalzone, 2021-08-02:
Based on the log the group is multi channel. Does a node dump include this info?

robertsLando, 2021-08-02:
It should

AlCalzone, 2021-08-02:
Ok, @Ascariota could you make a JSON dump of the node in question and upload it here?

Ascariota, 2021-08-02:
@AlCalzone :  yes here a full Json dump
The nodeId of this device is "nodeId": 6 

Thx

[zwave_js_dump (3).txt](https://github.com/zwave-js/node-zwave-js/files/6917886/zwave_js_dump.3.txt)

AlCalzone, 2021-08-02:
@robertsLando The association groups don't seem to be included there.

kpine, 2021-08-02:
That's a dump from HA. Doesn't look like the HA Python module exposes the groups.

robertsLando, 2021-08-02:
The dump shouls be done from z2m interface

Ascariota, 2021-08-02:
From Z2m

[node_6.txt](https://github.com/zwave-js/node-zwave-js/files/6918387/node_6.txt)

kpine, 2021-08-02:
Unless I'm missing something, the z2m dump does not include the _defined_ associations either, it only shows the available groups.

AlCalzone, 2021-08-02:
> if group `multiChannel` is true

Ok it looks like that is working correctly. That's not a Z-Wave+ device, and the lifeline is the last non-multichannel group. That means the fix I did doesn't actually affect this device.

AlCalzone, 2021-08-02:
@Ascariota can you make another log where you just toggle the device so I can see if anything gets received at all?

Ascariota, 2021-08-03:
Yes here the Silly Log

My action : 
Switch1 No return Status in HA
Switch2 Return Status in HA correctly
On the same device node 006

13:03 Switch Of Manually Switch 1 
13:04 Switch On Manually Switch 1
13:05 Switch Off Manually Switch 2
13:06 Switch On Manually Switch 2

[zwave_js (1).log](https://github.com/zwave-js/node-zwave-js/files/6923580/zwave_js.1.log)

AlCalzone, 2021-08-03:
Okay, I know what's going on. Endpoint 1 reports via the root endpoint, so we need to map the reports back. This was changed to opt-in in v8.

PR is up in a few seconds.

AlCalzone, 2021-08-03:
You can work around this until the release:
Take this file [fgs221_2.1-2.3.json.txt](https://github.com/zwave-js/node-zwave-js/files/6923666/fgs221_2.1-2.3.json.txt), rename it back to `.json` and put it in your user config dir in zwavejs2mqtt:
https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

After a restart, things should work again.

Ascariota, 2021-08-03:
@AlCalzone : 
I'm not sure of the path : //root/config/custom_components/
I need to create a folder zwavejs2mqtt ?

[image]

Big thx for help again :)

AlCalzone, 2021-08-03:
Zwavejs2mqtt has a web UI that let's you manage files in its store directory.

Ascariota, 2021-08-03:
I made a mistake ? 
[image]

After I reboot the complet HOST but nok.

I change the label 
"label": "Relay 1: Response to Water Flood Alarm"  to "label": "Relay 1: Response to Water Flood Alarma"  just to see if the news JSON is ok but I ZW2M I see the old label, so I sure I made a mistake with th json path/name... 

Sorry usually I'm not so noob.

AlCalzone, 2021-08-03:
The driver log should show you at the start if it picked up the new file or the included one.

kpine, 2021-08-03:
> You can work around this until the release:
> Take this file [fgs221_2.1-2.3.json.txt](https://github.com/zwave-js/node-zwave-js/files/6923666/fgs221_2.1-2.3.json.txt), rename it back to `.json` and [...]

@AlCalzone FYI, this attached file doesn't have the compat flag. I think [this](https://raw.githubusercontent.com/zwave-js/node-zwave-js/fd5460644c7a8e5d79959b0b0e06322d4c446b82/packages/config/config/devices/0x010f/fgs221_2.1-2.3.json) is the file?

Ascariota, 2021-08-04:
@AlCalzone 

I try unistall and install again Zwave addon.
I put the file in the config folder.
Zwave run in HA I can use my devices, but the driver log 
Driver: Error Resource temporarily unavailable Cannot lock port.

And I think the JSON is not load.

```
[08:29:11] INFO: Starting NGinx...
2021-08-04 08:29:12.950 INFO APP: GET / 304 17.372 ms - -
2021-08-04 08:29:13.100 INFO APP: GET /api/auth-enabled 304 7.406 ms - -
2021-08-04 08:29:13.354 INFO APP: GET /api/settings 200 114.642 ms - 11864
2021-08-04 08:29:13.384 INFO APP: GET /api/auth-enabled 304 2.022 ms - -
2021-08-04 08:29:14.655 INFO ZWAVE: Connecting to /dev/serial/by-id/usb-0658_0200-if00
2021-08-04 08:29:14.681 INFO ZWAVE: Controller status: Driver: Error Resource temporarily unavailable Cannot lock port
2021-08-04 08:29:14.682 WARN ZWAVE: Retry connection in 3 seconds...
2021-08-04 08:29:17.690 INFO ZWAVE: Connecting to /dev/serial/by-id/usb-0658_0200-if00
2021-08-04 08:29:17.747 INFO ZWAVE: Controller status: Driver: Error Resource temporarily unavailable Cannot lock port
2021-08-04 08:29:17.748 WARN ZWAVE: Retry connection in 3 seconds...
```

AlCalzone, 2021-08-04:
Yeah, you're right. 🙈

closed 2021-08-04 by AlCalzone

Ascariota, 2021-08-05:
Ok 👍

PR #3142 (2021-08-02): fix: interview MC first and require MCA on root if no endpoint has associations
fixes: #3119
PR #3149 (2021-08-04): fix(config): map root reports to endpoint 1 for Fibaro FGS211/221
fixes: #3119