zwave-js/zwave-js #3128: Unable to update firmware on Heltun devices such as TPS03 and RS01

johanschelin, 2021-07-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What happens:
Unable to update firmware of my Heltun devices 

What do you observe?
In the Web UI I get the errormessage:
"Error while calling api beginFirmwareUpdate: Unable to extract firmware from file: Could not parse HEX firmware file!"

What did you expect to happen?
Firmware upgrade to complete succesfully

Steps to reproduce the behavior:
  1. Go to 'ZWave-JS UI' and click the device you want to upgrade
  2. Click on 'Advanced'
  3. Click on Begin in the section 'Firmware Update'
  4. Click OK to accept target 0
  5. Choose the firmware file
  6. Now the UI gives you the errormessage "i get the errormessage:
  "Error while calling api beginFirmwareUpdate: Unable to extract firmware from file: Could not parse 
   
[Uploading Heltun firmwares.zip…]()
HEX firmware file!""

### Device information

Manufacturer: Heltun
Model name: Multiple, in my example tested the TPS03
Node ID in your network: <4>

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwavejs2mqtt: 5.4.2
zwave-js: 8.0.5
home id: 4110551221
home hex: 0xf50208b5

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2021-07-31.log](https://github.com/zwave-js/node-zwave-js/files/6911697/zwavejs_2021-07-31.log)

AlCalzone, 2021-08-01:
Uploading the firmware files didn't work, can you try again?

johanschelin, 2021-08-01:
[Heltun firmwares.zip](https://github.com/zwave-js/node-zwave-js/files/6912380/Heltun.firmwares.zip)

johanschelin, 2021-08-01:
Working now?

AlCalzone, 2021-08-01:
yep

AlCalzone, 2021-08-01:
`.ota` files usually contain data in the Intel HEX format. These files seem to be binary and its the first time I've seen that.

I'm not sure if its a good idea to detect that and attempt the firmware update without HEX decoding first. You can try the update on your own risk by renaming the files to `.bin` first.

johanschelin, 2021-08-01:
I tested just renaming the files from .ota to .bin and it actually worked. Not sure what we should proceed with, I can report back to Heltun that they could provide the files with the .bin suffix instead - but I have succesfully updated firmwares to my Heltun products via both Z-Wave.me and Simplicity Studio - so maybe we can just add support for the .ota suffix to ZWaveJS and everything will work.

AlCalzone, 2021-08-02:
> so maybe we can just add support for the .ota suffix to ZWaveJS and everything will work.

The `.ota` suffix is supported and expected to contain data in the Intel HEX format. As always with Z-Wave this is poorly documented and I've only found a reference in the document INS13477

> **OTA:**  
> Over the air (e.g. making a firmware update wireless). Also used as **S-Record hex** 
file extension (*.ota) for programs to be sent over the air achieving a firmware 
update of the device in question. The device must of course have a bootloader 
supporting the **S-Record hex file** format.

johanschelin, 2021-08-02:
Ok - I will inform @***@***.******@***.***> about that so they can modify their .ota files.
I can test the files for them.

Hämta Outlook för iOS<https://aka.ms/o0ukef>

johanschelin, 2021-08-04:
Got this answer from Heltun, so a fix in ZWaveJS sounds like the correct way to go?
----------
Our OTA files are encrypted using SiLab’s OTA encryption utilities and cannot be read as Intel HEX file.
This is the problem on the Controller side which is trying to decode this OTA file. It is where “Unable to extract firmware from file: Could not parse“ message comes from.
Actually, Controller has to open file as is and should not try any decoding because OTA already means file is fully prepared to be sent over the Air.

There are two way to resolve this problem
•	Fix Controller SW to not try to parse OTA files
•	Before of update, user may just change file extension: from *.ota to *.bin for example

AlCalzone, 2021-08-04:
> Controller has to open file as is and should not try any decoding because OTA already means file is fully prepared to be sent over the Air.

I don't agree with this statement. That would mean other devices would be able to decode the Intel HEX data thats found in their `.ota` update files.

I'll do some inquiries.

johanschelin, 2021-08-04:
Thanks!
For information - the .ota file works perfect as is when updating via Razberry and Simplicity Studio, its only with ZWaveJS i cannot get it to work.
And it works if renaming to .bin in ZWaveJS

AlCalzone, 2021-08-04:
So, I've asked around. `.ota` files containing data in Intel HEX format used to be a convention, but binary data like in this case is also possible.

AlCalzone, 2021-08-04:
Yep, I'll do as I typed above. Test if the contents are Intel HEX, if not treat them as binary.

PR #3281 (2021-09-05): fix: fall back to interpreting OTA/OTZ firmware files as binary
fixes: #3128
closed 2021-09-05 by AlCalzone