zwave-js/zwave-js #3145: Device configuration missing values

Seraph1m89, 2021-08-02:
Hi, I have a dimmer from ConnectHome manufacturer (0x0179). This device is not yet in the database but I have a config that is working with no problem. I have 4 of such devices and the problem is - each one of them have different set of values in it's config. For example:
3 devices have a section 32-0-currentValue and 32-0-targetValue
1 device has necessary sections to work as a dimmer:
38-0-targetValue, 38-0-duration, 38-0-currentValue, 38-0-Up, 38-0-Down
I tried to reinterview the nodes, add and remove them to the network but it does not give any result.
Sometimes after interviewing the node they are getting subset of additional values, the process is highly unstable fo me here

robertsLando, 2021-08-02:
PLease attach some silly level logs to help debug this

AlCalzone, 2021-08-02:
> Sometimes after interviewing the node they are getting subset of additional values, the process is highly unstable fo me here

That sounds like the devices have a bad connectivity to the controller and responses are getting lost. But without a driver logfile like @robertsLando wrote this is just guessing.

Seraph1m89, 2021-08-02:
Is this the logs? I don't see anything related to issues
[zwavejs_2021-08-02.log](https://github.com/zwave-js/node-zwave-js/files/6918529/zwavejs_2021-08-02.log)

Seraph1m89, 2021-08-02:
BTW - I tried to reinterview the device that was working correctly just to end up being in the same issue as with every other device. I have a config (probably from openzwave) that was left by my previous employee that has this line
<CommandClass id="32" name="COMMAND_CLASS_BASIC">
    <Compatibility>
        <Mapping>38</Mapping>
    </Compatibility>
    <State>
        <CCVersion>1</CCVersion>
    </State>
    <Instance index="1" />
</CommandClass>
But I'm just not sure how to do the same mapping in this system. Besides at one point it did work properly

AlCalzone, 2021-08-02:
I can't decipher what you posted there, but I think I know what it's supposed to mean. Node 12 first claims it supports Multilevel Switch CC:
```
2021-08-02T15:42:03.958Z CNTRLR « [Node 012] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Version
                                  · All Switch
                                  · Association
                                  · Multi Channel Association
                                  · Multilevel Switch
                                  · Manufacturer Specific
                                  · Configuration
                                  · Device Reset Locally
                                  · Association Group Information
                                  · Firmware Update Meta Data
                                  · Multilevel Sensor
                                  · Manufacturer Specific
```
but then it doesn't, when asked for the version:
```
2021-08-02T15:42:04.745Z CNTRLR » [Node 012]   querying the CC version for Multilevel Switch...
2021-08-02T15:42:04.943Z CNTRLR   [Node 012]   does NOT support CC Multilevel Switch (0x26)
```
Same happens for Nodes 16 and 17, which I guess are the same?

You can override that information by adding this to the config file - we don't seem to have an official one yet:
```json
	"compat": {
		"commandClasses": {
			"add": {
				"0x26": {
					// The device supports Multilevel Switch CC, but it responds to the CC version request with 0
					"isSupported": true,
					"version": 1
				}
			}
		}
	}
```

Seraph1m89, 2021-08-02:
Posted raw xml file peace, my mistake. Thank you, it solves the problem. What does version mean? From the documentation, it's unclear (perhaps I don't know z-wave protocol that well) - does it support versioning of command classes?

closed 2021-08-02 by Seraph1m89

AlCalzone, 2021-08-02:
Command Classes are versioned, yes. Usually the manual mentions which version a device implements, maybe the `1` needs to be something else.