zwave-js/zwave-js #3153: TKB TZ06: Switch state not automatically received by ZwaveJS - manual refresh required

BillsGate, 2021-08-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Among all my Z-wave devices I own a couple of Zipato PH-pan04 double switch modules, which are in fact rebrands of TKB TZ06 modules. They show up as TKB TZ06 in my ZwaveJS-panel.

Since a few updates of ZwaveJS, the switch state of these modules isn't received by ZwaveJS: If I push a physical switch connected to the module, the light turns on, but this state-change isn't updated in the Binary switch-panel of the module in ZwaveJS

[image]

Pressing the REFRESH button at the right side of the panel updates the values correctly, but having to refresh manually defeats the purpose of automating switch actions :)

Other Zwave-devices in my network don't have this issue.

I've added a log of re-interviewing a node and pressing both connected physical buttons afterwards. However, pressing the physical buttons doesn't show up in the log, not even in loglevel Silly.

[Related question in Q&A](https://github.com/zwave-js/node-zwave-js/discussions/3151)

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 5.4.2
zwave-js: 8.0.5
home id: 4039718769
home hex: 0xf0c93771

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

A few ZwaveJS versions back this worked perfectly. 

A while ago only the virtual switch, which gets turned on if either of the physical switches turned on, was the only thing responding in this module.
After the latest update of ZwaveJS, this switch isn't updated anymore as well.

### Attach Driver Logfile

// re-interviewing node
2021-08-04 12:09:13.051 INFO ZWAVE: Calling api refreshInfo with args: [ 16, [length]: 1 ]
2021-08-04 12:09:13.150 INFO ZWAVE: Node 16: interview started
2021-08-04 12:09:13.195 INFO ZWAVE: Success zwave api call refreshInfo undefined
2021-08-04 12:09:13.229 INFO ZWAVE: Node 16: interview stage PROTOCOLINFO completed
2021-08-04 12:09:13.348 INFO ZWAVE: Node 16: interview stage NODEINFO completed
2021-08-04 12:09:13.476 INFO ZWAVE: Node 16: value added: 114-0-manufacturerId => 280
2021-08-04 12:09:13.482 INFO ZWAVE: Node 16: value added: 114-0-productType => 1
2021-08-04 12:09:13.486 INFO ZWAVE: Node 16: value added: 114-0-productId => 4
2021-08-04 12:09:13.635 INFO ZWAVE: Node 16: value added: 134-0-libraryType => 3
2021-08-04 12:09:13.640 INFO ZWAVE: Node 16: value added: 134-0-protocolVersion => 3.41
2021-08-04 12:09:13.643 INFO ZWAVE: Node 16: value added: 134-0-firmwareVersions => 1.0
2021-08-04 12:09:15.088 INFO ZWAVE: Node 16: value added: 37-1-currentValue => false
2021-08-04 12:09:15.102 INFO ZWAVE: Node 16: value added: 37-1-targetValue => undefined
2021-08-04 12:09:15.237 INFO ZWAVE: Node 16: value added: 37-2-currentValue => false
2021-08-04 12:09:15.240 INFO ZWAVE: Node 16: value added: 37-2-targetValue => undefined
2021-08-04 12:09:15.366 INFO ZWAVE: Node 16: value added: 37-3-currentValue => false
2021-08-04 12:09:15.368 INFO ZWAVE: Node 16: value added: 37-3-targetValue => undefined
2021-08-04 12:09:15.511 INFO ZWAVE: Node 16: metadata updated: 112-0-1
2021-08-04 12:09:15.515 INFO ZWAVE: Node 16: metadata updated: 112-0-2
2021-08-04 12:09:15.519 INFO ZWAVE: Node 16: metadata updated: 112-0-3
2021-08-04 12:09:15.524 INFO ZWAVE: Node 16: metadata updated: 112-0-4
2021-08-04 12:09:15.529 INFO ZWAVE: Node 16: metadata updated: 112-0-5
2021-08-04 12:09:15.533 INFO ZWAVE: Node 16: metadata updated: 112-0-6
2021-08-04 12:09:15.594 INFO ZWAVE: Node 16: value added: 112-0-1 => 3
2021-08-04 12:09:15.713 INFO ZWAVE: Node 16: value added: 112-0-2 => 2
2021-08-04 12:09:56.148 INFO ZWAVE: Node 16: metadata updated: 113-0-alarmType
2021-08-04 12:09:56.164 INFO ZWAVE: Node 16: metadata updated: 113-0-alarmLevel
2021-08-04 12:09:56.170 INFO ZWAVE: Node 16: interview stage COMMANDCLASSES completed
2021-08-04 12:09:56.179 INFO ZWAVE: Node 16: interview stage OVERWRITECONFIG completed
2021-08-04 12:09:56.185 INFO ZWAVE: Node 16: interview stage COMPLETE completed
2021-08-04 12:09:56.220 INFO ZWAVE: Node 16: value added 16-114-0-manufacturerId => 280
2021-08-04 12:09:56.223 INFO ZWAVE: Node 16: value added 16-114-0-productType => 1
2021-08-04 12:09:56.225 INFO ZWAVE: Node 16: value added 16-114-0-productId => 4
2021-08-04 12:09:56.227 INFO ZWAVE: Node 16: value added 16-134-0-libraryType => 3
2021-08-04 12:09:56.229 INFO ZWAVE: Node 16: value added 16-134-0-protocolVersion => 3.41
2021-08-04 12:09:56.231 INFO ZWAVE: Node 16: value added 16-134-0-firmwareVersions => 1.0
2021-08-04 12:09:56.236 INFO ZWAVE: Node 16: value added 16-112-0-1 => 3
2021-08-04 12:09:56.244 INFO ZWAVE: Node 16: value added 16-112-0-2 => 2
2021-08-04 12:09:56.246 INFO ZWAVE: Node 16: value added 16-112-0-3 => undefined
2021-08-04 12:09:56.251 INFO ZWAVE: Node 16: value added 16-112-0-4 => undefined
2021-08-04 12:09:56.254 INFO ZWAVE: Node 16: value added 16-112-0-5 => undefined
2021-08-04 12:09:56.257 INFO ZWAVE: Node 16: value added 16-112-0-6 => undefined
2021-08-04 12:09:56.259 INFO ZWAVE: Node 16: value added 16-113-0-alarmType => undefined
2021-08-04 12:09:56.262 INFO ZWAVE: Node 16: value added 16-113-0-alarmLevel => undefined
2021-08-04 12:09:56.285 INFO ZWAVE: Node 16: value added 16-37-1-currentValue => false
2021-08-04 12:09:56.287 INFO ZWAVE: Node 16: value added 16-37-1-targetValue => undefined
2021-08-04 12:09:56.305 INFO ZWAVE: Node 16: value added 16-37-2-currentValue => false
2021-08-04 12:09:56.307 INFO ZWAVE: Node 16: value added 16-37-2-targetValue => undefined
2021-08-04 12:09:56.325 INFO ZWAVE: Node 16: value added 16-37-3-currentValue => false
2021-08-04 12:09:56.327 INFO ZWAVE: Node 16: value added 16-37-3-targetValue => undefined
2021-08-04 12:09:56.331 INFO ZWAVE: Node 16 ready: TKB Home - TZ06 (In Wall Dual Relay(1 Way) Switch Module 2x 1.5kW)
2021-08-04 12:09:56.359 INFO ZWAVE: Node 16: interview COMPLETED, all values are updated
2021-08-04 12:10:25.263 INFO APP: GET /health/zwave 200 13.614 ms - 1875

// switching on in app
2021-08-04 12:11:55.082 INFO ZWAVE: Node 16: value updated: 37-3-currentValue false => true
2021-08-04 12:11:56.175 INFO ZWAVE: Node 16: value updated: 37-3-currentValue true => true

// switching off in app
2021-08-04 12:12:00.240 INFO ZWAVE: Node 16: value updated: 37-3-currentValue true => false
2021-08-04 12:12:01.315 INFO ZWAVE: Node 16: value updated: 37-3-currentValue false => false

// toggling the physical switch doesn't appear in logs, even with loglevel Silly

BillsGate, 2021-08-04:
The node in question:
```       "16": {
            "id": 16,
            "interviewStage": "Complete",
            "deviceClass": {
                "basic": 4,
                "generic": 16,
                "specific": 1
            },
            "isListening": true,
            "isFrequentListening": false,
            "isRouting": true,
            "supportedDataRates": [
                40000,
                100000
            ],
            "protocolVersion": 3,
            "nodeType": "Routing End Node",
            "supportsSecurity": false,
            "supportsBeaming": true,
            "isSecure": false,
            "commandClasses": {
                "0x25": {
                    "name": "Binary Switch",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 1
                        },
                        "1": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        },
                        "2": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        },
                        "3": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        }
                    }
                },
                "0x27": {
                    "name": "All Switch",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        },
                        "1": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        },
                        "2": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        },
                        "3": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        }
                    }
                },
                "0x60": {
                    "name": "Multi Channel",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 3
                        }
                    }
                },
                "0x70": {
                    "name": "Configuration",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 1
                        }
                    }
                },
                "0x71": {
                    "name": "Notification",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 1
                        }
                    }
                },
                "0x72": {
                    "name": "Manufacturer Specific",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 2
                        }
                    }
                },
                "0x85": {
                    "name": "Association",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 1
                        }
                    }
                },
                "0x86": {
                    "name": "Version",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 1
                        }
                    }
                }
            }
        },```

AlCalzone, 2021-08-04:
What you posted is not a driver logfile => https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

Also please attach it as a file.

BillsGate, 2021-08-04:
> What you posted is not a driver logfile => https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs
> 
> Also please attach it as a file.

Very sorry, this is te first time I've reported an issue. Driver logfile is attached now.

AlCalzone, 2021-08-04:
I believe all we need to do here is restore the mapping from endpoint 0 to 1.

@zwave-js-bot add compat flag to 0x0118:0x0001:0x0004
```
"mapRootReportsToEndpoint": 1
```

AlCalzone, 2021-08-04:
@BillsGate can you test this file https://raw.githubusercontent.com/zwave-js/node-zwave-js/add-compat-flag-1628108692/packages/config/config/devices/0x0118/tz06.json as described [here](https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config)? zwavejs2mqtt lets you upload that file with the included file browser.
You'll have to restart zwavejs2mqtt for the change to take.

BillsGate, 2021-08-05:
@AlCalzone Thank you for your swift reply! 

Adding that config file makes the group switch respond again if I toggle a physical switch, so that's half the problem solved:

[image]

However, the switch I toggled still remains off in the binary switch panel until I click refresh.
Is there anyway that can be fixed? Is there anything I can do/try to fix this?

AlCalzone, 2021-08-05:
Can you make a driver log of that too please?

BillsGate, 2021-08-05:
[zwavejs_2021-08-05.log](https://github.com/zwave-js/node-zwave-js/files/6936224/zwavejs_2021-08-05.log)

I've tried to set "isLifeline": false on association 1 and rebooted ZwaveJS as per [this suggestion](https://zwave-js.github.io/node-zwave-js/#/config-files/file-format?id=associations), but that didn't help.

AlCalzone, 2021-08-05:
Your log looks strange and is randomly missing a bunch of info. Can you try again?

BillsGate, 2021-08-05:
[zwavejs_2021-08-05-1.log](https://github.com/zwave-js/node-zwave-js/files/6936773/zwavejs_2021-08-05-1.log)
Here's a new log. I've generated it like this:

- Enable Zwave-JS logging to file, loglevel Silly, only logging node 16
- Wait approx 30 seconds
- Turn on switch 1, wait 10 seconds, turn off switch 1, wait 10 seconds
- Turn on switch 2, wait 10 seconds, turn off switch 2
- Wait approx 30 seconds
- Disabled logging to file, downloaded generated file

AlCalzone, 2021-08-05:
> only logging node 16

Ah, maybe that's why it looks like some logs are missing.

Both your switches send a report via the root endpoint without saying which switch the report came from. With the updated config that report now gets mapped to endpoint 1 (switch 1), but it doesn't look like we'll be able to figure out which switch was toggled.

Looking at the manual for the [Zipato PAN04](https://www.zipato.com/wp-content/uploads/2015/09/ph-pan04-Zipato-Micro-Module-Switch-Double-User-Manual-v1.4.pdf), I also don't understand how that is supposed to work, since the device doesn't support multi channel associations and seems to report everything without source endpoint information.

What I think **could** work:
* We change the config file so that the root endpoint gets preserved
* You configure the device to report ON when either relay is on and OFF when both are off (param 3: default)
* and set up an automation that polls the endpoint values whenever the root value is updated

BillsGate, 2021-08-05:
That sounds like something we could absolutely try, thank you!

> Ah, maybe that's why it looks like some logs are missing.
Not sure if still needed, but this is a log file following same steps as above, but not limited to node 16
[zwavejs_2021-08-05-2.log](https://github.com/zwave-js/node-zwave-js/files/6937551/zwavejs_2021-08-05-2.log)

I'm not sure if the manual of the TKB TZ06 is any more helpful: 
https://doc.eedomus.com/files/Manual%20-%20TZ06%20manual.pdf

AlCalzone, 2021-08-07:
Here's a config file for you to test again [tz06.json.txt](https://github.com/zwave-js/node-zwave-js/files/6949478/tz06.json.txt)
Like we agreed, this preserves the root endpoint's values, so you can poll the endpoint values automatically whenever the root changes. Let me know if that helps.

BillsGate, 2021-08-08:
Thank you so much for your effort!

As far as I can see, the latest config adds a switch entity which turn if either of the physical switches turns on. 
However, such an endpoint already existed and since your config update of three days ago that worked perfectly:
[image]

The only issue left is that the physical buttons now aren't monitored individually, which means (in case of this specific screenshot) I can't automate a different timer for each light on this switch module. For me that's easy to work around, it's absolutely fine for me to have one timer for both bulbs.

In short: You've helped me a great deal! I can automate my light switches again. Thank you ever so much!

AlCalzone, 2021-08-08:
> The only issue left is that the physical buttons now aren't monitored individually

That's what I meant above. You need to set up an automation in HA that polls (refreshes) the individual switches' states whenever the "group" state changes.

PR #3164 (2021-08-10): fix(config): preserve root endpoint values for TZ06
fixes: #3153
closed 2021-08-10 by AlCalzone