zwave-js/zwave-js #3154: [bug] Inovelli LZW31-SN Target #1 firmware update issue

kreene1987, 2021-08-04:
### Version

**Checklist:**

- [X ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X ] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [X ] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt: 5.4.2
zwave-js: 8.0.5

### Describe the bug

Attempt to update target 1 of my LZW31-SN and receive the following error:

"Error while calling api beginFirmwareUpdate: Failed to start the update: Firmware target zwave-js/zwavejs2mqtt#1 not found on this node!"

### To Reproduce

Steps to reproduce the behavior:

1. Go to Node, select Firmware Update button.
2. Click on Target 1
3. Select File from Desktop location (local on PC)
4. See error

### Expected behavior

Prior functionality allowed updating of Target 0 and Target 1.

### Additional context

Hasn't been working for approximately 1 week.

AlCalzone, 2021-08-04:
I'll need to see a [driver logfile](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs) (loglevel debug) of such a failed upgrade attempt.

The selection of firmware targets is based on what the device reports, so there was probably something wrong with that.

kreene1987, 2021-08-04:
[zwavejs_2021-08-04.log](https://github.com/zwave-js/node-zwave-js/files/6934388/zwavejs_2021-08-04.log)

See if that contains what you are looking for? Should be a cold boot and attempt at the firmware update at the end.

AlCalzone, 2021-08-04:
`additional firmware IDs: [4660]`
Seems weird if you ask me, but seems to be correct. I was under the impression that these were supposed to be 0, 1, 2...

kreene1987, 2021-08-04:
I have performed a firmware update using ZJ2M a few times before with this device type and exact firmware files on Targets 0 and 1, just to give all info that comes to mind.

This is an Inovelli LZW31-SN, which is having trouble reporting power. I am working with them to diagnose it and this firmware version seems to be working. Most of my other devices had already been updated but this one was a new/fresh install.

AlCalzone, 2021-08-04:
I think I know what's wrong. Two different assumptions in the code about the firmware target that contradict each other.

kreene1987, 2021-08-06:
Issue persists on latest version (5.4.5).

AlCalzone, 2021-08-07:
With the same error message?

kreene1987, 2021-08-07:
Yes correct. Can get logs but identical message about not finding target 1 on node.

AlCalzone, 2021-08-07:
Nevermind - had an off-by-1 error in the check. The benefit of not having such devices to test 🤷🏻‍♂️

closed 2021-08-10 by AlCalzone

stephenjamieson, 2021-08-10:
I'm not sure if this is fixed, but then again I'm not sure I'm building zwavejs2mqtt correctly. I pulled the latest node-zwave-js with git, ran `yarn; yarn run build`, then in zwavejs2mqtt, ran `yarn add ../node-zwave-js/packages/zwave-js`.  It built successfully and I used it to initiate a firmware upgrade on target 1 but saw the same message. But again, I am not familiar with npm/node/yarn so I may be doing something wrong.

```
2021-08-10 21:44:10.468 INFO ZWAVE: Calling api beginFirmwareUpdate with args: [
  26,
  'LZW31-SN_1.44.bin',
  <Buffer 50 13 00 20 ad 28 00 00 0f 3c 00 00 15 3a 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff e1 3d 00 00 ff ff ... 15482 mo
  1,
  [length]: 4
]
2021-08-10 21:44:10.646 INFO ZWAVE: Failed to start the update: The node does not support upgrading a different firmware target than 0! beginFirmwareUpdate undefined
```

AlCalzone, 2021-08-11:
@stephenjamieson You have a different message. Your node doesn't seem to support updating a different firmware than the main one (CC version too old)

stephenjamieson, 2021-08-11:
Ok, thank you @AlCalzone, I'll wait for others to chime in because I'm using the same device (LZW31-SN) which requires target 1 to be upgraded too

AlCalzone, 2021-08-11:
Try re-interviewing the device - maybe the known version information is wrong.

stephenjamieson, 2021-08-11:
No luck, same error.  If I try again after the first time it also shows the following

```
2021-08-11 16:11:09.556 INFO ZWAVE: Calling api beginFirmwareUpdate with args: [
  26,
  'LZW31-SN_1.44.bin',
  <Buffer 50 13 00 20 ad 28 00 00 0f 3c 00 00 15 3a 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff e1 3d 00 00 ff ff ... 15482 more bytes>,
  1,
  [length]: 4
]
2021-08-11 16:11:09.557 INFO ZWAVE: Failed to start the update: A firmware upgrade is already in progress! beginFirmwareUpdate undefined
```

But if I try to abort:

```
2021-08-11 16:12:45.759 INFO ZWAVE: Calling api abortFirmwareUpdate with args: [ 26, [length]: 1 ]
2021-08-11 16:12:45.760 INFO ZWAVE: Success zwave api call abortFirmwareUpdate undefined
```

Hoping @kreene1987 can try

kreene1987, 2021-08-11:
> No luck, same error. If I try again after the first time it also shows the following
>
> ```
> 2021-08-11 16:11:09.556 INFO ZWAVE: Calling api beginFirmwareUpdate with args: [
>   26, [...]

If you are getting a firmware update already in progress, you can restart the docker container and it should resolve that. We would be looking for the identical issue from OP.

stephenjamieson, 2021-08-11:
I'm not exactly sure what happened, but after I restarted zwave2mqtt again, after re-interviewing, it successfully updated target 1. Thank you for the help!

mtvg, 2021-08-13:
@AlCalzone Is there an ETA for the next release that will contain this fix?

AlCalzone, 2021-08-13:
Some time this weekend

PR #3158 (2021-08-04): fix: firmware target selection for target > 0
fixes: #3154
PR #3163 (2021-08-10): fix: use correct index to verify existence of additional firmware
fixes: #3154