zwave-js/zwave-js #3166: Refresh Configuration CC values on load if the device config Configuration info has changed

raman325, 2021-08-08:
**Is your feature request related to a problem? Please describe.**
A user reported an issue here: https://github.com/home-assistant/core/issues/54181 related to an out of date config file. They had updated to a version of zwave-js that had the fixed config file, but the state of the Configuration CC must have been cached from previously. We asked them to do a re-interview (really a refreshCCValues would have done it) and that resolved the problem.

**Describe the solution you'd like**
It seems to me like there should be a better way to handle this that doesn't require user intervention. The solution I suggested in the summary may not be the best one, but it does express the intent I am trying to get to.

**Describe alternatives you've considered**
Re-interview or refresh CC values will do it but requires user intervention.

**Additional context**
Add any other context or screenshots about the feature request here. If you want to propose an API, share your thoughts how it could be used (syntax).

AlCalzone, 2021-08-09:
We've discussed similar things in the past. Some config file changes require a re-interview for the changes to be applied, but I'm opposed to doing that automatically. Worst case something goes wrong while the user has no chance to intervene. If someone has built an automation that controls a door lock, breaking that could be really bad.

What I could imagine is some kind of status that notifies the user that re-interviewing might be necessary. Which is a
Duplicate of #2909

AlCalzone, 2021-08-09:
Duplicate of #2909

closed 2021-08-09 by AlCalzone