zwave-js/zwave-js #3173: ZW111 not reporting values for kwh, volts or amps

omriasta, 2021-08-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I have 3 Aeotec ZW111 nano dimmers, all 3 are not reporting kwh, volts or amps properly
What do you observe?
Volts and amps are both always at 0
kwh is at a static value which is different for each dimmer (one is at 39.9kwh, one at 19.5kwh and the last at 21.8kwh)
The only field that appears to be properly updating is watts
What did you expect to happen?
volts and amps should report properly. As far as kwh that value should be changing with time and there should be a way to reset that value.
The logs also only show the watts being updated, no entries for volts or amps
Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Aeotec
Model name: ZW111
Node ID in your network: 31,33,35

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.34
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Not sure, just noticed this with the new release of Energy Dashboard

### Attach Driver Logfile

```
[... 78 lines cut ...]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      32.68
                                      time delta: 0 seconds
2021-08-10T16:42:49.521Z SERIAL « 0x01090004002303260300f7                                            (11 bytes)
2021-08-10T16:42:49.524Z CNTRLR   [Node 035] [~] [Multilevel Switch] currentValue: 0 => 0           [Endpoint 0]
2021-08-10T16:42:49.526Z CNTRLR   [Node 035] Scheduled poll canceled because value was updated
2021-08-10T16:42:49.527Z SERIAL » [ACK]                                                                   (0x06)
2021-08-10T16:42:49.531Z DRIVER « [Node 035] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: 0
2021-08-10T16:42:50.829Z SERIAL « 0x0114000400230e3202217400000000000000000000a7                      (22 bytes)
2021-08-10T16:42:50.831Z CNTRLR   [Node 035] [~] [Meter] value[66049]: 32.68 => 0                   [Endpoint 0]
2021-08-10T16:42:50.834Z SERIAL » [ACK]                                                                   (0x06)
2021-08-10T16:42:50.836Z DRIVER « [Node 035] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds

```

AlCalzone, 2021-08-10:
Did you configure reporting properly? -> Parameters 101-103/111-113, https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x006f:0.0

omriasta, 2021-08-10:
Hmmm 101,102 are not editable for me in homeassistant. I can edit 103 but it seems to revert to 0...
111 is set to 3, 112,113 are set to 600.
Any idea what number I would put in 103 to get everything reporting?

omriasta, 2021-08-10:
That was it, I installed the zwave mqtt addon instead and was able to change the values over there, now all is working properly. I will try to open a ticket for the built in addon that these fields can't be configured.

closed 2021-08-10 by omriasta

AlCalzone, 2021-08-10:
We have those defined as several partial parameters with clear names and enable/disable options (see the link I posted). Chose the ones you need and enable them. Most are disabled by default.