zwave-js/zwave-js #3180: Add support for ZRH-12

Kingcool88, 2021-08-12:
Would love to have support for Fakros ZRH12 remote, config available on openzwave git
https://github.com/OpenZWave/open-zwave/blob/master/config/fakro/zrh12.xml

[zrh12.txt](https://github.com/zwave-js/node-zwave-js/files/6975132/zrh12.txt)

AlCalzone, 2021-08-12:
Update: We do have a config file for it, so you should be all set.

Kingcool88, 2021-08-12:
ZRH12 is a remote to the windows. So its not the roller shutter.

Kingcool88, 2021-08-12:
> I'm confused. The OZW file you linked to resolves to this device:
> https://products.z-wavealliance.org/products/677 - **ARZ Z-Wave Roof Window Roller Shutter**
> which we do have here: https://devices.zwave-js.io/?jumpTo=0x0085:0x0002:0x0002:0.0
> 
> Is that just the same device under a different name?

ZRH12 is a remote to the windows. So its not the roller shutter ARZ.
When adding right now, we get unknown device so not all functions work as they should.
Unknown manufacturer 0x0085 Product 0x0001  Unknown product 0x0004

AlCalzone, 2021-08-12:
Okay then that OZW file is wrong because it contains the info for the shutter.
Please post a driver log, level debug where you interview the remote.

Kingcool88, 2021-08-12:
> Okay then that OZW file is wrong because it contains the info for the shutter.
> Please post a driver log, level debug where you interview the remote.

> 2021-08-12 15:10:34.611 INFO ZWAVE: Node 64: interview started
2021-08-12 15:10:34.622 INFO ZWAVE: Success zwave api call refreshInfo undefined
2021-08-12 15:10:34.636 INFO ZWAVE: Node 64: interview stage PROTOCOLINFO completed
2021-08-12 15:10:54.180 INFO ZWAVE: Node 64 is now awake
2021-08-12 15:10:54.353 INFO ZWAVE: Node 64: interview stage NODEINFO completed
2021-08-12 15:10:54.437 INFO ZWAVE: Node 64: value added: 132-0-wakeUpInterval => 0
2021-08-12 15:10:54.439 INFO ZWAVE: Node 64: value added: 132-0-controllerNodeId => 1
2021-08-12 15:10:54.491 INFO ZWAVE: Node 64: value added: 114-0-manufacturerId => 133
2021-08-12 15:10:54.493 INFO ZWAVE: Node 64: value added: 114-0-productType => 1
2021-08-12 15:10:54.495 INFO ZWAVE: Node 64: value added: 114-0-productId => 4
2021-08-12 15:10:54.610 INFO ZWAVE: Node 64: value added: 134-0-libraryType => 2
2021-08-12 15:10:54.612 INFO ZWAVE: Node 64: value added: 134-0-protocolVersion => 3.83
2021-08-12 15:10:54.614 INFO ZWAVE: Node 64: value added: 134-0-firmwareVersions => 1.1
2021-08-12 15:10:54.615 INFO ZWAVE: Node 64: value added: 134-0-hardwareVersion => 1
2021-08-12 15:10:55.962 INFO ZWAVE: Node 64: value added: 128-0-level => 83
2021-08-12 15:10:55.969 INFO ZWAVE: Node 64: value added: 128-0-isLow => false
2021-08-12 15:10:56.112 INFO ZWAVE: Node 64: value updated: 128-0-level 83 => 83
2021-08-12 15:10:56.115 INFO ZWAVE: Node 64: value updated: 128-0-isLow false => false
2021-08-12 15:10:56.307 INFO ZWAVE: Node 64: value updated: 128-0-level 83 => 83
2021-08-12 15:10:56.309 INFO ZWAVE: Node 64: value updated: 128-0-isLow false => false
2021-08-12 15:10:56.428 INFO ZWAVE: Node 64: value updated: 128-0-level 83 => 83
2021-08-12 15:10:56.430 INFO ZWAVE: Node 64: value updated: 128-0-isLow false => false
2021-08-12 15:10:56.676 INFO ZWAVE: Node 64: value updated: 132-0-wakeUpInterval 0 => 0
2021-08-12 15:10:56.678 INFO ZWAVE: Node 64: value updated: 132-0-controllerNodeId 1 => 1
2021-08-12 15:11:00.664 INFO ZWAVE: Node 64: interview stage COMMANDCLASSES completed
2021-08-12 15:11:00.668 INFO ZWAVE: Node 64: interview stage OVERWRITECONFIG completed
2021-08-12 15:11:00.670 INFO ZWAVE: Node 64: interview stage COMPLETE completed
2021-08-12 15:11:00.700 INFO ZWAVE: Node 64: value added 64-128-0-level => 83
2021-08-12 15:11:00.701 INFO ZWAVE: Node 64: value added 64-128-0-isLow => false
2021-08-12 15:11:00.703 INFO ZWAVE: Node 64: value added 64-134-0-libraryType => 2
2021-08-12 15:11:00.705 INFO ZWAVE: Node 64: value added 64-134-0-protocolVersion => 3.83
2021-08-12 15:11:00.706 INFO ZWAVE: Node 64: value added 64-134-0-firmwareVersions => 1.1
2021-08-12 15:11:00.708 INFO ZWAVE: Node 64: value added 64-134-0-hardwareVersion => 1
2021-08-12 15:11:00.709 INFO ZWAVE: Node 64: value added 64-114-0-manufacturerId => 133
2021-08-12 15:11:00.711 INFO ZWAVE: Node 64: value added 64-114-0-productType => 1
2021-08-12 15:11:00.712 INFO ZWAVE: Node 64: value added 64-114-0-productId => 4
2021-08-12 15:11:00.713 INFO ZWAVE: Node 64: value added 64-132-0-wakeUpInterval => 0
2021-08-12 15:11:00.714 INFO ZWAVE: Node 64: value added 64-132-0-controllerNodeId => 1
2021-08-12 15:11:00.725 INFO ZWAVE: Node 64 ready: Unknown manufacturer 0x0085 - Unknown product 0x0004 (0x0001)
2021-08-12 15:11:00.751 INFO ZWAVE: Node 64: interview COMPLETED, all values are updated

AlCalzone, 2021-08-12:
That's not the [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs).

Kingcool88, 2021-08-12:
> That's not the [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs).

My fault! Here it is
[zwavejs_2021-08-12.log](https://github.com/zwave-js/node-zwave-js/files/6975876/zwavejs_2021-08-12.log)

AlCalzone, 2021-08-12:
Okay, that looks like you should still be able to use it even if there is no name yet.
There's not a lot you can do though - set up associations, read the battery level and that's it.
The associations have semi-nice names like "Control5" but that's just cosmetical.

AlCalzone, 2021-08-12:
If you'd like to improve this, please provide a config file:
https://zwave-js.github.io/node-zwave-js/#/config-files/file-format

It would need these fields:

- manufacturer
- manufacturerId
- label
- description
- devices
- firmwareVersion
- associations (for better labels)
- metadata (for inclusion/exclusion instructions)

Kingcool88, 2021-08-12:
> Okay, that looks like you should still be able to use it even if there is no name yet.
> There's not a lot you can do though - set up associations, read the battery level and that's it.
> The associations have semi-nice names like "Control5" but that's just cosmetical.

Ive tried to associate it in the same group (2) as the windows, but its not possible to control the windows with the remote.  The windows are fully controllabe through HA tho, just not with the ZRH 12 remotes.
So dlnt know whats wrong..

AlCalzone, 2021-08-12:
> Ive tried to associate it in the same group (2) as the windows

I'm not sure what you mean by that. You need to add the device you want to control (the shutter) into an association group of the device that should control it (the remote).

Kingcool88, 2021-08-12:
> > Ive tried to associate it in the same group (2) as the windows
> 
> I'm not sure what you mean by that. You need to add the device you want to control (the shutter) into an association group of the device that should control it (the remote).

Yeah and it doesnt work  to control the shutter with the remote after..

AlCalzone, 2021-08-12:
Did you confirm that the association was created? You need to wake up sleeping devices for that to take effect.

Kingcool88, 2021-08-12:
> Did you confirm that the association was created? You need to wake up sleeping devices for that to take effect.

Yes the windows are listed in the groups section of the remote. Still not working.