zwave-js/zwave-js #3183: Ability to assigned "labels" to endpoints

kpine, 2021-08-12:
**Is your feature request related to a problem? Please describe.**

Some devices provide many endpoints. It's not always clear what their function is, especially when these endpoints get represented in an application. A good example of this is the Aeotec Siren 6, which has 8 endpoints. Each endpoint provides some slightly different functionality. Luckily Aeotec provides good tech documents, so it's somewhat easy for a technically inclined person to determine the functionality, but this is still out of reach of must users.

In Home Assistant, with 8 endpoints you get 8 duplicate entities that interact with each endpoint. Since all we have to go by are the endpoint numbers, you get meaningless names like:

```
siren.foo
siren.foo_2
siren.foo_3
...
siren.foo_8
```

The numbers (sans the "1") should correspond to the endpoint numbers, but most users wont' have a clue how `siren.foo_2` differs from `siren.foo_8`.

I would like to propose an enhancement that might allow applications to provide some clarification about how endpoints are used for various devices.

**Describe the solution you'd like**

As suggested in the HA Discord, Z-Wave JS already has the ability to configure endpoints via device config file. Adding something like a `"label"` field to the endpoint might be useful. This could be included as part of the node state and applications could choose to name things based off of the endpoint label, as opposed to just the endpoint number itself.

**Describe alternatives you've considered**

Applications could manually customize naming on a per-device basis. This would mean each application duplicates the implementation and could have different descriptions.

**Additional context**
Here's the definition of the endpoint functionality for a Siren 6. AFAIK this info is not available from the device itself.

[image]

In the device config I might set something like this:

```json
"endpoints": {
    "1": {
        "label": "Browse"
    },
    "2": {
        "label": "Tampering"
    },
    "3": {
        "label": "Doorbell 1"
    },
    ...
    "8": {
        "label": "Emergency"
    }
}
```
And with the label, the entities could be named descriptively:
```
siren.foo_browse
siren.foo_tampering
siren.foo_doorbell_1
...
siren.foo_emergency
```

The naming is not a foolproof solution: what do "browse" and "emergency" mean? IMO, still better than plain numbers. A description field could even be added to provide more information, although I don't have a clue how something like HA would present that kind of information (an attribute of the primary entity?).

Another example is the Fibaro RGBW devices. I think they have 6 endpoints, including the root. The root controls RGB directly, other endpoints control each color channel separately, etc. Naming these would be pretty useful, e.g. `light.foo_rgbw`, `light.foo_red`, `light.foo_green`, etc.

I can't say how many devices this would be useful for, so it may be limited in application. I'll say though that a common question in HA in the past has been along the lines of "why does my node have so many duplicate entities?". Even something like a double replay would benefit to being named `switch.foo_relay_s1` and `switch.foo_relay_s2` as opposed to `switch.foo` and `switch.foo_2`.

martijndemulder, 2021-11-30:
I'm trying to migrate from Openzwave to Zwave-JS and I'm really missing these labels. For the rest Zwave-JS is awesome!
Am I overlooking something or is it still missing from the current release?

kpine, 2021-11-30:
> Am I overlooking something or is it still missing from the current release?

The feature request is still open, which means it hasn't been implemented.

I actually didn't realize OZW had this capability, but here it is: https://github.com/OpenZWave/open-zwave/blob/cb2320965e45650e509887a982d8096723ae4c8c/config/aeotec/zw164.xml#L35-L44

I didn't think it was used in HA (I can't find any code using the labels). Does your experience show otherwise?

martijndemulder, 2021-12-01:
Good find in the OZW code! This should make it much easier to implement, right?

I don't think the label is currently used in HA. In fact I don't think the usage of sirens is really supported at all.
Personally, I use the labels in MQTT-explorer for my own reference when interfacing with the Siren's/ Doorbell I've got. Being able to make use of the labels in HA would make it an awesome improvement and really a big step to more/better support for sirens.

PR #5403 (2023-02-02): feat: support assigning labels to endpoints, add to Aeotec Siren 6
fixes: #3183
closed 2023-02-02 by AlCalzone