zwave-js/zwave-js #3194: Lock entity with device 012501 Popp & Co / Electric Strike Lock Control not present

gcosta74, 2021-08-15:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The Lock entity is missing while creating the entities for the Popp Electric Stricke Lock

All other entities like binary sensor are fully available. I use one of binary sensors for the bell ringing and it is responding fine.

Without the lock entity I can not use as an electric strike lock. Was working fine in HA when used with the deprecated ZWAVE Implementation. After move to ZWAVE JS the lock is not showing.

FW Version seems to be 1.3 and the component is not fully recognized

[image]

[image]

[image]

Config capabilities are not shown since component is not identified correctly

[image]

### Device information

Manufacturer: Popp & Co (0x0154)
Model name: Electric Strike Lock (012501)
Node ID in your network: 62 (secure inclusion)

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 0.1.35

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Hope this log helps:

[zwave_js-2.log](https://github.com/zwave-js/node-zwave-js/files/6988363/zwave_js-2.log)

AlCalzone, 2021-08-15:
We seem to have no config file for this, which is why you're not seeing configuration parameters.
The Door Lock CC interview etc. seem fine, so I don't know why you're not getting entities for that.

raman325, 2021-08-15:
@gcosta74  can you share a state dump of your device? In Home Assistant, Configuration > Integrations > Z-Wave JS > Configure and then click on the link at the bottom of the page. All I care about is the lock node data but if you provide the full dump (all nodes), just let me know what the node is.

gcosta74, 2021-08-16:
> @gcosta74 can you share a state dump of your device? In Home Assistant, Configuration > Integrations > Z-Wave JS > Configure and then click on the link at the bottom of the page. All I care about is the lock node data but if you provide the full dump (all nodes), just let me know what the node is.

@raman325 Node is 62. I have attached the dump:
[zwave_js_dump.json.zip](https://github.com/zwave-js/node-zwave-js/files/6990210/zwave_js_dump.json.zip)

raman325, 2021-08-16:
thanks! Our discovery logic wouldn't work for this lock but we can fix that for the next release. PR incoming soon.

BTW - you mentioned that an HA developer told you to post here - where did that happen? I don't recall seeing an issue about this but the issue of no entity is strictly an HA problem.

gcosta74, 2021-08-16:
> thanks! Our discovery logic wouldn't work for this lock but we can fix that for the next release. PR incoming soon.
>
> BTW - you mentioned that an HA developer told you to post here - where did that [...]

Hi Raman, first of all thanks for the quick help.

In regards to my Statement I have chosen the Option "YES, BUT a Home Assistant developer has told me to come here" simply because the other Option was "NO". I was convinced it was an ZWAVE JS Issue and therefore did not pos it in HA. 

Sorry for the confusion.

raman325, 2021-08-16:
OK no worries. While I appreciate the confidence, going forward please open issues in home-assistant/core to start when you are facing issues with the integration - there are a lot of moving pieces in the zwave-js integration, and while it does lead to false positives in the core repo, we have more people responding to those issues vs here so we can help filter things down for Al.

Anyway, back to the issue, this will likely get released in 2021.9 since it's technically a new feature.

raman325, 2021-08-17:
this can be closed as the new discovery logic has been merged

closed 2021-08-17 by AlCalzone