zwave-js/zwave-js #3197: Missing device configuration: Wenzhou MTLC Electric Appliances Co.,Ltd. ZWN-BPC PIR Sensor (FW < 5.9)

aaron9060, 2021-08-16:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Wenzhou MTLC Electric Appliances Co., Ltd. / PIR Sensor

### Manufacturer ID

0x011a

### Product Type

0x0601

### Product ID

0x0901

### Firmware Version

2.3

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2165

AlCalzone, 2021-08-16:
The product you linked mentions a firmware version 5.10. Yours is 2.3.
Do you have a manual for yours to check for any differences in association groups, config parameters, etc. compared to https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2165/20170111%20ZWN-BPC.pdf?

aaron9060, 2021-08-16:
Sorry, unfortunately I don't have a manual for this device. There is some printing on the back:

Ceiling Mounted, PIR Motion Sensor
Cat.No.: ZWN-BPC
Z-Wave Frequency: 908.42mhz
FCC ID: ZZHZWN-SC7

AlCalzone, 2021-08-18:
Ok, it seems there is another version of this device by another manufacturer where we have a firmware file for versions 5.9 and below. It looks like the single config parameter changed.

Could you test that in zwavejs2mqtt?
=> https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

Put a `.json` file into that directory with the following content:
```jsonc
{
	"manufacturer": "Wenzhou MTLC Electric Appliances Co., Ltd.",
	"manufacturerId": "0x011a",
	"label": "ZWN-BPC",
	"description": "PIR Sensor",
	"devices": [
		{
			"productType": "0x0601",
			"productId": "0x0901"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "5.9"
	},
	"associations": {
		"1": {
			"label": "Lifeliine",
			"maxNodes": 5,
			"isLifeline": true
		}
	},
	"paramInformation": {
		"0": {
			"label": "Motion timeout",
			"unit": "minutes",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 1,
			"unsigned": true
		}
	}
}
```

Restart zwavejs2mqtt and re-interview your device.
Then please verify that you can change the *Motion timeout* parameter to values above 7 (test 255 to be sure).

aaron9060, 2021-08-18:
This fixed the manufacturer and product for the device and I was able to set value 255 for Motion Timeout. 

The basic sensor does go from 0 -> 255 when motion is detected. But, the binary sensor doesn't appear to be working, it is always True. 

```
18:25:48.500 CNTRLR [Node 032] treating BasicCC::Set as a report
2021-08-18 18:25:48.502 INFO ZWAVE: Node 32: value updated: 32-0-currentValue 0 => 255
18:26:12.908 CNTRLR [Node 032] treating BasicCC::Set as a report
2021-08-18 18:26:12.909 INFO ZWAVE: Node 32: value updated: 32-0-currentValue 255 => 0
```

AlCalzone, 2021-08-19:
Okay, can you please re-interview the device and upload the [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs) of that here?
Don't copy/paste, just drag the file into the textbox.

Its likely we need to add a compat flag for this problem, but I need to see what happens during the interview.

AlCalzone, 2021-08-29:
@aaron9060 ping

aaron9060, 2021-08-30:
sorry for delay, logs attached. thanks

[zwavejs_2021-08-30.log](https://github.com/zwave-js/node-zwave-js/files/7078345/zwavejs_2021-08-30.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/7078346/zwavejs2mqtt.log)

AlCalzone, 2021-09-01:
Ok, try putting this in your config file - same procedure as above:
```jsonc
{
	"manufacturer": "Nortek Security & Control LLC",
	"manufacturerId": "0x014f",
	"label": "ZWN-BPC",
	"description": "PIR Sensor",
	"devices": [
		{
			"productType": "0x0601",
			"productId": "0x0901"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "5.9"
	},
	"associations": {
		"1": {
			"label": "Reports",
			"maxNodes": 5,
			"isLifeline": true
		}
	},
	"paramInformation": {
		"0": {
			"label": "Motion timeout",
			"unit": "minutes",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 1,
			"unsigned": true
		}
	},
	"compat": {
		"enableBasicSetMapping": true
	}
}
```

AlCalzone, 2021-09-05:
@aaron9060 ping :)

markleary, 2021-09-07:
I tried the driver above and it seems to work, I just had to modify the manufacturerID.  Mine is labeled Enerwave, not sure if this device was sold under two different names but this driver correctly matched with the device and got the binary_sensor working:

```
{
	"manufacturer": "Enerwave",
	"manufacturerId": "0x011a",
	"label": "ZWN-BPC",
	"description": "PIR Sensor",
	"devices": [
		{
			"productType": "0x0601",
			"productId": "0x0901"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "5.9"
	},
	"associations": {
		"1": {
			"label": "Reports",
			"maxNodes": 5,
			"isLifeline": true
		}
	},
	"paramInformation": {
		"0": {
			"label": "Motion timeout",
			"unit": "minutes",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 1,
			"unsigned": true
		}
	},
	"compat": {
		"enableBasicSetMapping": true
	}
}
```

AlCalzone, 2021-09-07:
Thanks for testing, I'll apply that change to both devices.

PR #3294 (2021-09-07): fix(config): enable basic set mapping for ZWN-BPC variants
fixes: #3197
closed 2021-09-07 by AlCalzone