zwave-js/zwave-js #3199: Fibaro FGR223 Roller Shutter 3 no shutter level update

pquan, 2021-08-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

When operating the roller shutter via the hardware buttons (S1/S2) or via the software buttons (I'm using HA+Zwave integration, actually), I don't see the new "level" update in HA.

The spontaneous level updates from Endpoint 0, however ZWaveJS thinks they should come from endpoint 1. So the level in HA is never updated. 

I opened a bug report in HA, but was directed here as this issue was deemed a zwave-js core issue, (quote) *"I would submit a bug report 1 with the node-zwave-js project. With release v8, the behavior of multi-channel devices has changed. There are several Fibaro devices with similar problems, that either require an update to the device config file, or new associations. If you provide the logs and dumps there, they should be able to solve it."* (end quote)

Here's an example of level update resulting from operation with a hardware button:

```
2021-08-07T10:08:24.664Z CNTRLR   [Node 009] [~] [Multilevel Switch] targetValue: 59 => 55          [Endpoint 0]
2021-08-07T10:08:24.667Z CNTRLR   [Node 009] [~] [Multilevel Switch] duration: "unknown" => "unknow [Endpoint 0]
                                  n"
2021-08-07T10:08:24.671Z CNTRLR   [Node 009] [~] [Multilevel Switch] currentValue: 59 => 55         [Endpoint 0]
2021-08-07T10:08:24.674Z DRIVER « [Node 009] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation]
                                    │ sequenced: false
                                    └─[MultilevelSwitchCCReport]
                                        current value: 55
                                        target value:  55
                                        duration:      unknown
2021-08-07T10:08:25.129Z SERIAL « 0x0108000400040298402d                                              (10 bytes)
2021-08-07T10:08:25.133Z SERIAL » [ACK]               
```

The Fibaros are configured with their lifeline on endpoint 0, 1 and 2 to the Node ID 1 which is an Aeontec controller.

The network dump shows the multilevel switch on endpoint 1. I'm showing here a service update from HA and below is a network dump from the FGR

```
Log Level changed to: Verbose
2021-08-07T18:21:17.081Z DRIVER » [Node 009] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      31
                                  └─[SecurityCCNonceGet]
2021-08-07T18:21:17.098Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-08-07T18:21:17.114Z DRIVER « [REQ] [SendData]
                                    callback id:     31
                                    transmit status: OK
2021-08-07T18:21:17.134Z DRIVER « [Node 009] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xebb819a2c91a0652
2021-08-07T18:21:17.150Z DRIVER » [Node 009] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      32
                                  └─[SecurityCCCommandEncapsulation]
                                    │ nonce id: 235
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 1
                                      └─[MultilevelSwitchCCGet]
2021-08-07T18:21:17.167Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-08-07T18:21:17.185Z DRIVER « [REQ] [SendData]
[... 193 lines cut ...]

Yes (please describe)

### If yes, where did it work?

"It worked in the past". I'm not sure which update broke it.

### Attach Driver Logfile

```
2021-08-12T22:14:02.180Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2021-08-12T22:14:02.187Z DRIVER   version 8.0.8
2021-08-12T22:14:02.188Z DRIVER
2021-08-12T22:14:03.762Z CONFIG   version 8.0.8
2021-08-12T22:14:04.807Z CNTRLR   beginning interview...
2021-08-12T22:14:04.807Z CNTRLR   querying version info...
2021-08-12T22:14:04.984Z CNTRLR   received version info:
                                    controller type: Static Controller
                                    library version: Z-Wave 4.54
```

AlCalzone, 2021-08-16:
Pretty sure that was the v8 update of zwave-js that caused this. Please re-interview your device (I need to see some of that to decide what's the appropriate fix) and make a driver log of the entire interview, on loglevel "debug". Please also control the device manually once after the interview and include that in the log.

Upload the logfile here by drag&drop into the text field.

pquan, 2021-08-16:
Ok, I did as requested. I re-interviewed a couple of nodes and I noticed that the values now came from endpoint 1 and those values were then reflected in the HA entities.
So, I operated another device, which was not re-interviewed, and the values now came from endpoint 0 (the problem). 
These nodes were last re-interviewed under driver 8.0.6 (a couple of days ago), while trying to fix the issue.
Today I updated to 8.0.8, but did not re-interview.
So there was probably an issue until 8.0.6 and it seems something has changed in 8.0.8 that allows a re-interview to heal the node.
For completeness, I am attaching the operation log from a device that was causing the problem, node 7, that was last interviewed under 8.0.6 (operation7pre.txt), then the re-interview process under 8.0.8 (interview7.txt) and the operation post-interview (operation7post.txt).
The problem seems cured by updating to 8.0.8 and re-interviewing all devices.
Can you confirm this looking at the logs?
Thank you.
[operation7pre.txt](https://github.com/zwave-js/node-zwave-js/files/6995598/operation7pre.txt)
[interview7.txt](https://github.com/zwave-js/node-zwave-js/files/6995599/interview7.txt)
[operation7post.txt](https://github.com/zwave-js/node-zwave-js/files/6995601/operation7post.txt)

AlCalzone, 2021-08-17:
The update from 8.0.6 to 8.0.8 includes these changes, which are very likely to have this impact that you're seeing:
https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md#807-2021-08-02
> Improved the heuristic for lifeline associations, which should resolve some reporting issues with devices:
>
>    If the root endpoint of a device is configured to use a node association, the fallback for the other endpoints no longer creates a multi channel association on the root endpoint
>    If the endpoints of a multi channel device don't support associations, the default lifeline on the root device will be configured as a multi channel association

So, re-interviewing the problematic devices should fix this.

closed 2021-08-17 by AlCalzone