zwave-js/zwave-js #3201: Ring keypad and contact sensor not working correctly over S2

IvanBrazza, 2021-08-16:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [ ] Docker
- [ ] PKG
- [ ] Snap package
- [x] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt version: 5.4.5.0d49676
zwave-js version: 8.1.0

### Describe the bug

After trying to get the jump on adding S2 support to zwavejs2mqtt (see https://github.com/IvanBrazza/zwavejs2mqtt/commit/3eb0cda05f174fff17d38f593f10a128ba469427), I believe I have enough down to at least include an S2 enabled device - even if some parts are hard coded for testing purposes like the inclusion strategy callbacks for DSK and security classes.

The devices I'm trying to include are the [Ring Keypad v2](https://products.z-wavealliance.org/products/4150?selectedFrequencyId=1) and [Ring Contact Sensor v2](https://products.z-wavealliance.org/products/4147?selectedFrequencyId=1).

After including the devices, everything appears to add as normal except that they don't report notifications at all - the keypad doesn't report any key presses in the logs or the UI, and the contact sensor doesn't report its opened/closed state. Configuration still appears to work as I am able to set configuration options.

I noticed in the silly logs that whenever a notification occurs (i.e. I wave my hand over the keypad motion sensor, or open the contact sensor) a deserialization error message appears in the log:

`2021-08-16T13:58:20.459Z DRIVER   Dropping message because it could not be deserialized: SupervisionCCGet: deserialization not implemented`

Full logs attached of including each device with no other nodes included.
[zwavejs_2021-08-16-ring-contact-sensor.log](https://github.com/zwave-js/zwavejs2mqtt/files/6993048/zwavejs_2021-08-16-ring-contact-sensor.log)
[zwavejs_2021-08-16-ring-keypad.log](https://github.com/zwave-js/zwavejs2mqtt/files/6993049/zwavejs_2021-08-16-ring-keypad.log)

I also found the same issue occurring when using an unencrypted inclusion on server version 8.0. Except the motion sensor reports notifications correctly on an insecure connection, nothing else reports however. I assumed this was due to the lack of an S2 connection but that doesn't seem to be the case with S2 support now in 8.1.

### To Reproduce

Steps to reproduce the behavior:

1. Set DSK pin in the [validateDSKAndEnterPIN callback](https://github.com/IvanBrazza/zwavejs2mqtt/blob/s2/lib/ZwaveClient.ts#L1617)
2. Start Default inclusion
3. Enable inclusion on the device

### Expected behavior

A clear and concise description of what you expected to happen.

### Additional context

Add any other context about the problem here.

AlCalzone, 2021-08-16:
We need to support responding to `SupervisionCCGet` commands to solve this.

xhemp, 2021-08-24:
Just to add: I am seeing the same thing if I add the ring 1st gen motion sensor and the ring  1st gen keypad with S2.

marcus-j-davies, 2021-08-24:
@AlCalzone,
I think its worth re-titling this issue?

@IvanBrazza , @xhemp 
The S2 paring process with ring actually works, and by definition, S2 is working with ring.
This is evident on the fact you can still control the device whilst under S2.

However,
Some devices, especially those that may be geared towards security, actually make use of the Supervision CC, Supervision is not fully implemented yet in Z-Wave JS.

But why does that matter?
It's on the assumption, the Supervision CC is used _more_ with S2 devices then those not using S2.

`Dropping message because it could not be deserialized: SupervisionCCGet: deserialization not implemented`
Actually occurs with some wall dimmers and switches, because Supervision just _seems_ to be more in use with S2 paired devices.

I have motion sensors, door sensors that work in S2 - but only if they are not using the Binary Report Mode, as that mode wants to use Supervision - not sure why, but they do.

Hopefully this helps, when you come across other devices that may not be sending events through.

Ikcelaks, 2021-08-24:
This is probably going to be a huge breaking issue for many (maybe most or all) devices that send notifications, because the best practices suggest always using the SupervisionCC when communicating with S2, because it prevents delay attacks and gives the sending node confirmation that the receiving node was able to decrypt and process the command.

https://www.silabs.com/documents/public/application-notes/APL13434-FAQ-%20S0-S2-Supervisio-CC-product-design-developments.pdf

IvanBrazza, 2021-08-24:
@marcus-j-davies makes total sense and I was coming to that same realisation the more I researched into it after raising this issue.

I've made a start on implementing the SupervisionCCGet deserialisation - I've managed to get far enough that it will now deserialise the message and execute a Notification Report (if that's the encapsulated command) which will update the value. I'm now struggling to get a Supervision Report sent back to the node in response, but this is my first foray into zwave and node so I'm slowly working my way through it.

Any pointers from anyone would be much obliged 😉

marcus-j-davies, 2021-08-24:
@AlCalzone is the Key holder for the driver,  and if you're making progress, it will be a great help to him and many others here.
I'm sure Al will have some answers if you get stuck.

AlCalzone, 2021-08-24:
@IvanBrazza I think your main point of interest is going to be here:
https://github.com/zwave-js/node-zwave-js/blob/b1f0bce76ece13506f4b68c4df5478f780d188da/packages/zwave-js/src/lib/driver/Driver.ts#L2377

We're also going need a way to remember how (with which CCs) a command was encapsulated after calling this:
https://github.com/zwave-js/node-zwave-js/blob/b1f0bce76ece13506f4b68c4df5478f780d188da/packages/zwave-js/src/lib/driver/Driver.ts#L2553
Not only for Supervision, but it's useful in other situations. Then you can react in line 2377ff if the command was supervision encapsulated.

marcus-j-davies, 2021-08-25:
@IvanBrazza,

Do you want to raise a PR with your latest work on this CC implementation and put it to draft?
once you're happy with it, set it to review stage.

`SupervisionCCGet`, is gaining attention, and I think it will be good for those  needing it (its growing), to monitor its progress.

This will allows those that know the internals off the driver, to help out?

dansrogers, 2021-09-03:
This also affects https://aeotec.com/hidden-z-wave-door-sensor/  If I pair in S2 mode, I do not get updates from the device.  Logging indicates that Supervision CC messages are being dropped.

raidboy, 2021-09-04:
+1 on the issue. Not sure how i could use HA with zwavejs2mqtt and S2 without this being resolved.
HA OS on RPI4 6.2,  , zwavejs2mqtt 0.24.1 

Can reproduce problem with Zooz ZEN30, ZEN34 included with S2_UNAUTHENTICATED, e.g.:
pressing button on ZEN34, debug log from zwavejs2mqtt:

2021-09-04T17:15:36.927Z CNTRLR Failed to execute controller command after 1/3 attempts. Scheduling next try i
n 100 ms.
2021-09-04T17:15:37.189Z DRIVER Dropping message because it could not be deserialized: SupervisionCCGet: deser
ialization not implemented
2021-09-04T17:15:37.222Z DRIVER Dropping message with invalid payload
2021-09-04T17:15:37.399Z DRIVER Dropping message because it could not be deserialized: SupervisionCCGet: deser
ialization not implemented

direct associations work, but wanted to build central scenes via HA.

marcus-j-davies, 2021-09-04:
@raidboy,
The implementation for this portion of the Supervision CC is very much in progress.
It takes time to get it right, as the driver is very much in favour of (and rightfully so) following specification.

you can track its progress with the linked PR: https://github.com/zwave-js/node-zwave-js/pull/3240

raidboy, 2021-09-04:
Thank you very much. I do not have the knowledge or time to test early versions, but will certainly be very happy to re-try whenever there is a new version available through the web-based update process and report on it.

PR #3240 (2021-09-09): feat: support reacting to SupervisionCC::Get
Will eventually close #3201
closed 2021-09-09 by AlCalzone