zwave-js/zwave-js #3222: Handle S2 keys not being defined

blhoward2, 2021-08-23:
Confirm how we handle S2 keys not being defined. Contract: August lock included securely under the present UI attempts S2 inclusion by default, but doesn't fallback to S0. This makes it impossible to include locks until S2 is added to the UI. 

```2021-08-22T23:50:06.736Z CNTRLR   [Node 006] supports Security S2, but no network key was configured. Continuing
                         interview non-securely.
```

AlCalzone, 2021-08-23:
To be clear: The device is included with S0, no S2 keys are configured and the **interview** skips S0, correct?

blhoward2, 2021-08-23:
The device is being included using the current function call for secure inclusion. It seems to attempt to use S2 for inclusion but the right keys aren’t there. Instead of falling back to S0, it falls back to insecure. Problem is that its a lock.

blhoward2, 2021-08-23:
@firstof9 duplicated on his setup. Can you confirm I have this right?

AlCalzone, 2021-08-23:
Ok I need to see a log of this.

firstof9, 2021-08-23:
Here's a log for you:
<details>
<summary>Log file</summary>

```
2021-08-22T23:40:43.550Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2021-08-22T23:40:43.557Z DRIVER   version 7.0.0
2021-08-22T23:40:43.558Z DRIVER   
2021-08-22T23:40:43.558Z DRIVER   starting driver...
2021-08-22T23:40:43.576Z DRIVER   opening serial port /dev/zwave
2021-08-22T23:40:43.620Z DRIVER   serial port opened
2021-08-22T23:40:43.621Z SERIAL » [NAK]                                                                   (0x15)
2021-08-22T23:40:43.633Z SERIAL « 0x01100004001e0a320221740000000000009a                              (18 bytes)
2021-08-22T23:40:43.637Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:43.638Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:43.643Z SERIAL « 0x0114000400080e3202a1640001d179000000000000b5                      (22 bytes)
2021-08-22T23:40:43.645Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:43.646Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:44.242Z SERIAL « 0x0110000400200a320221340000004a0000ae                              (18 bytes)
2021-08-22T23:40:44.245Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:44.246Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:45.132Z DRIVER   loading configuration...
2021-08-22T23:40:45.258Z SERIAL « 0x01140004002a0e600d0100320221320000001e00009a                      (22 bytes)
2021-08-22T23:40:45.263Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:45.264Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:46.654Z SERIAL « 0x01140004002a0e600d0200320221320000001e000099                      (22 bytes)
2021-08-22T23:40:46.657Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:46.658Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:46.658Z SERIAL « 0x01140004002a0e600d0200320221320000001e000099                      (22 bytes)
2021-08-22T23:40:46.660Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:46.661Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:46.665Z SERIAL « 0x01140004002a0e600d0300320221320000001e000098                      (22 bytes)
2021-08-22T23:40:46.668Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:46.668Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:46.671Z SERIAL « 0x01140004002a0e600d0400320221320000001e00009f                      (22 bytes)
2021-08-22T23:40:46.673Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-08-22T23:40:46.674Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:47.418Z DRIVER   beginning interview...
2021-08-22T23:40:47.419Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2021-08-22T23:40:47.420Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2021-08-22T23:40:47.420Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2021-08-22T23:40:47.421Z CNTRLR   beginning interview...
2021-08-22T23:40:47.421Z CNTRLR   querying version info...
2021-08-22T23:40:47.438Z SERIAL » 0x01030015e9                                                         (5 bytes)
2021-08-22T23:40:47.440Z DRIVER » [REQ] [GetControllerVersion]
2021-08-22T23:40:47.445Z SERIAL « [ACK]                                                                   (0x06)
2021-08-22T23:40:47.450Z SERIAL « 0x011001155a2d5761766520332e3935000199                              (18 bytes)
2021-08-22T23:40:47.451Z SERIAL » [ACK]                                                                   (0x06)
2021-08-22T23:40:47.453Z DRIVER « [RES] [GetControllerVersion]
                                    payload: 0x5a2d5761766520332e39350001
2021-08-22T23:40:47.464Z CNTRLR   received version info:
                                    controller type: Static Controller
                                    library version: Z-Wave 3.95
2021-08-22T23:40:47.465Z CNTRLR   querying controller IDs...
2021-08-22T23:40:47.481Z SERIAL » 0x01030020dc                                                         (5 bytes)
2021-08-22T23:40:47.483Z DRIVER » [REQ] [GetControllerId]
[... cut ...]
                                    controller supports timers: false
                                    nodes in the network:       1, 6, 8, 9, 10, 14, 15, 16, 17, 18, 19, 20, 21, 
                                  22, 23, 26, 27, 28, 29, 30, 31, 32, 33, 36, 37, 38, 39, 41, 42, 43, 44, 45, 46
                                  , 47
2021-08-22T23:40:49.213Z DRIVER   Failed to initialize the driver: Invalid input format!
2021-08-22T23:40:49.216Z DRIVER   destroying driver instance...
```
</details>

AlCalzone, 2021-08-23:
Okay, but that confirms what I wrote above. This is not about the inclusion, this is about the interview afterwards.

firstof9, 2021-08-23:
I'm unsure as I had to roll back to allow my secure nodes to operate again.

AlCalzone, 2021-08-24:
The log message is a bit misleading and the one for S0 is never printed. Maybe your S0 key is actually missing from the config?
The fact that S2 is mandatory for a "Fan Switch" leads to further confusion because there is no mention of S2 in the NIF.

Other than that, I can't get this to reproduce. I've tried these with z2m 5.4.6 without S2 support:
* a node that is included insecurely
* a node that ONLY supports S0
* a node that supports S0 and S2 but was included using S0

The last two log the same message about missing S2 keys, but interview just fine.

firstof9, 2021-08-24:
I've tested with 5.4.6 and everything works as expected.
I will try updating to the latest z2m again and see what happens.

firstof9, 2021-08-24:
Seems there may be a side effect with S2. My MeterCC on my LZW36 are having problems being discovered.
I've attached the re-interview log, `Node 043`.

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7042022/zwave_js.log)

AlCalzone, 2021-08-24:
I think this is unrelated. The rest of the communication seems fine, but Meter CC **is** having problems - only for the value queries though:
```
2021-08-24T20:39:44.227Z CNTRLR   [Node 043] Interviewing Meter...
2021-08-24T20:39:44.228Z CNTRLR » [Node 043] querying meter support...
2021-08-24T20:39:44.238Z DRIVER » [Node 043] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      80
                                  └─[MeterCCSupportedGet]
...
2021-08-24T20:39:44.397Z DRIVER « [Node 043] [REQ] [ApplicationCommand]
                                  └─[MeterCCSupportedReport]
                                      type:                 Electric
                                      supports reset:       true
                                      supported scales:
                                      · kWh
                                      · W
                                      supported rate types:
2021-08-24T20:39:44.405Z CNTRLR » [Node 043] querying meter value (type = Electric, scale = kWh)...
2021-08-24T20:39:44.418Z DRIVER » [Node 043] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      81
                                  └─[MeterCCGet]
                                      scale: kWh
2021-08-24T20:39:44.431Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-08-24T20:39:44.506Z DRIVER « [REQ] [SendData]
                                    callback id:     81
                                    transmit status: OK
...
2021-08-24T20:39:54.516Z CNTRLR   [Node 043] Timed out while waiting for a response from the node

...etc.
```

firstof9, 2021-08-24:
Thanks for the insight Al

AlCalzone, 2021-08-24:
FYI, there are 2 fresh PRs
1. Misleading log messages: https://github.com/zwave-js/node-zwave-js/pull/3229
2. Incorrect detection of security classes after migration: https://github.com/zwave-js/node-zwave-js/pull/3228

AlCalzone, 2021-08-25:
Until someone can come up with an actual repro, I think this is solved with 8.2.0

closed 2021-08-25 by AlCalzone