zwave-js/zwave-js #3224: Yale / Keyfree Connected Lock 0129 not reporting state

DirkFace, 2021-08-23:
I've removed and re-added the zwave device, re-interviewed, healed, and even changed the state manually in the developer tools - but for some reason my lock has stopped reporting the current state !!! It worked fine up until very recently.

I am able to send it commands via zwave to lock / unlock, or even put into passage mode and they respond correctly. It reports a ping when requested too.

If I look at the device in Alexa, it reports as jammed.

Some of my automations rely on wether the lock state is locked, but unfortunately it now renders them useless.

Help!
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/7033972/zwavejs2mqtt-store.zip)

AlCalzone, 2021-08-24:
Check if your security keys are set up correctly. Node 10 reliably acknowledges the commands but does not respond to any GET-type requests. It is very likely that it cannot decrypt them.

DirkFace, 2021-08-24:
Thanks, make sense. 

Could you point me in the direction of how to check the security keys please? 

I've tried Google but to no avail. 

King regards ...

Sent from my iPhone.

AlCalzone, 2021-08-24:
Depends on the software you use. It should have options to configure the network key(s). Make sure that the configured one is the same that was used to include the device.

DirkFace, 2021-08-24:
I'm using Z-Wave JS to MQTT in the supervisor and Z wave JS in integrations (url pointing at ..zwavejs2mqtt:3000).
I can see my network key in MQTT settings. I haven't changed it since using Zwave for both my locks.
The node 6 works fine, but node 10 seems to have glitched somewhere!

What you suggest as the cause for node 10 not report the current state makes perfect sense. Unfortunately my Home assistant skills are pretty basic, so verifying the network key for node 10 is beyond my knowledge at present.

AlCalzone, 2021-08-24:
If both have worked securely before and now only node 6 does, there is something wrong with node 10 and the key is likely correct. Otherwise, there is no way to retroactively retrieve the key from the device.

You can try re-interviewing Node 10 and see if that fixes it. If not, you'll probably have to re-include the device.

DirkFace, 2021-08-24:
I re-interviewed Node 10 a few hours ago to see if this would rectify the problem, but it hasn't.

Is there any way I can see in the node 10 logs to check if it has a network key?

AlCalzone, 2021-08-24:
> Is there any way I can see in the node 10 logs to check if it has a network key?

During the interview zwave-js uses the configured network key to find out if the node was included with Security. If it received the list of securely supported commands, the node has the same one. If not, it probably doesn't.