zwave-js/zwave-js #3227: Better allow for node transition from S0 to S2

blhoward2, 2021-08-24:
The zwave spec does not provide for a way to migrate an already included device from S0 to S2. To do so you'd have to exclude/include, which would change the node numbering, device naming, etc. You can't just reset the device and replace the failed node, because many devices will remove themselves from the database when they are reset locally.

In consultation with @AlCalzone, we propose to add a property to the node/driver that would ignore the next DeviceResetLocallyNotification from that node. That would allow the UIs to present a workflow similar to:

1. User selects a node to migrate
2. UI disables excluding the device on next local reset
3. If they exist, reset instructions are displayed from the device config file
4. User resets device locally following the instructions
5. UI checks for the failed node, and after the stick marks it failed, initiates replace failed node
6. User is prompted to reinclude the device

In addition to adding the property, we need to modify this routine to skip it if the property is set.

https://github.com/zwave-js/node-zwave-js/blob/b1f0bce76ece13506f4b68c4df5478f780d188da/packages/zwave-js/src/lib/driver/Driver.ts#L2375-L2394

blhoward2, 2021-08-24:
@robertsLando @cgarwood @raman325 @marcus-j-davies Thoughts?

raman325, 2021-08-25:
seems like a reasonable approach to me, I am not familiar with `DeviceResetLocallyNotification` but your explanation makes sense

robertsLando, 2021-08-25:
I could add this to the nodes manager dialog in my case. It's ok for me

marcus-j-davies, 2021-08-25:
Seems simple enough.
Might be one of those that will become obsolete however, but sure, this will be helpful.

robinsmidsrod, 2021-10-10:
> **Note:** In the interim, an eager user can power down zwavejs, reset the device, power back up, then check failed node, and replace failed node.

This workaround mentioned actually doesn't work (for me on Heatit ZDim). It goes along until you type in the DSK, and then it just ends up hanging, and eventually time out on S2 negotiation and then reverting to `None` security. If I unpair and re-pair properly it correctly manages to use S2_Authenticated security.

AlCalzone, 2021-10-10:
Have you tried resetting it twice before re-including? I remember that some locks need that.

robinsmidsrod, 2021-10-10:
@AlCalzone I have not tried double reset, but now I have done the tedious work of getting all my S2-compatible devices excluding and re-included, so I'm not really inclined to test it out... Thought I'd just mention it, for future reference.