zwave-js/zwave-js #3253: Nortec GD00Z-4 garage door opener not inclusing as secure node and therefore cannot control it

pdawg17, 2021-08-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

When I try to incluse the opener, it gets added as a node but with security "None". Therefore, I cannot control the opener since it is not a secure node.

What do you observe?

When starting inclusion, whether I select "force security" or not, it gets added but with security "None"

What did you expect to happen?

This opener does support s0, so I expected it to be added as s0 legacy.

Steps to reproduce the behavior:
  1. Go to inclusion
  2. Click on start
  3. Click button on opener which starts the inclusion
  4. Screen says "node xx added with security "none"
 

### Device information

Manufacturer: Nortek
Model name: GD00Z-4
Node ID in your network:  currently node 30

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.2.1
`zwavejs2mqtt` branch: 5.5.1

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2021-08-30.log](https://github.com/zwave-js/node-zwave-js/files/7079686/zwavejs_2021-08-30.log)

IN LOG FILE UPLOADED, NODE 29 IS WITH SECURITY FORCED AND NODE 30 IS WITHOUT

kpine, 2021-08-30:
@AlCalzone I was helping with this problem. I am pretty sure this device doesn't actually support S2, even though it reports it  during the interview, so maybe that's the issue.

kpine, 2021-08-31:
@pdawg17 A user reported [here](https://community.home-assistant.io/t/device-randomly-loses-security-network-key-and-no-longer-able-to-add-securely/334336/10?u=freshcoast) that including the device in HA (with secure option) worked, so you might try that.

In HA 2021.8 (with server v1.10.1) the inclusion in the above post will [use](https://github.com/zwave-js/zwave-js-server/blob/35c2f5385eb08f780d7a00ab042376d1286a254f/src/lib/controller/message_handler.ts#L252) strategy  `InclusionStrategy.Security_S0`. Compare that to zwavejs2mqtt, which _I believe_ always uses strategy `InclusionStrategy.Default` but sets the option `forceSecurity` to true or false.

So perhaps the difference is that HA still only supports explicit S0 or Insecure inclusions, but zwavejs2mqtt uses the new Default inclusion strategy and the `forceSecurity` flag is not working due to the misreporting of S2 support?

AlCalzone, 2021-09-01:
It looks like in both cases, inclusion using S2 is attempted, but the device doesn't react to the key exchange commands. If inclusion using S0 only works, then I'd agree with your assessment.

/cc @robertsLando maybe we do need an S0-only inclusion option in z2m

AlCalzone, 2021-09-01:
@pdawg17 honestly, you should contact the manufacturer and report this. I wonder if they have many users that actually use S2.

robertsLando, 2021-09-01:
Question is whouldn't this be handled automatically when using Default strategy and force security is true?

AlCalzone, 2021-09-01:
After we've tried S2 (which fails here), we can't try S0 anymore. So the decision must be done before attempting to do security bootstrapping, before any command is sent to the device.

kpine, 2021-09-01:
Would removing the Security S2 class via a compat flag for this device change the inclusion behavior so that it would skip S2 and use S0 for the Default strategy?

AlCalzone, 2021-09-01:
Sadly no. Security bootstrapping must happen before we can know which device it is.

dansrogers, 2021-09-03:
This workaround no longer works.  I just tried to add the device in home assistant's integration using the "use secure inclusion" checkbox" and it did not work.  The following log file captures the behavior, but it appears to be the same as described above.  S2 is attempted, the device doesn't respond to S2 negotiation, then refuses to communicate insecurely.  This makes me very sad, because I had them working, but excluded them to readd them to get improved security, and instead I got a broken device.
[zwavejs_2021-09-02.zip](https://github.com/zwave-js/node-zwave-js/files/7103515/zwavejs_2021-09-02.zip)

I have 3 garage door openers.  2 are a -4, but one is a -7.  The -7 has proper S2 support with a device id printed on in and work fine in S2 mode.

We need a way to specify, in advance, not to use S2 security, and to really force S0.

AlCalzone, 2021-09-03:
> We need a way to specify, in advance, not to use S2 security, and to really force S0 for devices with broken or incompatible s2 support.

The driver supports that, but I initially told @robertsLando to only offer the modes you currently see. Looks like we need to expose the S0 only mode too.