zwave-js/zwave-js #3254: Schlage locks not working after conversion from zwave2mqtt [question]

hi-flyer, 2021-08-31:
I have upgraded from zwave2mqtt to zwavejs2mqtt. Blinds are working fine (e.g they have no security). My schlage locks are not working. While zwavejs2mqtt knows they are there, the commands to actuate the lock are not there. I noticed on upgrade it picked up the security parameter that zwave2mqtt used but zwavejs2mqtt did not like (it was in red). I remove the 0x and spaces between the parameters to make such that the software seemed to like it but seemed to still be the same security code. But no luck and the interview process still does not come up with other than undefined commands. Do I need to re-connect the 3 locks to get them to work again. Two of the lock are BE469ZP the third is BE469

Thanks for any suggestions.
ps have attached 2 log files where I interrogated the 23, 24 and 28 which are the 3 schlage locks.
[zwavejs2mqtt.log](https://github.com/zwave-js/zwavejs2mqtt/files/7078608/zwavejs2mqtt.log)
[zwavejs_2021-08-30.log](https://github.com/zwave-js/zwavejs2mqtt/files/7078610/zwavejs_2021-08-30.log)

marcus-j-davies, 2021-09-01:
I'm not an expert at deciphering the logs/zwave2mqtt , and this may not be the reason.

But these stand out a mile.

```
2021-08-30T17:52:26.294Z DRIVER   Unexpected error during SerialAPI command: ZWaveError: The network key for the
                                   security class S2_Unauthenticated has not been set up yet!
```

AlCalzone, 2021-09-01:
Looks like some of the S2 keys are not configured and this causes problems during the interview for the S2 capable devices. The problem **should** go away if you provide a key for each security class and re-interview. The driver needs to catch this.

hi-flyer, 2021-09-01:
Thank you all for your comments. It turns out that I needed 2 different keys. The S2 Authenticated and S2 Access Control keys. I also needed the 5 digit S2 code for each of the S2 locks. If I am quick enough on entering the info, the locks sync up.

Any clue what the purpose of the S2 Unauthenticated key is for (as I have not set one)?

marcus-j-davies, 2021-09-01:
S2 'partitions' security modes, to. better protect the network.

A key for S2 - Access Control.
A key for S2 - Authenticated.
A key for S2 - Unauthenticated.

AlCalzone, 2021-09-01:
@hi-flyer you should enter all the keys. Authenticated is basically Unauthenticated plus PIN verification (ensures that you include the correct device and not one of an attacker in the next room). Access Control is like Authenticated but with a different key to avoid compromising the entire network when one of the other keys is hacked:
https://zwave-js.github.io/node-zwave-js/#/getting-started/security-s2?id=granting-security-classes

AlCalzone, 2021-09-01:
> If I am quick enough on entering the info, the locks sync up.

You should have 4 minutes for the entire process (minus a few seconds for the data exchange) - if you really need to be much quicker, the device isn't following the specs.

hi-flyer, 2021-09-01:
Thanks to all. The information is appreciated. 
And A big thank you for all of your hard work to provide an excellent piece of software.

AlCalzone, 2021-09-01:
I'll keep this open. We should fail more gracefully.

PR #3265 (2021-09-02): fix: handle missing S2 keys more gracefully
fixes: #3254
closed 2021-09-02 by AlCalzone