zwave-js/zwave-js #3266: Trane XR524 setting incorrect paremeters

tmjpugh, 2021-09-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
- Configuration Parameters use illegal values
- Configuration Parameters do not match device settings
- When Configuration Parameters are changed, they revert back to illegal value

What do you observe?
device>>configuration>>112-0-1
Options in zwavejs are

    0 Gas/Electric (default)
    1 HeatPump

option in Trane 524 menu are

    none
    AC
    HP

if i select AC on Trane, [1]HeatPump is selected in zwaveJS
if i select [1]HeatPump in zwaveJS, AC is slected on Trane

ALSO
112-0-18 Cooling Delta Stage 1 ON equal 18
I change value to 3 and receive message in log that show value is change From 3 to 18
value is change back from 3 to 18 in zwavejs UI as well
8 is maxValue for this setting

Refresh does not change these result and appears to cause device setting on device to become 0 for value that exceed maxValue

What did you expect to happen?
I expect zwavejs value to change to similar option on device Gas/Electric to AC or Heatpump to HP
I expect value change in zwaveJS not to revert OR not to allow change to illegal value

Steps to reproduce the behavior:
  1. Go to 'node'
  2. Click on 'node >> Configuration's
  3. Scroll down to 'parameter 18'
  4. change data for parameter and watch it revert to 18, this will occur for many parameter

### Device information

Manufacturer: Trane
Model name: XR524
Version: 1.38 and 1.14
Node ID in your network: 93 and 58

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.2.1
`zwavejs2mqtt` branch: 5.5.2
`docker image` zwavejs/zwavejs2mqtt:5.5.2

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2021-09-02.log
](https://github.com/zwave-js/node-zwave-js/files/7096803/zwavejs_2021-09-02.log)

AlCalzone, 2021-09-02:
Any chance you have a manual for your device?

tmjpugh, 2021-09-02:
Available here but I’ll upload if needed

https://www.nexiahome.com/wp-content/uploads/2015/01/Trane524UserGuide.pdf

cfrudolphy, 2021-09-03:
I am seeing some of the same issues.  However my main concern is the current_temperature: 24 this is in Celsius and it needs to be in Fahrenheit.  I re-interviewed the node and the log is attached below along with some screen shots.
[image]
[image]
[image]
[zwave_js_TraneXR525_Re-interview.log](https://github.com/zwave-js/node-zwave-js/files/7108699/zwave_js_TraneXR525_Re-interview.log)
If you need any more info please let me know.

AlCalzone, 2021-09-03:
> However my main concern is the current_temperature: 24 this is in Celsius and it needs to be in Fahrenheit

You might need to configure preferred scales so zwave-js knows which units to request. I'm not sure where it can be done in HA though.

cfrudolphy, 2021-09-03:
Have configured scales in HA and on the Trane XR524 device itself.  Both set to "F" for Fahrenheit.

cfrudolphy, 2021-09-03:
You asked in past comment to the OP if he had a manual for this device.  I didn't see where you got one.  Here it is attached.  Hope this helps.
[Trane524UserGuide.pdf](https://github.com/zwave-js/node-zwave-js/files/7108763/Trane524UserGuide.pdf)

tmjpugh, 2021-09-03:
> Have configured scales in HA and on the Trane XR524 device itself. Both set to "F" for Fahrenheit. Also note that is the one value only that is in Celsius. All other values are in Fahrenheit and match what the XR524 displays on its screen. Also the XR524 displays the current or air temperature in Fahrenheit on its screen. The only place I have this issue is on my HA dashboard with the standard HA/Lovelace Thermostat card.

You need  to set Preferred Scales in ZwaveJS UI >> Settings >> Zwave >> Preferred Scales >> select "temperature: Fahrenheit (F)"

after this, you may go to the device in ZwaveJS UI >> Control Panel >> Trane Device  >> Values >> [refresh] the Multilevel sensor

I use zwavejs in docker so if using the addon I'm not sure how that works or how to set preferred scales there

cfrudolphy, 2021-09-04:
I am running the ZwaveJS add-on in a Home Assistant OS installation.  Are these configuration options available in the ZwaveJS2MQTT management console?  If so i will switch to ZwaveJS2MQTT.  If not I will try to find or wait for another solution.

blhoward2, 2021-09-04:
@tmjpugh 

>112-0-18 Cooling Delta Stage 1 ON equal 18
I> change value to 3 and receive message in log that show value is change From 3 to 18
>value is change back from 3 to 18 in zwavejs UI as well
>8 is maxValue for this setting

Do you have any idea what the correct range is for this value? The manual does not provide the zwave parameters so what we have was reverse engineered by people in the past and taken from other projects.

cfrudolphy, 2021-09-04:
@tmjpugh 

> Do you have any idea what the correct range is for this value? The manual does not provide the zwave parameters so what we have was reverse engineered by people in the past and taken from other projects.
> 
> The interview could help us here. Can you post a Silly level log of the interview? In HA you would go to the logs in integration in one window, then in another trigger a reinterview. Download and post.

Please see the previous posts.  Once post has a log of the re-interview and another post has the owners manual.  Starting about page 7 or 8 and different settings and the prescribed ranges for each.

Regards
Chuck

tmjpugh, 2021-09-04:
> Do you have any idea what the correct range is for this value? 

I believe I may not have been clear.  All Configuration Value  for this device 112-0-1 to 112-0-24 are using incorrect parameter or range. I only add some item in my original post as example

This Value/Range are listed in manual, added to previous post as upload and link

> The interview could help us here. 

Just in case i attach another Silly log showing nodes 58 and 93, the Trane XR524 thermostats

Thank you for quickly looking at this.  I am able to make this correction but not familiar with device specific config file location in zwavejs to make the change.  Is it possible I make the changes locally then provide changes or github merge? Not sure if this is easier for you or if you have other task.

[zwavejs_2021-09-04.log](https://github.com/zwave-js/node-zwave-js/files/7110556/zwavejs_2021-09-04.log)

blhoward2, 2021-09-04:
It evidently doesn't, though. The zwave parameters may have different units. Doesn't it says param 18 is 0-8, which is what we had? And the device rejected that and set 18?

blhoward2, 2021-09-05:
Ok, here is what I can do. The ranges in the manual make no sense compared to the previously reverse engineered numbers. I have no idea how to model .5-8, as the values are whole numbers. Is 1 a 10th degree? Something else? You tried to set 3 and it failed so it can't be 1 to 1. It is possible its tenths and 5 is the minimum number (0.5). Separately, the schedule parameters have very large numbers. What does 100681294 mean? 

I'm pushing a PR that sets all of these to the highest range permitted by their value size. Someone that has the device will need to do more testing and deduce what numbers mean what, so we can formally fix the file and display the correct unit for each parameter. I did flip the HeatPump/AC ones based on your testing.

AlCalzone, 2021-09-05:
> Are these configuration options available in the ZwaveJS2MQTT management console? If so i will switch to ZwaveJS2MQTT

They are (settings -> Z-Wave -> Preferred scales), but I'm not sure if they carry over to HA.

xannor, 2021-09-05:
I added a comment to the PR but I realize I should repeat it here, (found the PR before I saw this issue):

There are two other "broken" parameters for the xr524, and have been so for a while (as per https://community.home-assistant.io/t/trane-xr524-with-control-of-the-schedule-run-hold-esm-modes/94461)

The 11-112-0-76 is not the fan off time, but the toggle for the Schedule, with 0 = Hold, and 1 = Run.
The 11-112-0-77 is not the 2nd Stage heat but the toggle for ESM (Economy Mode), with 0 = Off and 2 = Run. Not sure why 2 but it works on my unit. 

I don't know if the two older options are available under another parameter or if they were swapped, but I think this happened because the 524 settings were copied from the 624 and nobody every corrected them.

blhoward2, 2021-09-05:
I can make that change. No way around it, this is going to be a community sourced file. I don't have one of these to try to test it so you all are going to have to report changes.

xannor, 2021-09-05:
As for the schedule parameters, I found (at one time) a doc that explained them, but I cannot remember where. I believe they are an encoding of the start time and high low temps for that day and "type" [Morning-1, Afternoon-2, Evening-3, Night-4]. I will try to see if I can find that info again and provide it here.

xannor, 2021-09-05:
I run the zwave js 2 mqtt addon for home assistant. If there is a way to test or patch in the changes to a running system I am willing to try, but I dont have a spare system to setup to play with.

blhoward2, 2021-09-05:
If they're that complex, it's unlikely we can model them in parameters and we'll have to let it a number. But if you can find it I'll have a look.

xannor, 2021-09-05:
I can attest that if I tell the system to manually set a parameter value, 76 will toggle the schedule and 77 will toggle the ESM when set to the values I stated above (prior to jwave js). and if I toggle the settings on the until those values will reflect in the config reported by zwave-js. I cannot set the ESM currently though because that value is out of range for the current config.

AlCalzone, 2021-09-05:
> If they're that complex, it's unlikely we can model them in parameters

Might just be fixed bitmasks so we might be able to model them as partial params.

xannor, 2021-09-05:
I think I see how this got confused/merged with the 624. There is an American Standard xr624, listed on the zwave alliance website, that looks identical (mostly some minor cosmetic differences) and has the same product id and type id listed in their DB, also there is an RCS XL524 (with no picture) but again the same product/type info. The actual model of my unit is TZEMT524AA21MA, which when google searched everyone calls a XR524, but it is not listed on the z-wave alliance site at all. 

It seems this is a generic unit that has been "rebranded" by the major A/C MFGs here in the US.

xannor, 2021-09-05:
I had a theory and tested it with mine. My first schedule entry value was 103629900 and the schedule was 7:00am H68 C76. I then changed it to 6:45am (rest the same) and the value became 103629900. I then used Numbers (I had an air book handy so that was the easier spreadsheet option) and converted the numbers to binary. I saw a pattern of 8 bits for the hour, 8 bits for the minutes, 8 bits for the heat point and 8 bits for the cool point. I then built some formulas, plugged in the other values from the sunday schedule and could reverse all of them.

so it looks like it is a 32bit integer of hour(24hr), minute, heat point and cool point as 8 bit chunks left to right. I dont know if/how you can express this in the config though.

AlCalzone, 2021-09-05:
> so it looks like it is a 32bit integer of hour(24hr), minute, heat point and cool point as 8 bit chunks left to right. I dont know if/how you can express this in the config though.

We can - as 4 separate [partial parameters](https://zwave-js.github.io/node-zwave-js/#/config-files/partial-parameters). You should then be able to set them 1-by-1 or in bulk, depending on what the application supports.

xannor, 2021-09-11:
I believe it is just a raw value, I used Fahrenheit because that is what I am used to. Since unit changes are uncommon, I would think it assumes the value is always the current unit. I cannot verify though.

blhoward2, 2021-09-11:
Can you set a schedule, then change parameter 4 and see if the number changes?

xannor, 2021-09-11:
Changing the units did change the reported config values, however changing it back is "corrupting" the values (they changed again but not back to the original reported value,) also the configured setpoints did not change. The change back seems to have "confused" zwave-js as it is still reporting the current temp in celsius. The unit its self is correct though, so I dont know if changing the units affected what the console reported or how zwave-js interpreted it. I didnt check the schedule on the unit after the change to see what value it had. But now I need to restart the system in the hopes that zwave js will start reading the correct values or my automations will go nuts.

blhoward2, 2021-09-12:
If not, try reinterviewing it.

AlCalzone, 2021-09-12:
> The unit its self is correct though, so I dont know if changing the units affected what the console reported or how zwave-js interpreted it.

If you could capture that in a driver log, we should be able to figure this out.

tmjpugh, 2021-09-13:
upgraded to zwave JS V 5.5.4
after reinterview all sensor values were off greatly and device showed as unknown.  went to check on unit and found unit was Blank screen with small flicker in screen.  power cycle did not clear this
I removed it and left on table about 1hr then reattached. Unit came back but sensor showed 0.  
Reset device and all back.

tmjpugh, 2021-09-17:
if anyone is working on this. I figure out many values but cannot determine parameter for the "Comfort Setting" menu items. I believe these may be in the parameters 19-24 but no matter how i try i could not confirm

currently I use the below
parameters 1, 4, 6, 9, 11, 12, 14, 15, 75, 76, 77 are verified

```
{
	"manufacturer": "Trane Corporation",
	"manufacturerId": "0x008b",
	"label": "Trane XR524",
	"description": "Touchscreen Comfort Control Thermostat",
	"devices": [
		{
			"productType": "0x5452",
			"productId": "0x5442"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"paramInformation": {
		"1": {
			"label": "Outdoor Unit Type",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 2,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "NONE",
					"value": 0
				},
				{
					"label": "AC",
					"value": 1
				},
				{
					"label": "HP",
					"value": 2
				}				
			]
		},
		"2": {
			"label": "Fan Type",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Gas",
					"value": 0
				},
				{
					"label": "Electric",
					"value": 1
				}
			]
		},
		"3": {
			"label": "Change Over Type",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
[... 504 lines cut ...]
		},
		"253": {
			"label": "Schedule Sat-2",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 0,
			"defaultValue": 0
		},
		"254": {
			"label": "Schedule Sat-3",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 0,
			"defaultValue": 0
		},
		"255": {
			"label": "Schedule Sat-4",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 0,
			"defaultValue": 0
		}
	}
}
```

blhoward2, 2021-09-17:
Please don't post a massive json file as text. Please remove it and upload the file as an attachment.

blhoward2, 2021-09-18:
@tmjpugh Could you compare whatever you changed to the PR at #3277 and see if anything is still off?

tmjpugh, 2021-09-18:
@blhoward2 the changes to scheduling may be OK.  I will check them If I get time tonight or this weekend.  I never use scheduling feature so I need to determine how to check it.

parameters 1, 4, 6, 9, 11, 12, 14, 15, 75, 76, 77 are all complete change from original and verified with 2 software version of XR524 unit ; labels and values are completely different. In these case the PR changes appear to be mostly incorrect, for example I confirm Parameter 15 is "Sensor Calibration"

Honestly original .json appear to be mostly wrong so I feel best to start with file I post and add to that as new items are confirmed. I believe I confirm all parameter in that file but I only document eleven parameters listed above so I will verify again at later time and document all verified parameter for clarity.

tmjpugh, 2022-01-04:
@blhoward2 
Sorry reply this late but I want to confirm changes below were made.  I could not find these any change request for these.

Config Parameters  1, 4, 6, 9, 11, 12, 14, 15, 75, 76 and 77 required change.
I confirm value and function and proper value is listed in attached file.  I only see you change 76 already but did not see change for other value or change was incomplete.

I did also confirm value and proper function of parameters 2, 3 ,10, 13, 29, 30,  78, 79, 141, 143 and 148.  These may be changed from values in zwavejs config file.

Config parameters 228-244, 246-255 I did not confirm but they not cause any issue for me so I believe they OK to remain.
These show as scheduling items in config file
 
Config parameters 16-21, 24 and 25 should be removed.
I was unable to verify these function and in some case they cause incorrect/unexpected function and may cause AC unit damage if not removed.

I am currently using custom trane_xr524.json file based on above.  This has been attached as .txt file

[trane_xr524.txt](https://github.com/zwave-js/node-zwave-js/files/7805230/trane_xr524.txt)

AlCalzone, 2022-01-04:
@tmjpugh are you able to add your changes to the updated file from https://github.com/zwave-js/node-zwave-js/pull/3277/files#diff-9642212e8c22dd32a7b5a6daf8f015a37b19a504d16d1293dab94e6e91a09d23

The above one is pretty outdated by now and has an old file format.

tmjpugh, 2022-01-04:
@AlCalzone i possible can but I believe I would need some permission access or simply do know how to do that.
If you can provide some guidance there I am willing to add.

cweakland, 2022-02-14:
> I added a comment to the PR but I realize I should repeat it here, (found the PR before I saw this issue):
>
> There are two other "broken" parameters for the xr524, and have been so for a while (as [...]

I have a 524 and a 624, and I can tell you that these parameters (11-112-0-76 and 11-112-0-77) are mislabeled/missing on the 624. The 524 looks very complete now, would it be possible to copy the 524's config over to the 624 (and keep the humidity sensor on the 624)? 

> There are two other "broken" parameters for the xr524, and have been so for a while (as per https://community.home-assistant.io/t/trane-xr524-with-control-of-the-schedule-run-hold-esm-modes/94461) [...]

I have a xr524 and a xl624 in operation. I can confirm that the xr524 now looks very complete, however, the xl624 is lackluster. The main difference between these two units is that the 624 has a humidity sensor, otherwise they are identical. Would it be possible to copy the trane_xr524's config over to the xl624? 

Thanks again!

AlCalzone, 2022-02-18:
> I can confirm that the xr524 now looks very complete

Thanks for confirming (and the PR). I'll close this issue then.

closed 2022-02-18 by AlCalzone

cfrudolphy, 2022-02-18:
Has this been committed to the latest distribution?

Chuck Rudolphy
346.383.6920
Mobile from my Pixel 4A

LenirSantiago, 2023-09-12:
> Have configured scales in HA and on the Trane XR524 device itself. Both set to "F" for Fahrenheit. Also note that is the one value only that is in Celsius. All other values are in Fahrenheit and [...]

This issue is still happening as of HA 2023.9 and HASSOS 10.5, has there been any updates to this?

LenirSantiago, 2023-09-12:
> > Have configured scales in HA and on the Trane XR524 device itself. Both set to "F" for Fahrenheit. Also note that is the one value only that is in Celsius. All other values are in Fahrenheit and match what the XR524 displays on its screen. Also the XR524 displays the current or air temperature in Fahrenheit on its screen. The only place I have this issue is on my HA dashboard with the standard HA/Lovelace Thermostat card.
> 
> You need to set Preferred Scales in ZwaveJS UI >> Settings >> Zwave >> Preferred Scales >> select "temperature: Fahrenheit (F)"
> 
> after this, you may go to the device in ZwaveJS UI >> Control Panel >> Trane Device >> Values >> [refresh] the Multilevel sensor
> 
> I use zwavejs in docker and conect with HA via websocket so if using the addon I'm not sure how that works or how to set preferred scales there

Does anybody know how to do change preferred scales in ZwaveJS in HA? perhaps via a config file?

AlCalzone, 2023-09-12:
> This issue is still happening

I'll need to see a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) on loglevel `debug` of when it happens. Attach it here as a file (drag & drop into the text field).

> Does anybody know how to do change preferred scales in ZwaveJS in HA?

Your best bet for an answer is to ask this on the [HA Discord](https://discord.com/invite/home-assistant).

LenirSantiago, 2023-09-12:
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12588164/zwave_js.log)

Logs attached.

LenirSantiago, 2023-09-12:
This is what I see in Home Assistant:
[image]
[image]
[image]

AlCalzone, 2023-09-12:
Ok the log confirms what @kpine explains. There is no preferred scale and Z-Wave JS's heuristic isn't good enough to look at other temperature values to figure out which scale to use. So the first query of the air temperature is going to be in °C.

LenirSantiago, 2023-09-13:
> Ok the log confirms what @kpine explains. There is no preferred scale and Z-Wave JS's heuristic isn't good enough to look at other temperature values to figure out which scale to use. So the first query of the air temperature is going to be in °C.

So HA allows me to set the Scale for the Air Temperature Sensor... Can that be used somehow?

[image]

AlCalzone, 2023-09-13:
No that info isn't mirrored back to node-zwave-js

LenirSantiago, 2023-09-13:
> You can't set preferred scales in Home Assistant. If you need this functionality, you'll have to switch to Z-Wave JS UI.
> 
> You can try forcing ZJS to request the temperature in F. Here's a service call you can copy and paste into Dev Tools Services (switch to YAML mode):
> 
> ```yaml
> service: zwave_js.invoke_cc_api
> data:
>   command_class: "49"
>   method_name: get
>   parameters:
>     - 1  # sensorType: Air temperature
>     - 1  # scale: F
> target:
>   entity_id: climate.z_wave_thermostat
> ```
> 
> Replace target with your device. After sending this command, ZJS should default to F for future requests, as long as the device responds with the value in F and doesn't report values in C later on.

The service call worked, it updated the current temperature in F. 
Now i'm waiting for the thermostat to update the temperature to see if it shows up correctly in HA.

LenirSantiago, 2023-09-13:
> No that info isn't mirrored back to node-zwave-js

Since preferred scales can be configured although only ZwaveJS UI, maybe this should show up as a configurable ZwaveJS parameter in HA:
1) Maybe user configurable setting in ZwaveJS configuration 
2) or maybe ZwaveJS using System->General->Unit System Settings in HA

[image]

LenirSantiago, 2023-09-14:
> The service call worked, it updated the current temperature in F. Now i'm waiting for the thermostat to update the temperature to see if it shows up correctly in HA.

I can confirm that a  day later it's still working after sending the service call that one time.

abillits, 2024-06-26:
> You need to set Preferred Scales in ZwaveJS UI >> Settings >> Zwave >> Preferred Scales >> select "temperature: Fahrenheit (F)"
>
> after this, you may go to the device in ZwaveJS UI >> Control Panel [...]

Thank you so fricking much! I was struggling with this issue for a month.