zwave-js/zwave-js #3267: S2 Status seems to get lost after a FW update - results in bad re-interview

marcus-j-davies, 2021-09-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

As discussed,

After a FW update, with a device that was enrolled with **S2 Authenticated.**
the re-interview seems to go wrong.

checking the log myself (this is on behalf of a user) it shows:
`2021-09-02T19:58:45.930Z CNTRLR « [Node 009] The node was NOT granted the security class S2_Authenticated`

### Device information

Device:
https://devices.zwave-js.io/?jumpTo=0x0371:0x0004:0x00bd:0.0

Node:
9

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
**node-zwave-js** 8.2.1:

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Just a FW update

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[firmware-update.txt.zip](https://github.com/zwave-js/node-zwave-js/files/7101786/firmware-update.txt.zip)

AlCalzone, 2021-09-03:
I **think** this is it:
```
2021-09-02T19:58:35.450Z DRIVER « [Node 009] [REQ] [BridgeApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 208
                                    └─[FirmwareUpdateMetaDataCCStatusReport]
                                        status:    OK_RestartPending
                                        wait time: 30 seconds
2021-09-02T19:58:35.454Z CNTRLR « [Node 009] Firmware update completed with status OK_RestartPending
2021-09-02T19:58:35.456Z CNTRLR   [Node 009] Firmware updated, scheduling interview in 30 ms...
```

We start the interview while the device is still rebooting.

marcus-j-davies, 2021-09-03:
I (and other users) have done many FW updates, is there a special case for S2?.
or have we always jumped the gun with re-inetrviewing, and its now showing its negative affect under S2 perhaps?

AlCalzone, 2021-09-03:
No, the other devices probably didn't need 30 **seconds** so the interview doesn't time out.

marcus-j-davies, 2021-09-03:
So `waitTime` should also be honorued  before an interview starts?

AlCalzone, 2021-09-03:
It already is. But a by factor of 1000 too short.

PR #3280 (2021-09-05): fix: interpret waitTime after firmware update as seconds
fixes: #3267
closed 2021-09-05 by AlCalzone