zwave-js/zwave-js #3270: Validate that multicast messages don't target S0 devices

mhutch, 2021-09-03:
**Is your feature request related to a problem? Please describe.**

I wasn't aware S0 didn't support multicast, so I was attempting to multicast to a set of devices that included S0 devices.

If all the devices were S0, zwave-js returned an error that was both incorrect and unhelpful: "This virtual node does not support the Command Class Thermostat Mode!".

If one of the devices was unsecured, and the rest were S0, zwave-js sent the multicast message *exactly* as requested, including the node IDs of the S0 devices in the message. The S0 devices of course ignored it, but z-wave js incorrectly updated the local values regardless. This lack of validation is inconsistent with the behavior of returning an error when all the devices are S0.

Basically, if multicast messages target S0 devices it will either silently do the wrong thing, or return an incorrect error message.

**Describe the solution you'd like**

If any of the devices targeted in a multicast call use S0, return an error "Multicast messages cannot target nodes with S0 security". 

**Describe alternatives you've considered**

n/a

**Additional context**

I was attempting to set up multicasting in Home Assistant 2021.9.1 using zwavejs2mqtt 5.5.1 (z-wave js 8.2.1) and for some reason only one of the devices was responding.

The driver log showed that the multicast message was getting sent exactly as requested:

```
2021-09-03T04:41:56.945Z DRIVER » [REQ] [SendDataMulticast]
                                  │ target nodes:     45, 44, 47, 46, 43, 49, 48
                                  │ transmit options: 0x25
                                  │ callback id:      31
                                  └─[ThermostatModeCCSet]
                                      mode: Heat
2021-09-03T04:41:56.949Z SERIAL « [ACK]                                                                   (0x06)
2021-09-03T04:41:56.965Z SERIAL « 0x0104011401ef                                                       (6 bytes)
2021-09-03T04:41:56.966Z SERIAL » [ACK]                                                                   (0x06)
2021-09-03T04:41:56.967Z DRIVER « [RES] [SendDataMulticast]
                                    was sent: true
```

After some trial end error I figured out that the device that was responding was the only one that didn't have S0.

So I wondered if the problem was mixing non-secured and S0 nodes in the multicast, and messaged the S0 devices only, and got the following error in the Home Assistant log 

```
zwave_js_server.exceptions.FailedZWaveCommand: Z-Wave error 32: This virtual node does not support the Command Class Thermostat Mode!
```

When I tried to troubleshoot this by searching "zwave multicast S0", it was hard to find relevant results, and I ended up reading the S0 *spec*... Validation would have made it *much* easier to diagnose the problem.

AlCalzone, 2021-09-03:
We had a similar discussion that went in the direction of silently doing the right thing. With S2, multicast is an entirely different beast (multiple groups with their own nonces to be aware of). One idea was to have the driver segment the multicast into multiple messages:
- S2 group A
- S2 group B
- ...
- unencrypted nodes

We could use this opportunity to also convert multicasts to S0 nodes into singlecasts and send them all in sequence. That way, you'd issue a single multicast to a number of nodes and the driver would figure out which commands are necessary for that.

blhoward2, 2021-09-03:
I don't recall having this exact convo, but maybe. I think that's right. I think if given a multicast call we should try to send it in as few multicast groups as we can and make it just work.

closed 2023-03-07 by AlCalzone