zwave-js/zwave-js #3273: GE/Jasco Model #14318 Zwave Switch will not update Home Assistant UI when operated manually

cfrudolphy, 2021-09-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe?

What did you expect to happen?

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error
GE/Jasco Model #14318 Zwave Binary Switch does not update the UI when operated manually.  When HA operates the switch it updates the UI with it's current state (on/off) but if I operate the switch manually it does not update the UI.  I have been working with @blhoward2 in the forum on this and I have attached at "Silly" log of the re-interview of the switch and I also operated the switch manually while capturing, 1 on tap and 1 off tap.  Any questions and you can email me at cfrudolphy@gmail.com.
[zwave_js (1).log](https://github.com/zwave-js/node-zwave-js/files/7108150/zwave_js.1.log)

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 
GE/Jasco
Model #14318
Node ID: 7

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
Driver Version: 8.1.1
Server Version: 1.10.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have just switched over from a hybrid solution.  I was running Homeseer and the mcsMQTT plugin.  Then I had Home Assistant running the Mosquitto MQTT add-on.  I configured all of my Zwave Devices to subscribe and post to topics in the broker going in both directions.  I decided to get rid of Homeseer and MQTT and go with the ZwaveJS add on.  

When these switches (1 have 3 of the them and they all exhibit the same behavior in HA) were included and operated in Homeseer with their Zwave implementation they updated instantly upon operation whether manually or by Homeseer.  So yes it worked and updated instantly but with Homeseer's Zwave implementation.  It works in HA but again when manually operated it doesn't update the UI.

When I was running it through the MQTT broker:
Homeseer -> mcsMQTT -> Mosquitto Broker -. Home Assistant it also updated instantly in HA.

### Attach Driver Logfile

[zwave_js (1).log](https://github.com/zwave-js/node-zwave-js/files/7108189/zwave_js.1.log)

blhoward2, 2021-09-03:
I edited the description a bit. For further context, our device file currently does not define any association groups, so I'm curious whether the Lifeline group is getting setup appropriately. If it is, either the device isn't truly zwave plus or we're missing something.

blhoward2, 2021-09-03:
@cfrudolphy So that we have a complete picture, can you also upload a silly log of what (if anything) comes across when you operate the switch locally?

cfrudolphy, 2021-09-03:
See the attached.  1 tap "On"  and 1 tap "Off".
[zwave_js (2).log](https://github.com/zwave-js/node-zwave-js/files/7108385/zwave_js.2.log)

AlCalzone, 2021-09-03:
According to the interview, the device supports Z-Wave+, has a single association group and the controller is part of that. So the association part should be correct.

In the second log, I see that the device sends two NIFs with a few seconds inbetween. I only see a report after the device is polled for its current status.

Maybe it has no return route to the controller - try healing the node.

cfrudolphy, 2021-09-03:
Attached are two log files.  One is when it healed the device.  The other is when I locally operated the switch.  1 did 1 tap up, 1 tap down, 1 tap up, 1 tap down.  Let me know if you need anything else.
[zwave_js_local_operation.log](https://github.com/zwave-js/node-zwave-js/files/7108557/zwave_js_local_operation.log)
[zwave_js_heal_device.log](https://github.com/zwave-js/node-zwave-js/files/7108558/zwave_js_heal_device.log)

blhoward2, 2021-09-03:
@AlCalzone Doesn't the compat flag "manualValueRefreshDelayMs" suggest we should be refreshing on the NIF? Also, why are you working on holiday?

AlCalzone, 2021-09-03:
I wonder why the device even has an association group if it doesn't use it to report. Maybe some configuration needs to be set? 

If not, I'd argue it isn't really ZW+ (although it answers that it is) and we need to remove the CC to get the auto-refresh behavior.

blhoward2, 2021-09-03:
There are no parameters for this. It's a pretty feature-less switch. I'll push a PR removing the CC later.

closed 2021-09-06 by AlCalzone

nmpu, 2024-01-03:
According to the **Z-Wave Alliance** [website](https://products.z-wavealliance.org/Search/Index?regionId=-1&searchText=14318), there have been 3 'certifications' for this device. My switches are labeled 14318-3 with date codes 19xx. Firmware is 5.24. My 3 switches update as expected even without:

```
"compat": {
  "commandClasses": {
    "remove": {
	// Removes the Z-Wave Plus CC as this device fails to send state updates. It sends NIFs, which are only refreshed for non-plus devices
	"Z-Wave Plus Info": { "endpoints": "*" }
    }
  }
}
```
Did @cfrudolphy specify his hardware revision? The date code is on the front lower right. The actual model number is back left. The firmware version is available in Z-Wave JS UI. I see 3 possibilities:

1) Chuck's switches are different HW/FW revisions
2) Something else has changed in Z-Wave JS
3) Chuck's switches are somehow defective

All 3 revisions show the 'Plus' logo. Is this yet another [blatant example](https://github.com/jascoproducts/firmware/issues/119) of bogus 'certification'?

blhoward2, 2024-01-03:
Why are you bumping a closed issue thread from two years ago?

nmpu, 2024-01-03:
Sorry to 'bump' a 'settled' issue. I'm just trying to understand how this works. It looks like a single person can report an issue and you make changes without any consensus? Isn't the 'lowest common denominator' a slippery slope? The average user can't evaluate the severity of a Plus device being demoted to non-Plus. What exactly are the implications? Do any of the other **compat** mechanisms give a visual indication that something is broken when it probably isn't?
[image]

blhoward2, 2024-01-03:
A single person can report an issue and we will investigate the issue, including reviewing log files. If we agree that it is a actual technical issue that is best addressed in the driver versus them working with the manufacturer to fix it, we will then work around the issue, yes. The maintainer makes the call. 

You can always run zwavejs-ui and run your own device file if you disagree with the compact flag. That's a supported feature.

blhoward2, 2024-01-03:
Also worth noting, I'm not sure that flag does what you think it does. There shouldn't be any real issue to you as a result. If your device calls in an update on its own any polling would be cancelled.

PR #3274 (2021-09-06): fix(config): remove Z-Wave Plus CC from GE 14318
Closes #3273