zwave-js/zwave-js #3275: Aeotec ZWA012 interview marked complete before complete, and does not resume on wakeup

blhoward2, 2021-09-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
While diagnosing an issue, I noticed that if the ZWA012 goes to sleep mid-interview, it does not resume on subsequent wakeups. Additionally, zjs2mqtt displays that the interview is complete.

### Device information

Manufacturer: Aeotec
Model name: ZWA012
Node ID in your network: 85

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.2.2
`zwavejs2mqtt` branch: 5.5.3

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### Attach Driver Logfile

```
3:41:39.777 SERIAL » 0x011000a9015504700c0004250000000088c3 (18 bytes)
13:41:39.777 DRIVER » [Node 085] [REQ] [SendDataBridge]
│ source node id: 1
│ transmit options: 0x25
│ route: 0, 0, 0, 0
│ callback id: 136
└─[ConfigurationCCInfoGet]
parameter #: 4
13:41:39.783 SERIAL « [ACK] (0x06)
13:41:39.783 SERIAL « 0x010401a90152 (6 bytes)
13:41:39.784 SERIAL » [ACK] (0x06)
13:41:39.784 DRIVER « [RES] [SendDataBridge]
was sent: true
13:41:39.797 SERIAL « 0x011d00a98800000100a57f7f7f7f01010300000000020100007f7f7f7f7f18 (31 bytes)
13:41:39.797 SERIAL » [ACK] (0x06)
13:41:39.798 DRIVER « [REQ] [SendDataBridge]
callback id: 136
transmit status: OK
13:41:49.803 CNTRLR [Node 085] Timed out while waiting for a response from the node
13:41:49.804 CNTRLR « [Node 085] received information for parameter #4:
parameter name: Visual LED indications
value format: BitField
value size: 1 bytes
min value: undefined
max value: [object Set]
default value: [object Set]
is read-only: false
is advanced (UI): false
has bulk support: false
alters capabilities: false
13:41:49.805 CNTRLR » [Node 085] querying parameter #5 information...
13:41:49.809 SERIAL » 0x011000a9015504700e0005250000000089c1 (18 bytes)
13:45:31.982 SERIAL « 0x011c00498455160407015e858e595586725a8773809f30705b71846c7a8b (30 bytes)
13:45:31.984 SERIAL » [ACK] (0x06)
13:45:31.985 DRIVER « [Node 085] [REQ] [ApplicationUpdateRequest]
payload: 0x8455160407015e858e595586725a8773809f30705b71846c7a
13:45:31.986 CNTRLR « [Node 085] Received updated node info
13:45:31.987 CNTRLR [Node 085] The node is now awake.
13:45:32.989 CNTRLR » [Node 085] Sending node back to sleep...
```

AlCalzone, 2021-09-05:
Any chance you got a complete log?

blhoward2, 2021-09-05:
Yes, I'll post it later.

thk-socal, 2021-09-05:
We be curious if this fixes it: #3271

closed 2021-09-16 by blhoward2