zwave-js/zwave-js #3289: MCO HOME MH-C221 defined as switch (not as a cover)

faustinio, 2021-09-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The relay is defined as a switch, not as a relay for controlling curtains (electric motors)

What do you observe?
[image]

What did you expect to happen?
After adding a relay, it is defined as a relay for controlling curtains. See https://devices.zwave-js.io/?jumpTo=0x015f:0x0301:0x1001:0.0

### Device information

Manufacturer: McoHome Technology Co., Ltd.
Model name: MH-C221
Node ID in your network: 30,31,32,33,34,35,36

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

This worked with the previous version of the Home Assistant ZWave integration without any problem

### Attach Driver Logfile

  "label": "MH-C221",
            "interviewAttempts": 1,
            "endpoints": [
              {
                "nodeId": 30,
                "index": 0,
                "installerIcon": 6656,
[... 479 lines cut ...]
                "isSecure": false
              },
              {
                "id": 50,
                "name": "Meter",
                "version": 3,
                "isSecure": false
              },
              {
                "id": 122,
                "name": "Firmware Update Meta Data",
                "version": 4,
                "isSecure": false
              }
            ],
            "interviewStage": "Complete",
            "deviceDatabaseUrl": "https://devices.zwave-js.io/?jumpTo=0x015f:0xc221:0x5102:3.3",
            "statistics": {
              "commandsTX": 28,
              "commandsRX": 36,
              "commandsDroppedRX": 0,
              "commandsDroppedTX": 0,
              "timeoutResponse": 0
            }
          }

AlCalzone, 2021-09-06:
What you posted is not a logfile.

Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs), loglevel `debug` or `silly` and **attach** (via drag&drop) it here as a file.

faustinio, 2021-09-06:
@AlCalzone There you are:
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7117611/zwave_js.log)

AlCalzone, 2021-09-06:
Ahh, this is one of **those** devices 🙄

```
2021-09-06T19:18:10.220Z CNTRLR « [Node 030] node info received
                                  supported CCs:
                                  ...
                                  · Multilevel Switch
                                  ... and once more for good measure:
                                  · Multilevel Switch

... but later:
2021-09-06T19:18:11.195Z DRIVER « [Node 030] [REQ] [ApplicationCommand]
                                  └─[VersionCCCommandClassReport]
                                      CC:      Multilevel Switch
                                      version: 0
2021-09-06T19:18:11.204Z CNTRLR   [Node 030]   does NOT support CC Multilevel Switch (0x26)
```

AlCalzone, 2021-09-06:
@zwave-js-bot add compat flag to 0x015f:0xc221:0x5102:3.3
```
		"commandClasses": {
			"add": {
				"0x26": {
					// This is a roller shutter, but it responds to the CC version request with 0
					"isSupported": true,
					"version": 1
				}
			}
		}
```

AlCalzone, 2021-09-06:
@faustinio PR is up and should be released as part of tonight's config release

PR #3292 (2021-09-06): fix(config): force Multilevel Switch CC support for MCOHome MH-C221
fixes: #3289

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-09-06 by AlCalzone