zwave-js/zwave-js #3295: Pairing with S2 giving different interview results from pairing without security

crxporter, 2021-09-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Pairing with S2 enabled causes the device to return extra binary switch endpoints. 

What do you observe?
The device should have "Binary Switch" on endpoints 5 and 6 only. This is how things are working when pairing without security enabled, I only see EP 5 and 6 for binary switch.

What did you expect to happen?
I would expect the interview to have the same results without regard to security class

I've done the following in the attached log file:
1- start up the driver
2- include the device with S2 as node number 70

(view the extra binary switch endpoints)

3- exclude the device
4- include the device 

(observe only 5 and 6 binary switch endpoints)

Screenshots from nodered are included here
S2 pair:
[image]

Non secure pair:
[image]

### Device information

Manufacturer:  Fibaro
Model name: Smart inject
Node ID in your network: 70 ... then 71.

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

node-red-contrib-zwave-js 5.1.2

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_07sept21.log](https://github.com/zwave-js/node-zwave-js/files/7124568/zwave_07sept21.log)

AlCalzone, 2021-09-08:
Looks like the driver incorrectly uses the node's secure command list to query the individual endpoints when it should be asking the endpoints for their secure commands.

AlCalzone, 2021-09-08:
@crxporter can you test https://github.com/zwave-js/node-zwave-js/pull/3298 please? The bot should post an install command for an updated version there shortly.

Please re-interview the device (re-inclusion isn't necessary) and make another log.

crxporter, 2021-09-08:
@marcus-j-davies has made a "special" version for me to test this. I'll get it done in a few hours once I'm back home from work today.

Thank's for the quick help! I'll provide the new log later today!

crxporter, 2021-09-08:
Alright - it's worked!

I've got a new log for you. It has this device as Node 72. In the log I did the following:

1- include the device with S2, wait for the first interview
2- restart the system, interview again
3- update to the beta including #3298 
4- restart the system, interview

The final interview returned only the expected Binary Switch endpoints and it was *much faster* than the previous interviews.

[zwave_08sept21.log](https://github.com/zwave-js/node-zwave-js/files/7131471/zwave_08sept21.log)

And the log, as promised. Please let me know if I can provide any more information.

crxporter, 2021-09-08:
One minor note that might be important: the final interview did show "interview failed" before showing "interview complete"

Once it was "complete" everything looked as expected.

marcus-j-davies, 2021-09-08:
> One minor note that might be important: the final interview did show "interview failed" before showing "interview complete"
> 
> Once it was "complete" everything looked as expected.

You can safely ignore that one.
I really only should react if it was the final attempt, before handing it off to userland

AlCalzone, 2021-09-08:
I've noticed that too. S2 is a bit better in terms of communication. When non-secure or S0 commands couldn't be decoded, the nodes just silently didn't respond. Now the nodes tell us about the decoding error and that causes the interview to abort instead of dropping the requests. We should be a bit more forgiving here.

Also the device exhibits a communication pattern that's not entirely in line with the specs - and that causes the error originally:
```
# zwave-js requests a value:
2021-09-08T19:49:18.008Z DRIVER » [Node 072] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      181
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 24
                                    └─[BinarySwitchCCGet]

# The device responds with a corrupted command:
2021-09-08T19:49:18.034Z DRIVER « [Node 072] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode

# zwave-js creates a new nonce and sends it to the device, which should now repeat the last command with its own nonce attached
2021-09-08T19:49:18.057Z DRIVER » [Node 072] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      182
                                  └─[Security2CCNonceReport]
                                      sequence number:  25
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0xfea6c15ac5476e9256e64b3aa271deaa

# The node instead sends only its own nonce
2021-09-08T19:49:18.082Z DRIVER « [Node 072] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceReport]
                                      sequence number:  130
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x7bbae16bd0c1c2996bad179f2ca11d29

# which zwave-js interprets as "hey, I didn't understand your last command" and sends it again with another new nonce
2021-09-08T19:49:18.118Z DRIVER » [Node 072] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      181
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 26
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x970bcc9e3a86e7f40bd5a7daa8189bce
                                    └─[BinarySwitchCCGet]

# causing the node to send yet another nonce
2021-09-08T19:49:18.195Z DRIVER « [Node 072] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceReport]
                                      sequence number:  131
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x2354b541b675941fb82866218a86bef9
```
and at that point zwave-js gives up.

We should probably use the nonce in return to our nonce to set up a new shared encryption state and just request the switch state without attaching a new nonce.

But: I don't think the device is behaving correctly in line with the specs here. This might be worth reporting to Fibaro.

marcus-j-davies, 2021-09-08:
Yup,

In the case of the UI, I am passing the interview error state (causing the UI to say, "it broked" - without taking into account the other 4 retries).

I'm surprised I didn't catch this before

AlCalzone, 2021-09-08:
So, IMO the PR is good to go. I've raised issues for the other things.

crxporter, 2021-09-08:
I believe I've found a firmware update for the fibaro. Flag me in another issue if you want to see results of another interview after updating.

AlCalzone, 2021-09-08:
You can update?! I thought they require you to buy their Home Center stuff to do updates.

crxporter, 2021-09-08:
Yeah so it seems like I was wrong about that.

Just saw a forum post about the update then joined their forum thinking I could download it...

false alarm I think

PR #3298 (2021-09-08): fix: endpoint interview for Security S2
With this PR, endpoint interviews for the S2 CC correctly use multi channel encapsulation and only use the highest security class the node has been granted.

fixes: #3295
closed 2021-09-08 by AlCalzone